espressif/esp-idf #4067: Arrow keys do not work in menuconfig (VSCode terminal & Windows Terminal Preview)

kaofishy, 2019-09-15:
## Environment

- IDF version (run ``git describe --tags`` to find it): 
   v4.1-dev-256-g9f145ff16
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:    Windows 10

## Problem Description

When using `idf.py menuconfig` in a Visual Studio Code integrated terminal or a Windows Terminal Preview terminal, the arrow keys do not work as menu navigation. This does not occur under regular Windows terminal.

### Expected Behavior
Arrow keys navigate the menu items

### Actual Behavior
The menu does not respond to arrow keys

### Steps to repropduce

1. Install Visual Studio Code or Windows Terminal Preview
2. Run `idf.py menuconfig` in a project
2. Use arrow keys to navigate the menu

versamodule, 2019-09-16:
Works for me.

kaofishy, 2019-09-16:
> Works for me.

Did you try it with Windows Terminal Preview https://github.com/microsoft/terminal and the latest git IDF? I am using the CMake Python-based menuconfig and not the old lxdialog menuconfig.

versamodule, 2019-09-16:
Did you move you mouse onto the terminal window and left click first 
before you tryed to use the cursor keys.

Sincerely,

Jimp@Versamodule.com

kaofishy, 2019-09-16:
Yes of course, and this occurs under both VSCode's integrated terminal and the Windows Terminal Preview, but not under regular conhost.exe command prompt.

Interestingly if I build the IDF under WSL then the arrow keys work under both terminals, which suggests to me it's something to do with `mconf-idf` under Windows since the Windows mconf is built differently from the other platforms (from what I can gather from the repo it uses pdpcurses instead of ncurses).

kaofishy, 2019-09-16:
After some more tests, I've definitively isolated the issue to PDCurses on Windows when using a non-conhost.exe terminal. I compiled demo programs for PDCurses and I am able to reproduce the same issue: arrows work under regular conhost.exe but not any other terminal on Windows.

I used MinGW and source from this git repo https://github.com/wmcbrine/PDCurses

kaofishy, 2019-09-18:
I have created a pull request to fix this issue in kconfig-frontend, see https://github.com/espressif/kconfig-frontends/pull/1

TinusScheppers, 2019-10-20:
Is there any workaround for this?

versamodule, 2019-11-18:
I'm new to the VSCode thing, so I many not clearly understand. But I ran into this issue too. What fixed it for me was to set the integrated terminal to bash. 
"terminal.integrated.shell.windows": "C:\\Program Files\\Git\\bin\\bash.exe"

Devenda, 2019-12-29:
After updating to the latest version of esp-idf, even the rendering in vscode is broken.
Even though the menu is hard to use, the j and k keys seems to navigate the menu correctly.

Hoping more is known about this, a new issue is made about both issues: #4555 

Currently using bash is not really feasible in my setup.