espressif/esp-idf #4069: [BLE Mesh] How to add custom information to the broadcast package of BLE mesh during provisioning?

zhanlej, 2019-09-16:
Because IOS can't get the mac address of the device during the provision scan, I want the unprovisioned device to add mac information or other custom information to the broadcast package during the provsioning to facilitate the APP to filter the broadcast packet.

In the BLE demo, I can implement custom information on the broadcast by modifying the value of the ``p_manufacturer_data`` member in the ``adv_data`` variable. So can I achieve similar functions in BLE mesh?

WCCWCC, 2019-09-16:
Hi zhanlej,

The data format of the broadcast packet is well defined. You can fill the MAC address of the device into the UUID field.

Currently the esp32 device is also filled like this.

[image]

zhanlej, 2019-09-17:
Hi WCCWCC.

Thanks!

Alvin1Zhang, 2019-09-17:
@zhanlej Thanks for reporting, has your issue been resolved? Thanks.

zhanlej, 2019-09-17:
@Alvin1Zhang  YES

zhanlej, 2019-09-17:
@WCCWCC Hi,

Now the no-provision node can get the mac when scanning. But after the provision, I still don't know the mac of the node when I scan the proxy node.

Campou, 2019-09-25:
Hi @zhanlej 

After the unprovisioned device is provisioned, the format of advertising are changed. And in this situation, Mesh Proxy Advertising is going to be used. 

During Mesh Proxy Advertising, the previous 60s (maximum) is advertised with Node Identity, and after that Network ID is going to be advertised. Please see the attached files for the formats of two advertising packets.

[image]
[image]

Thanks.

Alvin1Zhang, 2020-01-15:
@zhanlej Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

AlpinProgrammer, 2020-01-22:
Hello !
I need to add extra information to the advertising as well. However, filling this data into the device UUID field is not suitable for us.

How is it possible to add the manufacturer specific data (0xFF) to the advertising mesh ?
I did not spot any mesh function in esp-idf-v4.0-rc able to do that. Is it possible to do so by directly modifying esp-idf ?

Thanks in advance !

stingzhan, 2022-05-30:
> Hello ! I need to add extra information to the advertising as well. However, filling this data into the device UUID field is not suitable for us.
>
> How is it possible to add the manufacturer specific [...]

Any updates on this issue？