espressif/esp-idf #4070: Ble ISSUE: BT_APPL: BTA got unregistered event id 32

syu-source, 2019-09-16:
I am working on esp32 ble, with ESP32 DevKitC V4 board.

below is code with problems: 

void ble_init(){
	esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
	esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT()	;
	ERROR_CHECK(esp_bt_controller_init(&bt_cfg));
	ERROR_CHECK(esp_bt_controller_enable(ESP_BT_MODE_BLE));
	ERROR_CHECK(esp_bluedroid_init());
	ERROR_CHECK(esp_bluedroid_enable());
	ERROR_CHECK(esp_ble_gap_register_callback(ble_gap_event_handler));
	ERROR_CHECK(esp_ble_gattc_register_callback(esp_gattc_cb));
	ERROR_CHECK(esp_ble_gattc_app_register(0));
	BLUFI_INFO("BLE ADDR: "ESP_BD_ADDR_STR"\n", ESP_BD_ADDR_HEX(esp_bt_dev_get_address()));
	BLUFI_INFO("BLUFI VERSION %04x\n", esp_blufi_get_version());
}

the callbacks ble_gap_event_handler and esp_gattc_cb is basically copied from gatt_client example.

The output is as followed, endless and finally cause watchdog 

␛[0;33mW (14074) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14084) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14084) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14094) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14094) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14104) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14114) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14114) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14124) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14124) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14134) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (14134) BT_APPL: BTA got unregistered event id 32
__________repeated endless________

What is event 32 and why I got  this?

syu-source, 2019-09-16:
After lots of trail-and-error, I found that the bugy code is this line:
`BLUFI_INFO("BLUFI VERSION %04x\n", esp_blufi_get_version());`

If I remove this line, everything works. 
If I keep this line and call `esp_blufi_profile_init()` after it, everything works.

esp_blufi_get_version() is supposed to get a const version number, but apperently it does a bit more than that.

I could get rid of this issue and move on. But since it tortured my long, I want to know more:
1. what does `esp_blufi_get_version `do other than return version numbers? why does it cause the the problem?
2. what is "BT_APPL:BTA event id 32", where can I find BTA event definitions?

syu-source, 2019-09-16:
More about the "innocent looking" `esp_blufi_get_version ()`:
If it is called without `esp_blufi_profile_init()`, then if wifi status changed, I got 
```
 BT_GATT: gatt_if out of range [ = 0]
 BT_APPL: Unknown connection ID: 0 fail sending notification_
```

syu-source, 2019-09-16:
More logs  

```
␛[0;32mI (843) TAG: (esp_wifi_init(&cfg))OK␛[0m
␛[0;32mI (843) TAG: (esp_wifi_set_storage(WIFI_STORAGE_FLASH))OK␛[0m
␛[0;32mI (853) TAG: (esp_wifi_set_mode(WIFI_MODE_STA))OK␛[0m
␛[0;32mI (943) phy: phy_version: 4101, 100880e, Jul 12 2019, 19:32:06, 0, 0␛[0m
I (943) wifi: mode : sta (a4:cf:12:43:59:34)
␛[0;32mI (943) TAG: (esp_wifi_start())OK␛[0m
␛[0;32mI (953) BTDM_INIT: BT controller compile version [4d3bde7]␛[0m
␛[0;32mI (953) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (963) TAG: (esp_bt_controller_init(&bt_cfg))OK␛[0m
I (1223) wifi: new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (2073) wifi: state: init -> auth (b0)
I (2073) wifi: state: auth -> assoc (0)
␛[0;32mI (2083) TAG: (esp_bt_controller_enable(ESP_BT_MODE_BLE))OK␛[0m
I (2083) wifi: state: assoc -> run (10)
␛[0;32mI (2093) TAG: (esp_bluedroid_init())OK␛[0m
I (2103) wifi: connected with YU, channel 1, 40U, bssid = f8:c3:9e:38:10:34
I (2113) wifi: pm start, type: 1

␛[0;32mI (2183) TAG: (esp_bluedroid_enable())OK␛[0m
␛[0;32mI (2183) TAG: (esp_ble_gap_register_callback(ble_gap_event_handler))OK␛[0m
␛[0;32mI (2183) TAG: (esp_ble_gattc_register_callback(esp_gattc_cb))OK␛[0m
␛[0;32mI (2193) TAG: esp_gattc_cb EVENT ESP_GATTC_REG_EVT gattc_if=3␛[0m
␛[0;32mI (2193) TAG: (esp_ble_gattc_app_register(1))OK␛[0m
␛[0;32mI (2203) TAG: BLE ADDR: a4:cf:12:43:59:36
␛[0m
␛[0;32mI (2203) TAG: (esp_ble_gap_set_scan_params(&ble_scan_params))OK␛[0m
␛[0;32mI (2213) TAG: ble_gap_event_handler 2-ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT␛[0m
␛[0;32mI (2223) TAG: (esp_ble_gap_start_scanning(2147483647))OK␛[0m
␛[0;32mI (2233) TAG: ble_gap_event_handler 7-ESP_GAP_BLE_SCAN_START_COMPLETE_EVT␛[0m
␛[0;32mI (2773) tcpip_adapter: sta ip: 192.168.3.98, mask: 255.255.255.0, gw: 192.168.3.1␛[0m
␛[0;33mW (2783) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (2783) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (2793) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (2803) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (2803) BT_APPL: BTA got unregistered event id 32
␛[0m
␛[0;33mW (2813) BT_APPL: BTA got unregistered event id 32

```

Alvin1Zhang, 2019-09-17:
@syu-source Thanks for reporting, we will look into. Thanks.

gengyuchao, 2019-09-23:
Hi, sorry for late reply. Could you tell me more about you version of the idf ? I can't reproduce your question, could you please provide an example that can reproduce this issue? Thank you.

Alvin1Zhang, 2019-12-03:
@syu-source Thanks for reporting. Would you please help share more details as suggested by @gengyuchao ? Thanks.

Alvin1Zhang, 2019-12-30:
@syu-source Thanks for reporting, sorry that we can't reproduce the issue locally, will close the issue due to not enough feedback and updates. Feel free to reopen with more details. Thanks.

closed 2019-12-30 by Alvin1Zhang

zcs007, 2020-03-27:
I have this problem

zcs007, 2020-03-27:
[image]

zcs007, 2020-03-27:
When you scan the device with ble Bluetooth, you can turn on WiFi and connect to AP