espressif/esp-idf #4071: Reset NVS on CHIP_PU/EN active

jkockx, 2019-09-16:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v4.0-dev-1191-g138c941fa
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

For the device I am working on, I would like one button to reset the esp32 and erase non-volatile storage at the same time. Is it possible somehow to make this happen in software? I mean, having the chip to erase its nvs at reset.
The NVS only needs to be kept on power off and power on, but I want to be able to reset it with my reset button.

KonstantinKondrashov, 2019-09-16:
Hi @jkockx!
Yes, could you try `make menuconfig` and go to bootloader part.
See these options https://github.com/espressif/esp-idf/blob/master/components/bootloader/Kconfig.projbuild#L65-L130.  BOOTLOADER_FACTORY_RESET BOOTLOADER_DATA_FACTORY_RESET

jkockx, 2019-09-16:
Thank you for the quick response. 
In menuconfig I change:
->bootloader config -> checked GPIO triggers factory reset
When I build and try to flash my project with this configuration, I run into another error:
```
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: /home/programmer/esp/gateway_mqtt/build/partitions.bin
/home/programmer/esp/esp-idf/components/esptool_py/Makefile.projbuild:83: recipe for target 'flash' failed
make: *** [flash] Error 2

```
I didn't find a solution right away although this issue has already been asked before. 
My file partitions.csv for custom partition table looks like this at the moment:
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild,,,,
nvs,      data, nvs,     ,        0x6000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        2M,
```

KonstantinKondrashov, 2019-09-16:
I think you need to move your partition table. See an option - https://github.com/espressif/esp-idf/blob/master/components/partition_table/Kconfig.projbuild#L38-L50 . It is defined as 0x8000 try 0x10000.

jkockx, 2019-09-16:
OK, thank you very much! Just now, I did find another issue in which you resolved this as well
https://github.com/espressif/esp-idf/issues/3013
It worked!

jkockx, 2019-09-16:
Working on the devkit, I thought that I only had to change the GPIO pin in the menuconifig (->bootloader config -> [checked] GPIO triggers factory reset -> ) from pin 4 to pin 0. But for some reason the reset of the nvs partition is still not working.
```
 (0)   Number of the GPIO input for factory reset                                │ │  
  │ │          [ ]   Clear OTA data on factory reset (select factory partition)                │ │  
  │ │          (nvs) Comma-separated names of partitions to clear on factory reset             │ │  
  │ │          [ ] GPIO triggers boot from test app partition                                  │ │  
  │ │          (5)   Hold time of GPIO for reset/test mode (seconds)                           │ │  
```

KonstantinKondrashov, 2019-09-16:
Please see https://dl.espressif.com/dl/schematics/ESP32-Core-Board-V2_sch.pdf. GPIO0 it is IO0 on the scheme. It is used in the reset part of the scheme. I suggest to change it to any unused GPIO or to try so - press `EN` then release it and press `BOOT` as close as possible to the release moment (and hold it 5secs).
My recommendation change the number of pin.
```
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (5086) boot: Detect a condition of the factory reset
I (5086) boot: Data partitions to erase: nvs
I (5086) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (5184) boot_comm:  0 nvs              data  00009000 00006000 [yes]
I (5184) boot_comm:  1 phy_init         data  0000f000 00001000 [no]
```

negativekelvin, 2019-09-17:
You can also add a delay to bootloader start so you have more time to press gpio0 button

jkockx, 2019-09-17:
When I try to push BOOT just after EN, I get following output (device makes ready to download flash as if you would push them both at the same time): 
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```
Then I tried to push EN at the moment where @KonstantinKondrashov  pushed it (half a second or a full second after releasing BOOT) and it worked:
```
I (90) boot: Partition Table:
I (94) boot: ## Label            Usage          Type ST Offset   Length
D (101) boot: load partition table entry 0x3f400000
D (106) boot: type=1 subtype=2
I (109) boot:  0 nvs              WiFi data        01 02 00011000 00006000
D (117) boot: load partition table entry 0x3f400020
D (122) boot: type=1 subtype=1
I (125) boot:  1 phy_init         RF data          01 01 00017000 00001000
D (132) boot: load partition table entry 0x3f400040
D (137) boot: type=0 subtype=0
I (141) boot:  2 factory          factory app      00 00 00020000 00200000
I (148) boot: End of partition table
I (5152) boot: Detect a condition of the factory reset
I (5152) boot: Data partitions to erase: nvs
D (5152) bootloader_flash: mmu set paddr=00010000 count=1 size=c00 src_addr=10000 src_addr_aligned=10000
D (5161) boot_comm: mapped partition table 0x10000 at 0x3f400000
D (5167) flash_parts: partition table verified, 4 entries
I (5172) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (5319) boot_comm:  0 nvs              data  00011000 00006000 [yes]
I (5319) boot_comm:  1 phy_init         data  00017000 00001000 [no]
```
Is it because GPIO0 has special functions after boot, that it won't work with GPIO0? So, If I would configure it with another GPIO, I will be able to erase my nvs at any time after boot as well?

KonstantinKondrashov, 2019-09-17:
Yes, you can configure it with another GPIO and you will be able to erase nvs at any time while booting. One more doc here: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/bootloader.html#factory-reset

jkockx, 2019-09-17:
"this GPIO must be pulled low on reset to trigger this"  doesn't the "on reset" part mean it only works after pulling down the reset signal?

jkockx, 2019-09-17:
I preferably want to make a system with only one reset button but as I  both have to bring reset low and up and have to bring the configured GPIO low, this will not be possible probably?

KonstantinKondrashov, 2019-09-17:
GPIO0 is used for booting purpose you can not use it, it was defined in the ROM bootloader.

Use one button it makes sense.  Could you try this - GPIO4 for FACTORY_RESET and handle this button in your application to reset the device (call `esp_restart()`). See example https://github.com/espressif/esp-idf/tree/master/examples/peripherals/gpio. Is it ok for you?

jkockx, 2019-09-17:
OK, I didn't think about a software solution. I'll use software interupts to call `nvs_flash_erase()` and `esp_restart()`. Thank you very much for your help!

KonstantinKondrashov, 2019-09-24:
Hi @jkockx! 
Could you close this issue if it is solved. Thanks.

Alvin1Zhang, 2019-10-14:
@jkockx Thanks for reporting, feel free to reopen if you still have the issue, thanks.

closed 2019-10-14 by Alvin1Zhang