espressif/esp-idf #4073: Multiple Client Data handling with ESP32 BLEServer

kowshik1234, 2019-09-16:
I am operating one ESP32 as a server. I have 2 clients to be connected with this server. One of those clients is an ESP32 and another client is my mobile phone. The strings coming from the mobile phone must be sent to the server and eventually to the client ESP32. How can this be done??

Alvin1Zhang, 2019-09-17:
@kowshik1234 Thanks for reporting, Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. And for "how" questions, we suggest you raise a post on our forum at https://esp32.com. Thanks.

Alvin1Zhang, 2019-09-17:
@kowshik1234 Hi, checked that this ticket is same with the post on https://esp32.com/viewtopic.php?f=2&t=12328&p=49299#p49299. Will close this ticket and track on the post. Thanks.

closed 2019-09-17 by Alvin1Zhang