espressif/esp-idf #4074: [BLE Mesh] Does losing power leads to losing the provisioning of a node in the a bluetooth mesh network?

John-nx, 2019-09-16:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
It seems that when a node loses power (USB is disconnected for example), the mentioned node loses also the provisioning information so that when connected to a power supply again it must be provisioned again, IS that right? if so when this problem will be fixed? 

//Detailed problem description goes here.

### Expected Behavior
After losing power and re-connectiong to a power supply the node must join the mesh network again! 
### Actual Behavior
Node loses provisioning information.
### Steps to reproduce

-Device1 (ESP32-DevKitC) that runs the ble_mesh_client_model project.
-Device2 (ESP32-DevKitC) runs the ble_mesh_node project.
-nRF Mesh app is used for provisioning these two nodes

### Code to reproduce this issue

## Debug Logs

## Other items if possible

Alvin1Zhang, 2019-09-17:
@John-nx Thanks for reporting, we will look into it. Thanks.

WCCWCC, 2019-09-17:
Hi John-nx，
If you want the node to communicate with other nodes after the node is powered back on, you can enable the node's power-down save function.

Componet config -->  Ble mesh Support --> Store BLE Mesh Node configuration persistently

John-nx, 2019-09-17:
@WCCWCC Nope didn't work!

John-nx, 2019-09-17:
This is just so principal about BLE mesh!

John-nx, 2019-09-17:
In fact, activating the "Store BLE Mesh Node configuration persistently", the iussue has become more severe as I neither see the device provisioned nor unprovisioned, now I need to flash it !

Campou, 2019-09-18:
Hi @John-nx 

Please enable the node NVS storage functionality through `make menuconfig`.
`Component config -> BLE Mesh Support -> Store BLE Mesh Node configuration persistently`

* **Note: about this functionality, we have pushed a bugfix to the branch `ble_mesh_release/esp-ble-mesh-v0.6.1` last week, please pull and have a try.**

After the device is provisioned successfully and restarted, the App will not show it in the **Scanner** of the nRF Mesh App, instead we need to click the **CONNECT** on the top right and the App will start to scan the proxy advertising of the device.

I have tested using the branch `ble_mesh_release/esp-ble-mesh-v0.6.1` with NVS storage on, each time after the device is restarted, the App can connect with it successfully and control the RGB light.

The attachment is the `ble_mesh_node` example test log.

[ble_mesh_node_nvs_test.log](https://github.com/espressif/esp-idf/files/3624280/ble_mesh_node_nvs_test.log)

Thanks.

John-nx, 2019-09-18:
Hi @Campou 

Thanks for your response.

Actually, I am working on the latest version of the ble_mesh_release/esp-ble-mesh-v0.6.1 branch. However, I conducted another test from scratch to make sure, the result is as I previously mentioned. 

-I enabled the Store BLE Mesh Node configuration persistently through menuconfig
-I flashed the ble_mesh_node example
-The device is provisioned successfully, but after resetting or disconnecting and reconnecting the power, the nRF Mesh App will see the device, neither as an unprovisioned device (using**Scanner**) nor as a provisioned device (clicking on the **CONNECT**)

Here is a screenshot of the monitored port after resetting it:
[image]

Thanks

Campou, 2019-09-19:
Hi @John-nx 

The difference between the two logs is : `No subnets to advertise on`, and this will cause the problem you have met.

But unfortunately, I couldn't reproduce this issue on my side. So could you please share me the followings which will be a great help.
* The generated binary files (i.e. **ble_mesh_node.bin**, **partitions_singleapp.bin**, **bootloader.bin**)
* The generated **sdkconfig** file

BTW, after pulling the latest code of the branch `ble_mesh_release/esp-ble-mesh-v0.6.1` and running `git describe --tags`, the log on my side is as following. Please make sure it's the same on your side.
[image]

Thanks.

John-nx, 2019-09-19:
Hi @Campou 

I will share with you the files, but now I cannot do that, because as I told you I had to flash it again the option disables in order to provision it again. Anyways I will reproduce the problem and send you the files. 

However, I noticed that running _git describe --tags_ I have slightly a different result :
Note that I do not have any uncommitted changes!

[image]

Running _git log --oneline_
[image]

Thanks

Campou, 2019-09-25:
Hi @John-nx 

Thanks for you support. When the files are provided, we can look into this issue immediately.

Thanks.

John-nx, 2019-10-02:
Hi @Campou,

Sorry for the late response.

-I pulled first 
[image]
-I enabled the "Store BLE Mesh Node configuration persistently" in the menu config and saved the sdkconfig file
(to be sure that the right sdkconfig  is used, I removed others (sdkconfig.defaults and sdkconfig.old))
-I flashed the ble_mesh_node example (using ESP32-DevKitC and idf.py in windows cmd)
-After a reset, the device not only loses the provisioning data but also cannot be provisioned anymore
-Here are the files:
try to unzip the file, I had to add .log nither binary nor zip files are not supported
[debug.7z.log](https://github.com/espressif/esp-idf/files/3682071/debug.7z.log)

Thanks

Campou, 2019-10-28:
Hi @John-nx 

We have fixed this issue on the branch `ble_mesh_release/esp-ble-mesh-v0.6.1`, please pull and have a try.

Thanks.

John-nx, 2019-11-13:
Hi, 
Im sorry to say that, but the new release does not seem to be working!
[image]

I cannot even provision a node le alone the rest!
[image]

Campou, 2019-11-14:
Hi @John-nx,

Seen from the log, and after enabling the BLE Mesh NVS functionality, when you reboot the device, it will restore the corresponding information, there is no need to provision it again.

If the device needs to be provisioned another time, please erase the flash firstly.

Thanks

John-nx, 2019-11-14:
Hi @Campou 

Please read my comments more carefully, every time I try my best to mention the problem as clear as possible but you interpret another thing!

The device cannot be provisioned IN THE FIRST PLACE! using nRF Mesh and your app.

Best

Campou, 2019-11-15:
Hi @John-nx 

If my above comments lead to some misunerstanding, then sorry here.

* But firstly, I have tried with the branch `ble_mesh_release/esp-ble-mesh-v0.6.1` and the nRF Mesh App, it works fine. Following is the screenshot of my test.
[image]

* Secondly in your attachment above, from the log, it indicates that after the reboot, the device restores its previous information from the flash. Please check the log in your attachment:
    ```
    (751) ble_mesh_node: net_idx: 0x0000, addr: 0x0002
    (751) ble_mesh_node: flags: 0x00, iv_index: 0x00000000
    ```
    This is the provisioning information of the node which is restored from the flash, so based on your attachment, I recommend to erase the flash firstly.

* Thirdly, if you are trying to provision the device in the first place, please make sure the log looks like the following, then you can try to provision it.
[image]

Thanks.

John-nx, 2019-11-15:
Hi @Campou 

I still cannot provision the device in the first place. Let me explain more: when I use the nRF app to provision the device, the nRF app hangs/stops during the provisioning process, as shown in photo below :

[image]

and then I have to terminate this step by force, by going to the app menu.
Now if I reset the device I get the error and log that I had previously sent you! 

Note:
1. I am on the ble_mesh_release/esp-ble-mesh-v0.6.1 branch and pulled the latest version
2. Each time I erase the micro and build and flash a clean version
3. I have used both ESP32-DevKitC and WROVER
4. I have flashed ble_mesh_node example (both with the Store "BLE Mesh Node configuration persistently" option enabled and disabled )
5. I use ESP-IDF Tools Installer **(version 2.1 and with python 3 )** and Windows 10
6. when I flash the example (and before provisioning) the log is correct as you shown above

Actually, I am not working on this project anymore I just wanted to give a try and see whether the problem of losing provisioning data after power reset is solved or not, however, now I cannot even provision the node!

Thanks
Here is the log:

I (0) cpu_start: App cpu up.
I (471) heap_init: Initializing. RAM available for dynamic allocation:
I (478) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (484) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (490) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (496) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (502) heap_init: At 3FFCA3A8 len 00015C58 (87 KiB): DRAM
I (508) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (515) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (521) heap_init: At 40091A28 len 0000E5D8 (57 KiB): IRAM
I (527) cpu_start: Pro cpu start user code
I (210) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (211) ble_mesh_node: Initializing...
I (211) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (241) BTDM_INIT: BT controller compile version [3861d9f]
I (241) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
BLE MESH LIB 0619
I (331) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
BLE MESH LIB 0619
BLE MESH LIB 0619
I (701) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_PROV_REGISTER_COMP_EVT
I (701) ble_mesh_node: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (711) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT
I (711) ble_mesh_node: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (721) ble_mesh_node: BLE Mesh Node initialized
I (620751) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT
I (620751) ble_mesh_node: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (627471) BLE_MESH: bt_mesh_attention, No Health Server context provided
W (631371) BLE_MESH: bt_mesh_attention, No Health Server context provided
I (631391) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT
I (631391) ble_mesh_node: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
I (631401) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (631411) ble_mesh_node: net_idx: 0x0000, addr: 0x0006
I (631421) ble_mesh_node: flags: 0x00, iv_index: 0x00000000
W (637561) BLE_MESH: Composition page 255 not available
W (639811) BLE_MESH: Ran out of retransmit attempts
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6332
load:0x40078000,len:11308
ho 0 tail 12 room 4
load:0x40080400,len:6700
[... 41 lines cut ...]
I (502) heap_init: At 3FFCA3A8 len 00015C58 (87 KiB): DRAM
I (508) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (515) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (521) heap_init: At 40091A28 len 0000E5D8 (57 KiB): IRAM
I (527) cpu_start: Pro cpu start user code
I (210) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (211) ble_mesh_node: Initializing...
I (211) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (241) BTDM_INIT: BT controller compile version [3861d9f]
I (241) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
BLE MESH LIB 0619
I (341) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
BLE MESH LIB 0619
BLE MESH LIB 0619
E (701) BT_GATT: GATTS_StopService service_handle: 46 is not in use
I (711) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (711) ble_mesh_node: net_idx: 0x0000, addr: 0x0006
I (711) ble_mesh_node: flags: 0x00, iv_index: 0x00000000
W (721) BOARD: led green is already off
I (721) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_PROV_REGISTER_COMP_EVT
I (731) ble_mesh_node: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (741) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT
I (751) ble_mesh_node: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code -120
I (751) ble_mesh_node: BLE Mesh Node initialized

Campou, 2019-11-15:
Hi @John-nx 

When I tried with the latest Android App (version 2.1.0), I met the same problem.  And I switch back to the old version which is version 1.1.0. Test about 3 times, it works fine, the following are the screenshots of the test.
<p align="center">
	[image]
	[image]
	[image]
	<p align="center">
	</p>
</p>

And in the Github website of Nordic, some developer has submitted an issue for this https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/268.

Also I have a test with the lastes iOS version, and everything goes well.

Thanks.

John-nx, 2019-11-15:
Hi @Campou ,

Alright, thanks. I would downgrade the app and test again.

John-nx, 2019-11-15:
Hi @Campou 

First things first, I downgraded the nRF mesh one version to the v2.0.5 (in my opinion, it does not make any sense to downgrade to 1.1.0), and I could provision the node successfully, and more importantly, after a reset, the node was working just fine and I was able to connect to it!  However, there was one minor problem: when I wanted to reconnect to the node after a rest, instead of showing the name of the node in the searching list, nRF was showing "UNKNOWN DEVICE".

Secondly, I noticed that the version 2.1.0 is released 15 days ago! So it seemed odd to me that such a bug issue still exists ! I thought maybe it works with the Nordic micros and I was right!
I tried to provision a nordic micro with the latest nRF mesh Android app (v2.1.0), and it worked successfully! At this point, I think it's your job to find the problem.
BTW you work as a developer and support for esspressif , right?

Best

Alvin1Zhang, 2020-05-28:
@John-nx Thanks for reporting and keeping sharing updates, and thanks @Campou for the support. Feel free to reopen the issue if it still happens, or file a new ticket for new issue. Thanks.

closed 2020-05-28 by Alvin1Zhang

ashish-iottive, 2021-12-01:
@Alvin1Zhang @Campou Could you provide proper screenshots of the whole BLE Node Provisioning Storage, I am running example of `onoff_server` from `esp_ble_mesh` example.
I am facing same problem how to store the ble mesh provision via nRF mesh Android Mobile Application cause the device lost the provisioning settings after its gets restart. 
Environment
Development Kit: [ESP32-DevKitC]
Kit version (DevKitC): [v1]
Module or chip used: [ESP32-WROOM-32]
IDF version :
// v4.0-dev-301-gc7c419da4
Build System: ESP-IDF Tools // [Make|CMake]
Operating System: [Windows]
Power Supply: [USB]

> Hi @John-nx
>
> Please enable the node NVS storage functionality through `make menuconfig`. `Component config -> BLE Mesh Support -> Store BLE Mesh Node configuration persistently`
>
> * **Note: about [...]

 
Here is the screenshot of the monitor after flash:
[image]

.