espressif/esp-idf #4075: Internet access captive portal

AlphaArslan, 2019-09-16:
Hi there
Thanks for the great contribution we really appreciate that.

I'm trying to build on your code by adding captive portal to it.
Can you help me with any tips or directions ?

Captive Portal that allows the user to access the internet. something like what you would see in hotels and so

Thanks

Alvin1Zhang, 2019-09-17:
@AlphaArslan Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. And for "how" questions, we suggest you raise a post on our forum at https://esp32.com. Thanks.

negativekelvin, 2019-09-17:
https://github.com/search?q=esp32+captive

septatrix, 2021-06-30:
> Would you please help provide more details as suggested... And for "how" questions, we suggest you raise a post on our forum...

I guess details are not necessary as this is mostly a how question. However I would really like to have code for a softAP featuring a captive portal to the examples.

closed 2021-09-04 by espressif-bot