espressif/esp-idf #4076: Boot loader log level

Zeni241, 2019-09-17:
I recently download esp-idf v3.3 and tried hello_world example.
I set CONFIG_BOOTLOADER_LOG_LEVEL to No output  but I am still getting very verbose out put at boot time.
[BootloaderLog.txt](https://github.com/espressif/esp-idf/files/3619245/BootloaderLog.txt)

Alvin1Zhang, 2019-09-17:
@Zeni241 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

KonstantinKondrashov, 2019-09-17:
HI @Zeni241!
Please note we have at least two options for logs: 
- for bootloader (CONFIG_BOOTLOADER_LOG_LEVEL)
- for application (CONFIG_LOG_DEFAULT_LEVEL) `Component config->Log output->Default log verbosity (Info).`

I looked at BootloaderLog.txt, you have these options:
CONFIG_BOOTLOADER_LOG_LEVEL=No output  (Info by default)
 and 
CONFIG_LOG_DEFAULT_LEVEL=Verbose  (Info by default)
Please correct them as you want.

Zeni241, 2019-09-17:
Thanks @Konstantin Kondrashov , I corrected as you guided and set both log levels to none. Now my log output is as given in file below. This is much better. Just out of curiosity can we reduce it further?
Thanks again.

[BootloaderLog2.txt](https://github.com/espressif/esp-idf/files/3620984/BootloaderLog2.txt)

KonstantinKondrashov, 2019-09-17:
@Zeni241. Your log consist of two parts: 
1. ROM bootloader log can not be disabled.
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:884
load:0x40078000,len:9716
ho 0 tail 12 room 4
load:0x40080400,len:5840
entry 0x4008069c
```
2. hello_world example uses ptintf directly. We do not have a mechanism to disable it but If you rewrite the example with ESP_LOGI("TAG", "..."); you could manage these logs.
```
Hello world!
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB extern
al flash
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
```

What is the purpose of disabling all logs? 
Look at `Component config->Common ESP-related->UART for console output-> None`, maybe it will be useful for you. But I do not recommend disabling all logs, because it will become very difficult to debug an application.

Zeni241, 2019-09-17:
Thanks, my question is answered by you, that is more than enough for a day:)

closed 2019-09-17 by Zeni241

igrr, 2019-09-18:
Some more information, ROM bootloader log can be disabled by pulling GPIO15 low at reset time.

Zeni241, 2019-09-18:
Thank you igrr.