espressif/esp-idf #4077: EXCEPTION Handling Guide

akshar001, 2019-09-17:
Version: - ESP-IDF v3.2.2-dirty 2nd stage bootloader

`#define APP_TAG "Fox"

#include <stdexcept>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_err.h"
#include "esp_log.h"

void inner_exception_test_task(  void *ignore  ) {
  ESP_LOGI(APP_TAG, "exception_test_task enter" );

  try {
    throw std::runtime_error( "Task exception message" );
  } catch ( const std::exception& e ) {
    ESP_LOGE(APP_TAG, "Task exception: %s", e.what() );
  }

  for(;;) {
        vTaskDelay( 1000 / portTICK_PERIOD_MS );
    }

    vTaskDelete( NULL );
}

void exception_test_task(  void *ignore  ) {
  inner_exception_test_task(ignore);
}

extern "C" void app_main(void)
{
  ESP_LOGI(APP_TAG, "Main thread enter" );

  try {
    // Serila.println()
    int i = 580;
    int k = i/0;
    printf("%d\n",k);
  } catch (const std::exception& e) {
    // ESP_LOGI(APP_TAG, "Error Occured" );
    ESP_LOGE(APP_TAG, "Exception: %s", e.what() );
  }
   ESP_LOGI(APP_TAG, "Hello here" );
  //   xTaskCreate(&exception_test_task, "exception_test_task", 4096, NULL, 6, NULL);

  // for(;;) {
  //   vTaskDelay(1000 / portTICK_PERIOD_MS);
  //   ESP_LOGI(APP_TAG, "+" );
  // }
}`
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Debug Log
rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6292
load:0x40078000,len:10180
ho 0 tail 12 room 4
load:0x40080400,len:6712
entry 0x40080768
I (31) boot: ESP-IDF v3.2.2-dirty 2nd stage bootloader
I (31) boot: compile time 12:07:54
I (33) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (66) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (74) boot:  2 app0             OTA app          00 10 00010000 00140000
I (81) boot:  3 app1             OTA app          00 11 00150000 00140000
I (89) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (96) boot: End of partition table
I (101) boot: No factory image, trying OTA 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x145b8 ( 83384) map
I (144) esp_image: segment 1: paddr=0x000245e0 vaddr=0x3ffb0000 size=0x01ec0 (  7872) load
I (147) esp_image: segment 2: paddr=0x000264a8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/akshar/Desktop/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (151) esp_image: segment 3: paddr=0x000268b0 vaddr=0x40080400 size=0x0933c ( 37692) load
I (175) esp_image: segment 4: paddr=0x0002fbf4 vaddr=0x00000000 size=0x0041c (  1052) 
I (176) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x14b40 ( 84800) map
0x400d0018: _stext at ??:?

I (217) boot: Loaded app from partition at offset 0x10000
I (217) boot: Disabling RNG early entropy source...
I (217) cpu_start: Pro cpu up.
I (221) cpu_start: Starting app cpu, entry point is 0x40080ea8
0x40080ea8: call_start_cpu1 at /Users/akshar/Desktop/esp/esp-idf/components/esp32/cpu_start.c:246

I (212) cpu_start: App cpu up.
I (232) heap_init: Initializing. RAM available for dynamic allocation:
I (238) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (244) heap_init: At 3FFB30B0 len 0002CF50 (179 KiB): DRAM
I (251) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (257) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (263) heap_init: At 4008973C len 000168C4 (90 KiB): IRAM
I (270) cpu_start: Pro cpu start user code
I (64) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (66) Fox: Main thread enter
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d2272  PS      : 0x00060e30  A0      : 0x800d0a9d  A1      : 0x3ffb9950  
0x400d2272: app_main at /Users/akshar/Desktop/esp/esp-idf/examples/get-started/blink/main/blink.cpp:38

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb9900  
A10     : 0x400dbd28  A11     : 0x00000244  A12     : 0x3f4039a0  A13     : 0x00000008  
0x400dbd28: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c:35

A14     : 0x00000002  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x400d2272:0x3ffb9950 0x400d0a9a:0x3ffb9970 0x40084409:0x3ffb9990
0x400d2272: app_main at /Users/akshar/Desktop/esp/esp-idf/examples/get-started/blink/main/blink.cpp:38

0x400d0a9a: main_task at /Users/akshar/Desktop/esp/esp-idf/components/esp32/cpu_start.c:497

0x40084409: vPortTaskWrapper at /Users/akshar/Desktop/esp/esp-idf/components/freertos/port.c:403

Rebooting...
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Problem or a question:- I want to enable a excetion handling in my esp-idf code. By this is a simple code i have written for any exception to handle. Right now i have tried to catch IntegerDeviceByZero. I also have tried putting catch(...) into that breackets. The problem is they are not being caught. I have insufficient knowledge about this topic. But i have tried most of the exceptions for IntegerDevideByZero for C++ . But it's still not being caught. A little guide or reference about how to handle default exceptions like this is in need.

negativekelvin, 2019-09-17:
https://github.com/espressif/esp-idf/tree/release/v3.2/examples/system/cpp_exceptions

akshar001, 2019-09-17:
@negativekelvin yes i have uploaded that program. But it just show the exception that we throws . What about the exceptions that inner library that throws. Like dividing integer by 0 or accessing sum null pointers.

negativekelvin, 2019-09-17:
Those errors are too low level to have exception handling.

akshar001, 2019-09-18:
Yes but there should be a way to handle them.

negativekelvin, 2019-09-18:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html

https://stackoverflow.com/questions/1823721/how-to-catch-the-null-pointer-exception

akshar001, 2019-09-18:
This is a very nice insights @negativekelvin.

closed 2019-09-18 by akshar001