espressif/esp-idf #4078: Secure Bootloader and Flash Encryption - flash read err, 1000

ManishMehra001, 2019-09-17:
#Environment

- Development Kit:      [ESP32-WROOM-32D| 
- Kit version 
- Module or chip used: |ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
  
- Build System:         [Make|
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB|external 5V]

## Problem Description

//Detailed problem description goes here.

After enabling Secure boot + Flash encryption and following the steps mentioned in steps to reproduce, getting error "flash read err, 1000" .

### Expected Behavior
Module should boot up successfully with Secure Boot and Flash Encryption feature enabled.
### Actual Behavior
flash read err, 1000
### Steps to repropduce

Secure Bootloader and Flash Encryption both are [enabled:](url)
1. Run command "make menuconfig"												

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3620677/sdkconfig.txt)

													
	Select "Scurity features"																									
	Enable for "Enable secure boot in bootloader"																									
	Select "Secure bootloader mode", select "Refalshable".																									
	Enable "Sign binaries during build"																									
	On "Secure boot provite signing key", input your file name, here is "my_secure_boot_signing_key.pem"																									
	Enable "Allow potentially insecure options"																									
	Select "Potentially insecure option", 
	Select "Leave ROM BASIC Interpreter available on reset"
	Select "Allow JTAG Debugging"
	Select "Allow app partition length not 64KB aligned"
	Select "Leave UART bootloader encryption enabled"
	Not Select "Leave UART bootloader decryption enabled"
	Not Select "Leave UART bootloader flash cache enabled"
	
	Select "Bootloader config"																									
	Select "Bootloader log verbosity"																									
	Select "No output"																									
	
	Select "Enable flash encryption on boot (READ DOCS FIRST) -  Enable usage mode (Development(NOT SECURE))"
	Save and Exit menu config

2. Create Flash Encryption key: hdbs_flash_encryption_key.bin
python $IDF_PATH/components/esptool_py/esptool/espsecure.py digest_private_key --keyfile my_secure_boot_signing_key.pem hdbs_flash_encryption_key.bin

3. make bootloader 
4. Enable Secure boot on board and Burn secure bootloader key 
	- Enable Secure boot on board by burn ABS_DONE_0 to 1,
	- Burn secure bootloader key (secure-bootloader-key-256.bin),
5. Enable Encription Flash on board by burn FLASH_CRYPT_CNT and Burn Encreption key (hdbs_flash_encryption_key.bin).

6. Encrypt bootloader : bootloader-reflash-digest.bin
python $IDF_PATH/components/esptool_py/esptool/espsecure.py encrypt_flash_data --keyfile hdbs_flash_encryption_key.bin --address 0x0 -o BinaryEncryptOut/bootloader-reflash-digest_En BinaryOut/bootloader-reflash-digest

Generated file : bootloader-reflash-digest_En,
Converted to bin by adding (.bin) : bootloader-reflash-digest_En_0x0.bin

7. Build and Encrypt Partition table
python $IDF_PATH/components/esptool_py/esptool/espsecure.py encrypt_flash_data --keyfile hdbs_flash_encryption_key.bin --address 0x8000 -o BinaryEncryptOut/partitions_spiffs_En BinaryOut/partitions_spiffs

Generated file : partitions_spiffs_En,
Converted to bin by adding (.bin) : partitions_spiffs_En_0x8000.bin

8. Build and Encrypt factory app
python $IDF_PATH/components/esptool_py/esptool/espsecure.py encrypt_flash_data --keyfile hdbs_flash_encryption_key.bin --address 0x10000 -o BinaryEncryptOut/panl24_fac_En BinaryOut/panl24_fac

Generated file : panl24_fac_En,
Converted to bin by adding (.bin) : panl24_fac_En_0x10000.bin

9. Build and Encrypt main app
python $IDF_PATH/components/esptool_py/esptool/espsecure.py encrypt_flash_data --keyfile hdbs_flash_encryption_key.bin --address 0x100000 -o BinaryEncryptOut/panl24_app_En BinaryOut/panl24_app

Generated file : panl24_app_En,
Converted to bin by adding (.bin) : panl24_app_En_0x100000.bin

Flash Binaries:
10. bootloader-reflash-digest_En
python $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyS29 --baud 230400 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 BinaryEncryptOut/bootloader-reflash-digest_En

11. partitions_spiffs_En
python $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyS29 --baud 230400 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x8000 BinaryEncryptOut/partitions_spiffs_En

12. panl24_fac_En
python $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyS29 --baud 230400 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 BinaryEncryptOut/panl24_fac_En

13. panl24_app_En
python $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyS29 --baud 230400 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x100000 BinaryEncryptOut/panl24_app_En

14. ota_data_initial
python $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyS29 --baud 230400 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xd000 BinaryOut/ota_data_initial

After reset, getting below error:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

Please let me know what is wrong or missing in my above steps.
// It helps if you attach a picture of your setup/wiring here.

// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2019-09-17:
Your binaries are preencrypted but did you set encryption key and crypt cnt efuses?

ManishMehra001, 2019-09-17:
[image]

ManishMehra001, 2019-09-17:
I have attached the summary log.

negativekelvin, 2019-09-17:
Ah, key tweak also needs to be set
https://github.com/espressif/esp-idf/blob/aa087667dffb8f024a5bf162cef4d5c0ffb152ab/examples/security/flash_encryption/example_test.py#L18

ManishMehra001, 2019-09-17:
Following command use to burn key :

python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key secure_boot ./build/bootloader/secure-bootloader-key-256.bin
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key flash_encryption hdbs_flash_encryption_key.bin
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse FLASH_CRYPT_CNT

ManishMehra001, 2019-09-17:
You mean, I need to execute a below command :

python $IDF_PATH/components/esptool_py/esptool/espefuse.py --do-not-confirm --port /dev/ttyS29 burn_efuse FLASH_CRYPT_CONFIG 0xf

ManishMehra001, 2019-09-17:
Or need to execute command something like,
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse FLASH_CRYPT_CNT --do-not-confirm

ManishMehra001, 2019-09-17:
I tried below command and getting error - "A fatal error occurred: New value required for efuse FLASH_CRYPT_CONFIG", 
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --do-not-confirm --port /dev/ttyS29 burn_efuse FLASH_CRYPT_CONFIG

ManishMehra001, 2019-09-17:
ohh Great bro :) thanks a lot :)
It works now with below command :)

python $IDF_PATH/components/esptool_py/esptool/espefuse.py --do-not-confirm --port /dev/ttyS29 burn_efuse FLASH_CRYPT_CONFIG 0xf

Alvin1Zhang, 2019-09-18:
@negativekelvin Thanks for the notes. @ManishMehra001 Thanks for reporting, has your issue been resolved? Thanks.

ManishMehra001, 2019-09-18:
Yes @negativekelvin, my issue is resolved. Thanks for your instant & right support.

closed 2019-09-18 by Alvin1Zhang

ManishMehra001, 2019-09-19:
Hi @negativekelvin, Please help me to know, 
with my above setup, where my board has 
- Secure boot key & Flash Encryption key, efuse burned as below, 
   1. python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key 
   secure_boot ./build/bootloader/secure-bootloader-key-256.bin

   2. python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key flash_encryption hdbs_flash_encryption_key.bin

 3.  python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0

4. python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse FLASH_CRYPT_CNT
 
5. python $IDF_PATH/components/esptool_py/esptool/espefuse.py --do-not-confirm --port /dev/ttyS29 burn_efuse FLASH_CRYPT_CONFIG 0xf

- Later Flashing pre-encrypted binaries
And able to boot successfully,

Now In order to do ota flash, do i need to generate ota binary with encrypted binary files or with Plaintext file, ?
Note able to understand this from the esp website.
Please help me in that.

Please let me know if I need to create a separate ticket for this query.

Thanks & Best Regards,
Manish

negativekelvin, 2019-09-19:
Plaintext but app has to be signed

ManishMehra001, 2019-09-19:
Thanks for your reply @negativekelvin,

So I will sign my binaries like below and then generate ota bin file,

1. Encrypt bootloader : bootloader-reflash-digest.bin
python $IDF_PATH/components/esptool_py/esptool/espsecure.py encrypt_flash_data --keyfile hdbs_flash_encryption_key.bin --address 0x0 -o BinaryEncryptOut/bootloader-reflash-digest_En BinaryOut/bootloader-reflash-digest

Generated file : bootloader-reflash-digest_En,
Converted to bin by adding (.bin) : bootloader-reflash-digest_En_0x0.bin

2. Build and Encrypt Partition table
python $IDF_PATH/components/esptool_py/esptool/espsecure.py encrypt_flash_data --keyfile hdbs_flash_encryption_key.bin --address 0x8000 -o BinaryEncryptOut/partitions_spiffs_En BinaryOut/partitions_spiffs
Generated file : partitions_spiffs_En,
Converted to bin by adding (.bin) : partitions_spiffs_En_0x8000.bin

3. Build and Encrypt factory app
python $IDF_PATH/components/esptool_py/esptool/espsecure.py encrypt_flash_data --keyfile hdbs_flash_encryption_key.bin --address 0x10000 -o BinaryEncryptOut/panl24_fac_En BinaryOut/panl24_fac
Generated file : panl24_fac_En,
Converted to bin by adding (.bin) : panl24_fac_En_0x10000.bin

4. Build and Encrypt main app
python $IDF_PATH/components/esptool_py/esptool/espsecure.py encrypt_flash_data --keyfile hdbs_flash_encryption_key.bin --address 0x100000 -o BinaryEncryptOut/panl24_app_En BinaryOut/panl24_app
Generated file : panl24_app_En,
Converted to bin by adding (.bin) : panl24_app_En_0x100000.bin

And then execute below command to get the ota bin,
- ./mkspiffs -c assets -b 4096 -p 256 -s 0x200000 BinaryOut/otadata

- ./script/catfile.py **BinaryEncryptOut/panl24_app_En** 0x180000 BinaryOut/otadata 0x200000 BinaryOut/otaall

ManishMehra001, 2019-09-23:
Hi @negativekelvin, Just to update this ticker here, my ota binary works well if I use plain text binary instead of encrypted one, like, 
./script/catfile.py **BinaryOut/panl24_app** 0x180000 BinaryOut/otadata 0x200000 BinaryOut/otaall

May be i misunderstood your earlier comment. 
Thanks again.

ManishMehra001, 2019-09-23:
Hi @negativekelvin,

Now i am trying another approach for "Secure Boot + Flash encryption" where I dont pre-encrypt my binary files and let secondary bootloader encrypt my normal plain text files.

I try to understand steps mentioned in esp web and executed below steps but getting same error again 

After reset, getting below error:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun 8 2016 00:22:57

Below are details of my steps:

1. Enable Secure Boot And Flash Encryption
Select "Scurity features"																									
Enable for "Enable secure boot in bootloader"																									
Select "Secure bootloader mode", select "One - time flash".																									
Enable "Sign binaries during build"																									
On "Secure boot provite signing key", input your file name, here is "my_secure_boot_signing_key.pem"																									
Enable "Allow potentially insecure options"																									
Select "Potentially insecure option", 
Select "Leave ROM BASIC Interpreter available on reset"
Select "Allow JTAG Debugging"
Select "Allow app partition length not 64KB aligned"
Select "Leave UART bootloader encryption enabled"
Not Select "Leave UART bootloader decryption enabled"
Not Select "Leave UART bootloader flash cache enabled"

Select "Bootloader config"																									
Select "Bootloader log verbosity"																									
Select "No output"																									

Select "Enable flash encryption on boot (READ DOCS FIRST) -  Enable usage mode (Development(NOT SECURE))"
Save and Exit menu config

2. make bootloader

3. Enable Secure boot on board and Burn secure bootloader key

	- Enable Secure boot on board by burn ABS_DONE_0 to 1,
	  python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0

	- Burn secure bootloader key (secure-bootloader-key-256.bin),
	  python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key secure_boot ./build/bootloader/secure-bootloader-key-256.bin
		
4. Flashed following binaries at below mentioned addresses:

	bootloader-reflash-digest_En_0x0.bin at address 0x0
	Build and Encrypt Partition table at address 0x8000
	Build and Encrypt factory app at address 0x10000
	Build and Encrypt main app at address 0x100000
	ota_data_initial at address 0xd000
	otadata_086b0e29f7c5570fb7ee208a8c7f8c0f at address 0x400000

5. Attaching here the espefuse summary.

6. Please help me to know what is missing in my above steps?

Thanks & Best Regards,
[image]

Manish

negativekelvin, 2019-09-24:
Don't set crypt fuses and don't encrypt binaries

ManishMehra001, 2019-09-24:
Hi negativekelvin, correct this is what i also understood so I only burn below espefuse of bootloader only,

- Enable Secure boot on board by burn ABS_DONE_0 to 1,
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0

- Burn secure bootloader key (secure-bootloader-key-256.bin),
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key secure_boot ./build/bootloader/secure-bootloader-key-256.bin

Because it was not working so later I burn FLASH_CRYPT_CONFIG also,
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --do-not-confirm --port /dev/ttyS29 burn_efuse FLASH_CRYPT_CONFIG 0xf

And I am not sure why FLASH_CRYPT_CNT is set to 1 .. even though i didnt burn it..

Please have a look on above attached summary log and let me know what can i do to make it work.

Thanks & Best Regards,
Manish

negativekelvin, 2019-09-24:
Set FLASH_CRYPT_CNT to 3 and try again

ManishMehra001, 2019-09-24:
Hi negativekelvin,  I set FLASH_CRYPT_CNT to 3 and getting below error after reboot:

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1088
ho 0 tail 12 room 4
load:0x40078000,len:16500
load:0x40080400,len:2996
entry 0x400805c4

Attaching below summary log:
Thanks & Best Regards,
Manish
[image]

negativekelvin, 2019-09-24:
Well you turned bootloader logs off so we don't know what is happening.  Eventually it should boot the app after a minute of encrypting.

ManishMehra001, 2019-09-26:
Hi negativekelvin,

This time before I execute my steps on new board, 
I want to verify below steps if these steps are ok to execute or i need to add any missing steps here:

1. Enable Secure Boot And Flash Encryption

Select "Security features"
Enable for "Enable secure boot in bootloader"
Select "Secure bootloader mode", select "Reflashable".
Enable "Sign binaries during build"
On "Secure boot privite signing key", input your file name, here is "my_secure_boot_signing_key.pem"
Enable "Allow potentially insecure options"
Select "Potentially insecure option",
Select "Leave ROM BASIC Interpreter available on reset"
Select "Allow JTAG Debugging"
Select "Allow app partition length not 64KB aligned"
Select "Leave UART bootloader encryption enabled"
Select "Leave UART bootloader decryption enabled"
Select "Leave UART bootloader flash cache enabled"
Select "Bootloader config"
Select "Bootloader log verbosity"
Select "No output"

Select "Enable flash encryption on boot (READ DOCS FIRST) - Enable usage mode (Development(NOT SECURE))"
Save and Exit menu config

2. make bootloader

3. Enable Secure boot on board and Burn secure bootloader key

3.1 Enable Secure boot on board by burn ABS_DONE_0 to 1,
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0

3.2 Burn secure bootloader key (secure-bootloader-key-256.bin),
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key secure_boot ./build/bootloader/secure-bootloader-key-256.bin

4. Build and Flash following bin files at their address using esp32 flash download tool:
	bootloader-reflash-digest_En_0x0.bin at address 0x0
	Build Partition table and flash at address 0x8000
	Build factory app and flash at address 0x10000
	Build main app and flash at address 0x100000
	Build ota_data_initial and flash at address 0xd000
	Build otadata_086b0e29f7c5570fb7ee208a8c7f8c0f and flash at address 0x400000

5. Reboot the H/W.

Please have a look on my above steps and if anything is missing here please suggest me.
It would be great help of yours if finally we can work with this approach where we dont need to create flash encryption key.

Thanks & Best Regards,
Manish

ManishMehra001, 2019-09-26:
I believe my steps are as per description mentioned int the link "https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html"

"On first power-on reset, all data in flash is un-encrypted (plaintext). First stage loader (ROM) will load the second stage loader in IRAM.
Second stage bootloader will read the flash_crypt_cnt (=00000000b) eFuse value and since the value is 0 (even number of bits set) it will configure and enable the flash encryption block. It will also program FLASH_CRYPT_CFG eFuse to value 0xF.
The flash encryption block will generate AES-256 bit key and store into BLOCK1 eFuse. This operation is performed in hardware and the key can not be accessed by software.
Next the flash encryption block will encrypt the flash contents (based on partition table flag value). Encrypting in-place can take some time (up to a minute for large partitions).
Second stage bootloader then sets the first available bit in flash_crypt_cnt (=00000001b) to mark the flash contents as encrypted (odd number of bits set)."

ManishMehra001, 2019-09-26:
Or Is there any order, I suspect means,
First enable only Flash encryption,
And once it can bootup successfully then 
Enable Secure Bootloader .....

ManishMehra001, 2019-09-26:
Hi negativekelvin,

**In my above 4th step of Secure boot + Flash encryption in place,
Do i need to flash 
i. bootloader.bin (at Address 0x1000)
OR
ii. bootloader-reflash-digest_0x0.bin at address 0x0**

Below copying Step - 4 again:
4. Build and Flash following bin files at their address using esp32 flash download tool:
**bootloader-reflash-digest_0x0.bin at address 0x0  (Question : Is it right here ?)**
Build Partition table and flash at address 0x8000
Build factory app and flash at address 0x10000
Build main app and flash at address 0x100000
Build ota_data_initial and flash at address 0xd000
Build otadata_086b0e29f7c5570fb7ee208a8c7f8c0f and flash at address 0x400000

negativekelvin, 2019-09-26:
Yes, with digest.  You may want to consider leaving some bootloader logs enabled until you know the process.  "Leave UART bootloader encryption enabled" means you can change it later.

ManishMehra001, 2019-09-27:
Hi negativekelvin, to avoid confusion, for this issue where configuration done for "Secure Boot (Reflashable) + Flash Encryption in place" , I created separate ticket and where i used fresh steps on fresh board and attached my error log,
**Please have a look on ticket:
                            https://github.com/espressif/esp-idf/issues/4124

Overall, for this issue, where configuration done for "Secure Boot (Reflashable) + Flash Encryption in place" , I tried with flashing

i. bootloader-reflash-digest_0x0.bin at address 0x0
And other plain text files at their respective address

ii. bootloader.bin (at Address 0x1000)
And other plain text files at their respective address

And in both log summary i can see, FLASH_CRYPT_CNT and FLASH_CRYPT_CONFIG are not set by the Hardware.**