espressif/esp-idf #4079: "idf.py fullclean" command removes some header files belong to the esp-idf repository!

John-nx, 2019-09-17:
## Environment

-Development Kit: [ESP32-DevKitC]
-Kit version (DevKitC): [v4]
-Module or chip used: [ESP32-WROOM-32D]
-IDF version (ran git describe --tags to find it):
-// v4.0-dev-301-gc7c419da4
-Build System: ESP-IDF Tools // [Make|CMake]
-Compiler version (ran gcc --version to find it):
-// gcc (MinGW.org GCC-6.3.0-1) 6.3.0
-Operating System: [Windows]
-Power Supply: [USB]

## Problem Description

Surprisingly using the "idf.py fullclean" command not only all the files under the build folder are removed, but also some header files under the following path are removed! Which result in the compilation error!
...\esp-idf\components\mbedtls\mbedtls\include\mbedtls

### Expected Behavior

### Actual Behavior

### Steps to reproduce
Run idf.py fullclean for the ble_mesh_node example.

## Debug Logs
## Other items if possible

dobairoland, 2019-09-17:
Please update to a newer v4.0. This has been fixed in 6ccdca511422feb921a98624ba176090b55f6708

John-nx, 2019-09-17:
Im working on the ble_mesh_release/esp-ble-mesh-v0.6.1 branch. There is no newer version unless I checkout to the master branch or fix it locally.
Anyways my intention was to report the bug, if it's already fixed then we're good.

closed 2019-09-17 by John-nx