espressif/esp-idf #4081: SSL: CERTIFICATE_VERIFY_FAILED

LCSimba, 2019-09-17:
Hello I have tried to use the guideline for installing the esp-idf. I receive the following error when trying to run the install.sh command. The following error is given:

Lions-MacBook:esp-idf LionMacBook$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp32-2019r1-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-macos.tar.gz to /Users/LionMacBook/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-macos.tar.gz.tmp
Traceback (most recent call last):
  File "/Users/LionMacBook/esp/esp-idf/tools/idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "/Users/LionMacBook/esp/esp-idf/tools/idf_tools.py", line 1313, in main
    action_func(args)
  File "/Users/LionMacBook/esp/esp-idf/tools/idf_tools.py", line 1107, in action_install
    tool_obj.download(tool_version)
  File "/Users/LionMacBook/esp/esp-idf/tools/idf_tools.py", line 541, in download
    urlretrieve(url, local_temp_path, report_progress if not global_non_interactive else None)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1038, in endheaders
    self._send_output(message_body)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 882, in _send_output
    self.send(msg)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 844, in send
    self.connect()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1263, in connect
    server_hostname=server_hostname)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 369, in wrap_socket
    _context=self)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 599, in __init__
    self.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 828, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)

I have tried multiple versions(latest stable and Beta). I have tried to reinstall Python2.7 and made sure it is the only version in the PATH variable. Removed it and tried again a few times. I also double checked that all dependencies where installed correctly. Any ideas why this would be?

Alvin1Zhang, 2019-09-23:
@LCSimba Thanks for reporting, we will look into. Thanks.

jonra1993, 2019-09-24:
I have the same problem described by @LCSimba . When I run install.sh in Mac OS

punee995, 2019-09-24:
Same for the Windows. looks like Certificate expired.

juliendland, 2019-09-24:
https://thomas-cokelaer.info/blog/2016/01/python-certificate-verified-failed/

punee995, 2019-09-24:
@juliendland that worked.

danb-swiftlabs, 2019-09-24:
Could someone help me understand how to use the above posted fix?

SaurabhBajaj, 2019-09-24:
@projectgus @igrr @renzbagaporo  looks like the SSL cert on the `https://dl.espressif.com/` website has expired - could you please update the certs otherwise setup would be broken for everyone. The above is just a temporary workaround. I am surprised it's been expired for over 7 days already.

SaurabhBajaj, 2019-09-24:
As @juliendland  mentioned, if you want to disable the SSL check, just update the `tools/idf_tools.py` file from esp-idf folder and add these lines in the import section
```
import ssl
ssl._create_default_https_context = ssl._create_unverified_context
```

Alvin1Zhang, 2019-09-25:
@SaurabhBajaj We have fixed the certificate, would you please help try again? Thanks.

LCSimba, 2019-09-25:
Hello I have downloaded the idf again and following the steps. It is still giving the same error:

Lions-MacBook:esp-idf LionMacBook$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp32-2019r1-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-macos.tar.gz to /Users/LionMacBook/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-macos.tar.gz.tmp
Traceback (most recent call last):
  File "/Users/LionMacBook/esp/esp-idf/tools/idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "/Users/LionMacBook/esp/esp-idf/tools/idf_tools.py", line 1313, in main
    action_func(args)
  File "/Users/LionMacBook/esp/esp-idf/tools/idf_tools.py", line 1107, in action_install
    tool_obj.download(tool_version)
  File "/Users/LionMacBook/esp/esp-idf/tools/idf_tools.py", line 541, in download
    urlretrieve(url, local_temp_path, report_progress if not global_non_interactive else None)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1038, in endheaders
    self._send_output(message_body)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 882, in _send_output
    self.send(msg)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 844, in send
    self.connect()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1263, in connect
    server_hostname=server_hostname)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 369, in wrap_socket
    _context=self)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 599, in __init__
    self.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 828, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)

tomcolen, 2019-10-09:
The issue is still there
@SaurabhBajaj fixed it for me, thanks!

Francois-Belanger, 2019-11-12:
Still not working...

projectgus, 2019-11-13:
The expired cert should be fixed now, and should auto-renew correctly from now on. Thanks everyone for your patience while we resolved this issue.

closed 2019-11-13 by projectgus

Jayrajghoniya, 2019-12-09:
This issue is still not solved. I am facing this issue from last 2 days.
Please fix it.
thank you.

kolipakakondal, 2020-01-21:
https://stackoverflow.com/a/41692664/1812235 this is what it helped me! We need to run "Install Certificates.command"  from /Applications/Python 3.6/. This is applicable even for Python 2.6 as well.

> /Applications/Python 3.6/Install Certificates.command

noyelseth, 2020-08-25:
Today(26-Aug-2020), I encountered the same problem while installing **ESP-IDF**.

```
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache
Installing xtensa-esp32-elf@esp-2019r2-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to C:\Users\hp-pavilion\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to C:\Users\hp-pavilion\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip.tmp
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to C:\Users\hp-pavilion\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to C:\Users\hp-pavilion\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip.tmp
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to C:\Users\hp-pavilion\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to C:\Users\hp-pavilion\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "D:\workspace\emmate-release\emmate-v2.1-esp32\sdk\esp-idf\tools\idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "D:\workspace\emmate-release\emmate-v2.1-esp32\sdk\esp-idf\tools\idf_tools.py", line 1488, in main
    action_func(args)
  File "D:\workspace\emmate-release\emmate-v2.1-esp32\sdk\esp-idf\tools\idf_tools.py", line 1181, in action_install
    tool_obj.download(tool_version)
  File "D:\workspace\emmate-release\emmate-v2.1-esp32\sdk\esp-idf\tools\idf_tools.py", line 598, in download
    raise DownloadError()
__main__.DownloadError
```

projectgus, 2020-08-26:
Hi @noyelseth,

The certificate on dl.espressif.com is up to date (expiry 19 November 2020) and showing as valid.

It looks like you're using Windows, is that right? Is it a new-ish Windows install?

There is a known issue with Python and Windows where the certificate store doesn't add certificates initially, it adds them on the fly. But Python doesn't do this.

You can try the following things to resolve:
* Navigate to https://dl.espressif.com/ in a web browser (ideally IE or Edge), and then retry.
* Follow the steps shown here: https://bugs.python.org/issue36011#msg335850

noyelseth, 2020-08-28:
Hey @projectgus,

Thanks. It worked.

UpstairTechnologiesLLP, 2020-11-23:
Hi.. guys 
I'm facing the Same issue with the **Linux 18.04 LTS**
here are the logs 
`Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz to /home/dnk127/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz to /home/dnk127/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz to /home/dnk127/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz to /home/dnk127/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz to /home/dnk127/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz to /home/dnk127/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/home/dnk127/esp/esp-idf/tools/idf_tools.py", line 1591, in <module>
    main(sys.argv[1:])
  File "/home/dnk127/esp/esp-idf/tools/idf_tools.py", line 1587, in main
    action_func(args)
  File "/home/dnk127/esp/esp-idf/tools/idf_tools.py", line 1280, in action_install
    tool_obj.download(tool_version)
  File "/home/dnk127/esp/esp-idf/tools/idf_tools.py", line 697, in download
    raise DownloadError()
__main__.DownloadError
`

Can any one help me here

aknabi, 2021-01-16:
Looks like it's back (with macOS) as well....

Adding:
ssl._create_default_https_context = ssl._create_unverified_context

let it get further... but still get the following after a while:

WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/abdul/.espressif/dist/xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/abdul/.espressif/dist/xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/abdul/.espressif/dist/xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/abdul/.espressif/dist/xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/abdul/.espressif/dist/xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
ERROR: Failed to download, and retry count has expired

hackaprende, 2021-03-17:
Still happening, I solved it with @SaurabhBajaj instructions about editing the tools/idf_tools.py

Seraffimo, 2021-04-30:
Still happening for me. Did not see a resolution in this thread. Running from MacOS terminal ...

Installing xtensa-esp32-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/.../.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/billybond/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/billybond/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/billybond/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/billybond/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/billybond/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
ERROR: Failed to download, and retry count has expired

neythex, 2021-05-20:
What helps me out is

1. going to my Python3 folder `cd /Applications/Python\ 3.7`
2. start `./Install\ Certificates.command`
3. going back to esp-idf folder
4. retry `./install.sh`

MGonzalez506, 2021-06-12:
I had the same Issue, tried with @nythex but in my case it did not work. 

But tryied with @SaurabhBajaj option and it did work. I am on MAC OS X 11.4.

Now I am having the problem on Ubuntu, I tried the same steps here and the error stills the same I cannot go through even when disabling SSL check. @projectgus any advance on this? Thanks

MGonzalez506, 2021-06-13:
> I had the same Issue, tried with @nythex but in my case it did not work.
>
> But tryied with @SaurabhBajaj option and it did work. I am on MAC OS X 11.4.
>
> Now I am having the problem on Ubuntu, I tried [...]

--
Solved not the right way.

I fresh installed ubuntu again and following this video [https://www.youtube.com/watch?v=Jt6ZDct4bZk](url) I was able to make it work.

What I do not know is if there is any problem about having virtualenv and virtualenvwrapper already installed on my ubuntu machines and that could lead to the problem described in this thread, that is the only thing I can think of that was installed in the previous try I did.

georgik, 2021-07-20:
Hi guys. I'd like to add summary for Windows, macOS and Linux including how to simulate the problem and possible fixes.

The problem is caused by missing certificate in the system. This is true for all operating systems. Python is using system certificates to establish connection to sites like https://dl.espressif.com.

## Linux

The problem can be simulated on Linux:
- add `!` as the first character of line `mozilla/DST_Root_CA.crt` in file `/etc/ca-certificates.conf`
- run command: `update-ca-certificates`
- following Python code fails with `SSL: CERTIFICATE_VERIFY_FAILED`:

```python
import urllib.request
with urllib.request.urlopen('https://dl.espressif.com') as response:
   html = response.read()
```

### How to fix the problem on Linux

```
mv /etc/ca-certificates.conf /tmp
apt-get update && apt-get install --reinstall ca-certificates
```

## Windows

The problem can be simulated on Windows:
- run mmc
- CTRL+M - select `Certificates`, click `Add`, click `Finish`, click `Ok`
- expand `Certificates - Current User`, expand `Trusted Root Certificates`, right-click `DST Root CA X3`, click `Delete`, confirm `Yes`
- open Python and paste the code above

### How to fix the problem on Windows

- open PowerShell and type the following command:
```
Invoke-WebRequest https://dl.espressif.com -OutFile index.html
```

Alternative solution
- open the following link in web browser https://dl.espressif.com

Windows is updating certificates in the background.

## macOS

The problem can be simulated on macOS:

- open application `Keychain Access`
- click `System Roots`, double-click `DST Root CA X3`, expand `Trust`, change `When using this certificates` to `Never Trust`, close the window, confirm your decision by providing a password

### How to fix the problem on macOS

- make sure that `DST Root CA X3` and `ISGR Root X1` is trusted and up-to-date

Alternative solution:

```
cd "/Applications/Python\ 3.7"
start ./Install\ Certificates.command
```

Alternative solution:
```
python3 -m pip install certifi
python3 -m certifi
```

TobyBorland, 2021-11-25:
Same error under a fresh Ubuntu 20.04 LTS installation,

WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1131)>

Resolved via,

$ sudo update-ca-certificates --fresh
$ export SSL_CERT_DIR=/etc/ssl/certs

JonathanWMorris, 2022-03-12:
> As @juliendland mentioned, if you want to disable the SSL check temporarily, just update the `tools/idf_tools.py` file from esp-idf folder and add these lines in the import section
> 
> ```
> import ssl
> ssl._create_default_https_context = ssl._create_unverified_context
> ```

I was trying to run `./install.sh` on MacOS 12 today and I ran into the issue again. This is the only solution that worked for me.

ceerqingtingml, 2022-07-12:
> $ sudo update-ca-certificates --fresh
> $ export SSL_CERT_DIR=/etc/ssl/certs

@TobyBorland ‘s solution also works for me, I am using Ubuntu 18.04 LTS.