espressif/esp-idf #4083: pthreads - future/promise - unresolved references when linking

rbiernat, 2019-09-17:
Hi,

I pulled latest changes and when I attempt to us standard c++ future/promise stuff it compiles fine but I get the following when trying to link

xtensa-esp32-elf/lib\libstdc++.a(condition_variable.o): in function `__gthread_cond_broadcast':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/xtensa-esp32-elf/bits/gthr-default.h:865: undefined reference to `pthread_cond_broadcast'

I'm builidng on windows not linux using the latest tools/framework installer. Building and flashing work fine when I don't try and use the future/promise stuff i.e regular pthread mutexes are ok so initially I thought not everything in pthrteads was implemented but one of the test cases in the framework explicitly tests the futures/promises stuff so is this one of those ones where on windows the library may not be being built correctly?

Thanks

Rob

igrr, 2019-09-18:
Could you please attach the project (source file and cmakelists) which reproduces the issue? There is a linking issue related to pthreads and libstdc++ which we are fixing now, need to check if yours is the same or not.

Alvin1Zhang, 2019-09-23:
@rbiernat Hi, thanks for reporting. Would you please help share more details? Thanks.

rbiernat, 2019-09-24:
Hi
Have been out of action with my other job. Will upload shortly

Rob

johnnyxwan, 2019-09-29:
I have similar error messages when using EventQueue in the following library:
https://github.com/wqking/eventpp

```
undefined reference to `pthread_cond_init'
undefined reference to `pthread_cond_destroy'
```

igrr, 2019-10-07:
It seems that this should be fixed by a recent commit, https://github.com/espressif/esp-idf/commit/f1b5c326f413beb6adc367861975fd94506a2056. Could you please try again with latest master?

igrr, 2019-10-29:
Assuming that this is fixed by the commit mentioned above. Please reopen if the issue persists.

closed 2019-10-29 by igrr

Abdulrahman-Yasser, 2024-01-06:
I just faced this problem.
adding this to my cmake solved the problem:

idf_component_register(SRCS src/main.cpp
                    INCLUDE_DIRS ""
                    REQUIRES freertos spi_flash pthread)