espressif/esp-idf #4087: can not build thingstream SDK

Pratikm78, 2019-09-18:
I can not build esp-idf code for the thingstream SDK. running into erro when trying to build the code

C:\Users\Pratik\Desktop\rebuild TS>idf.py build
Checking Python dependencies...
Python requirements from C:\dev\esp\esp-idf\requirements.txt are satisfied.
Running ninja in directory 'C:\Users\Pratik\Desktop\rebuild TS\build'
Executing "ninja all"...
[1/5] Linking CXX executable rebuild.elf
FAILED: rebuild.elf
cmd.exe /C "cd . && C:\PROGRA~1\ESPRES~1\ESP-ID~1\tools\bin\XT1FE9~1.EXE    -nostdlib CMakeFiles/rebuild.elf.dir/dummy_main_src.c.obj  -o rebuild.elf  -Wl,--gc-sections -Wl,--cref -Wl,--Map=rebuild.map -Wl,--start-group soc/libsoc.a log/liblog.a heap/libheap.a freertos/libfreertos.a vfs/libvfs.a newlib/libnewlib.a esp_ringbuf/libesp_ringbuf.a driver/libdriver.a esp_event/libesp_event.a ethernet/libethernet.a lwip/liblwip.a tcpip_adapter/libtcpip_adapter.a app_update/libapp_update.a spi_flash/libspi_flash.a mbedtls/libmbedtls.a micro-ecc/libmicro-ecc.a bootloader_support/libbootloader_support.a nvs_flash/libnvs_flash.a pthread/libpthread.a smartconfig_ack/libsmartconfig_ack.a wpa_supplicant/libwpa_supplicant.a xtensa-debug-module/libxtensa-debug-module.a esp32/libesp32.a cxx/libcxx.a app_trace/libapp_trace.a asio/libasio.a jsmn/libjsmn.a coap/libcoap.a console/libconsole.a nghttp/libnghttp.a esp-tls/libesp-tls.a esp_adc_cal/libesp_adc_cal.a tcp_transport/libtcp_transport.a esp_http_client/libesp_http_client.a esp_http_server/libesp_http_server.a esp_https_ota/libesp_https_ota.a expat/libexpat.a wear_levelling/libwear_levelling.a sdmmc/libsdmmc.a fatfs/libfatfs.a freemodbus/libfreemodbus.a json/libjson.a libsodium/liblibsodium.a mdns/libmdns.a mqtt/libmqtt.a openssl/libopenssl.a protobuf-c/libprotobuf-c.a protocomm/libprotocomm.a spiffs/libspiffs.a ulp/libulp.a wifi_provisioning/libwifi_provisioning.a -Wl,--undefined=uxTopUsedPriority -L C:/dev/esp/esp-idf/components/newlib/lib -lc -lm -L C:/dev/esp/esp-idf/components/esp32/lib -L C:/dev/esp/esp-idf/components/esp32/ld/wifi_iram_opt -lcoexist -lcore -lespnow -lmesh -lnet80211 -lphy -lpp -lrtc -lsmartconfig -lwpa2 -lwpa -lwps -L C:/Users/Pratik/Desktop/rebuild TS/build/esp32 -T esp32_out.ld -L C:/dev/esp/esp-idf/components/esp32/ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld C:/dev/esp/esp-idf/components/esp32/libhal.a -lgcc -u call_user_start_cpu0 -u ld_include_panic_highint_hdl -lstdc++ -u __cxa_guard_dummy -u __cxx_fatal_exception -lgcov && cd ."
XT1FE9~1.EXE: error: TS/build/esp32: No such file or directory
[2/5] Performing build step for 'bootloader'
ninja: no work to do.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\Pratik\Desktop\rebuild TS>

Pratikm78, 2019-09-18:
it is always fails when it try tries to build the bootloader
[61/62] Linking C executable bootloader.elf
FAILED: bootloader.elf

Alvin1Zhang, 2019-09-19:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks. @Pratikm78

Pratikm78, 2019-09-19:
Development Kit: ESP32-DevKitC
Module or chip used: [ESP32-WROOM-32|
idf version: ESP-IDF v4.1-dev-281-g96b96ae24
using ninja to build
xtensa version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
Operating System: [Windows]
Usb powered

Pratikm78, 2019-09-20:
I have the thingstream sdk example from thingstream.

C:\Users\Pratik\Desktop\ts_example\example>idf.py reconfigure
Executing action: reconfigure
CMakeLists.txt not found in project directory c:\users\pratik\desktop\ts_example\example
there file does not contain a CMakeLists.txt

I created a CMakelist.txt  with the follow content:
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(TS_example)

then I ran a reconfigure and tried to build the code however ran to errors

C:\Users\Pratik\Desktop\ts_example\example>idf.py reconfigure
Executing action: reconfigure
Running cmake in directory c:\users\pratik\desktop\ts_example\example\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\pratik\desktop\ts_example\example"...
Warn about uninitialized values.
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
CMake Warning at C:/dev/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Xtensa toolchain C:/Program Files/Espressif/ESP-IDF
  Tools/tools/bin/xtensa-esp32-elf-gcc.exe version 5.2.0 is not the supported
  version 8.2.0.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  C:/dev/esp/esp-idf/components/esp32/project_include.cmake:20 (gcc_version_check)
  C:/dev/esp/esp-idf/tools/cmake/build.cmake:305 (include)
  C:/dev/esp/esp-idf/tools/cmake/build.cmake:458 (__build_process_project_includes)
  C:/dev/esp/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Warning at C:/dev/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain C:/Program Files/Espressif/ESP-IDF
  Tools/tools/bin/xtensa-esp32-elf-gcc.exe crosstool-ng version
  crosstool-ng-1.22.0-80-g6c4433a5 doesn't match supported version
  esp32-2019r1.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  C:/dev/esp/esp-idf/components/esp32/project_include.cmake:21 (crosstool_version_check)
  C:/dev/esp/esp-idf/tools/cmake/build.cmake:305 (include)
  C:/dev/esp/esp-idf/tools/cmake/build.cmake:458 (__build_process_project_includes)
  C:/dev/esp/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/Users/Pratik/Desktop/ts_example/example/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/dev/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/dev/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/dev/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/dev/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/dev/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/dev/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/dev/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/dev/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-locale.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/dev/esp/esp-idf/components/app_trace C:/dev/esp/esp-idf/components/app_update C:/dev/esp/esp-idf/components/asio C:/dev/esp/esp-idf/components/bootloader C:/dev/esp/esp-idf/components/bootloader_support C:/dev/esp/esp-idf/components/bt C:/dev/esp/esp-idf/components/cbor C:/dev/esp/esp-idf/components/coap C:/dev/esp/esp-idf/components/console C:/dev/esp/esp-idf/components/cxx C:/dev/esp/esp-idf/components/driver C:/dev/esp/esp-idf/components/efuse C:/dev/esp/esp-idf/components/esp-tls C:/dev/esp/esp-idf/components/esp32 C:/dev/esp/esp-idf/components/esp_adc_cal C:/dev/esp/esp-idf/components/esp_common C:/dev/esp/esp-idf/components/esp_eth C:/dev/esp/esp-idf/components/esp_event C:/dev/esp/esp-idf/components/esp_gdbstub C:/dev/esp/esp-idf/components/esp_http_client C:/dev/esp/esp-idf/components/esp_http_server C:/dev/esp/esp-idf/components/esp_https_ota C:/dev/esp/esp-idf/components/esp_https_server C:/dev/esp/esp-idf/components/esp_local_ctrl C:/dev/esp/esp-idf/components/esp_ringbuf C:/dev/esp/esp-idf/components/esp_rom C:/dev/esp/esp-idf/components/esp_websocket_client C:/dev/esp/esp-idf/components/esp_wifi C:/dev/esp/esp-idf/components/espcoredump C:/dev/esp/esp-idf/components/esptool_py C:/dev/esp/esp-idf/components/expat C:/dev/esp/esp-idf/components/fatfs C:/dev/esp/esp-idf/components/freemodbus C:/dev/esp/esp- [... cut ...] dev/esp/esp-idf/components/bootloader/subproject/components/micro-ecc C:/dev/esp/esp-idf/components/partition_table C:/dev/esp/esp-idf/components/soc C:/dev/esp/esp-idf/components/spi_flash C:/dev/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Pratik/Desktop/ts_example/example/build/bootloader
ninja: no work to do.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\Pratik\Desktop\ts_example\example>

renzbagaporo, 2019-09-23:
For the first case, is it the case that the project path has a space? We don't support spaces in build paths.

For the second case, have you defined a `main` component with a source file defining `app_main`? You can use the template project https://github.com/espressif/esp-idf-template as reference.

renzbagaporo, 2019-10-14:
Hi @Pratikm78, any update on the issue?

renzbagaporo, 2019-10-18:
Closing due to lack of updates.

closed 2019-10-18 by renzbagaporo