espressif/esp-idf #4088: ULP instruction precise cycles

eminarcissus, 2019-09-18:
In current ULP document we have two versions of running cycles.

All instructions are 32-bit. Jump instructions, ALU instructions, peripheral register and memory access instructions are executed in 1 cycle. Instructions which work with peripherals (TSENS, ADC, I2C) take variable number of cycles, depending on peripheral operation.

On the top.

And at each instruction under we get a totally different description,which is correct here?
 For example.

ADD - Add to register

Syntax
ADD Rdst, Rsrc1, Rsrc2

ADD Rdst, Rsrc1, imm

Operands
Rdst - Register R[0..3]
Rsrc1 - Register R[0..3]
Rsrc2 - Register R[0..3]
Imm - 16-bit signed value
Cycles
### 2 cycles to execute, 4 cycles to fetch next instruction
Description
The instruction adds source register to another source register or to a 16-bit signed value and stores result to the destination register.

ST – Store data to the memory

Syntax
ST Rsrc, Rdst, offset
Operands
Rsrc – Register R[0..3], holds the 16-bit value to store
Rdst – Register R[0..3], address of the destination, in 32-bit words
Offset – 10-bit signed value, offset in bytes
Cycles
### 4 cycles to execute, 4 cycles to fetch next instruction
Description
The instruction stores the 16-bit value of Rsrc to the lower half-word of memory with address Rdst+offset. The upper half-word is written with the current program counter (PC), expressed in words, shifted left by 5 bits:

Alvin1Zhang, 2019-09-19:
@eminarcissus Thanks for reporting, we will look into. Thanks.

igrr, 2019-09-20:
@eminarcissus The sentence about 1 cycle execution time is outdated, thanks for pointing this out. Will update the document.

eminarcissus, 2019-09-20:
wow, so indeed run a NOP will cost 2 cycles running time and 4 cycles of fetch time? in total of 750ns for a NOP?

boarchuz, 2019-11-13:
I had to debug some timing issues today so I may as well share these until the docs are updated in case it's of use to anyone:

12 cycles:
REG_WR

8 cycles:
REG_RD, LD, ST

6 cycles:
WAIT

4 cycles:
JUMP (all), STAGE (all), MOVE, and all of the basic arithmetic and binary ops

All others untested.