espressif/esp-idf #4091: BLE_MESH, BT_GATT error

tatsutaigu, 2019-09-19:
Hi,

I found error log on BLE_MESH_NODE as below.
> E (45330) BT_GATT: GATT_SendServiceChangeIndication can't send service change indication manually, please configure the option through menuconfig

How do I resolve it?
I cannot find `GATT_SendServiceChangeIndication` on menuconfig.

my ESP-IDF version is here.
> $ git log -2
> commit a7e8d87d3e5ccc9e5ffcd701a1bac587ba4f43ea (HEAD -> master, origin/master, origin/HEAD)
> Merge: d1f2ea7cd ba2ec73d5
> Author: Angus Gratton <angus@espressif.com>
> Date:   Fri Aug 30 16:27:28 2019 +0800
> 
>     Merge branch 'bugfix/error_out_on_uninit_git' into 'master'
> 
>     cmake: fix build failure when git repo is initialized but no first commit
> 
>     Closes IDFGH-1676
> 
>     See merge request espressif/esp-idf!5914
> 
> commit d1f2ea7cdf7bf377c8705edbd74c5c1c095b172f
> Merge: 879778a67 c50a67567
> Author: Angus Gratton <angus@espressif.com>
> Date:   Fri Aug 30 16:27:21 2019 +0800
> 
>     Merge branch 'bugfix/make_build_issue_if_build_dir_symlink' into 'master'
> 
>     ldgen: fix build issue on make when build dir is a symlink
> 
>     Closes IDFGH-1340
> 
>     See merge request espressif/esp-idf!5915

Campou, 2019-09-19:
Hi @tatsutaigu 

Please enable sending **GATT Service Indication manually** in the `menuconfig`, the path is:
`Component config -> Bluetooth -> Bluedroid Options -> GATTS Service Change Mode -> GATTS manually send service change indication`

Also this configuration has been added in the **sdkconfig.defaults** file of the example, before compiling the example, you can run `make defconfig` to load the default configurations.

Thanks.

tatsutaigu, 2019-09-19:
Hi, @Campou 

Thank you.
I have solved the problem

BTW, `idf.py` does not have `defconfig` option.

[https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#idf-py](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#idf-py)

> $ idf.py defconfig
> Usage: /tatsutaigu/muto/esp/esp-idf/tools/idf.py [OPTIONS] COMMAND1 [ARGS]... [COMMAND2 [ARGS]...]...
> Try "/home/tatsutaigu/esp/esp-idf/tools/idf.py --help" for help.
> 
> Error: No such command "defconfig".
>

closed 2019-09-19 by tatsutaigu