espressif/esp-idf #4092: idf.py issue

tatsutaigu, 2019-09-19:
Hi,

I found below 2 issue.

- idf.py ignores `menuconfig` --> `Serial flasher config` --> `before flashing (No reset)` option.
- idf.py does not support `defconfig` option.

Alvin1Zhang, 2019-09-19:
@tatsutaigu Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

tatsutaigu, 2019-09-19:
Hi, @Alvin1Zhang 

**Environment**
- Development Kit: ESP32-DevKitC
- Kit version (for DevKitC): v4
- Module or chip used: ESP32-WROOM-32D|ESP32-SOLO-1
- IDF version (run git describe --tags to find it): v4.1-dev-281-g96b96ae24
- Build System: idf.py
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: Linux
- Power Supply:USB
 
**Problem Description**

- ignored before flashing (No reset)
sample code:[blink](https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink)
I set `idf.py menuconfig` --> `Serial flasher config` --> `before flashing (No reset)` option.
But, `idf.py` ignore `before flashing (No reset)` option when I do `idf.py -p /dev/ttyS20 flash`.
And, I cannot find `--before no_reset` option on below message.[esptool.py reset before operation](https://github.com/espressif/esptool/wiki/Advanced-Options#reset-before-operation)

> $ idf.py build
> [687/687] Generating binary image from built executable
> esptool.py v2.8-dev
> Generated /home/muto/esp/esp-idf/examples/get-started/blink/build/blink.bin
> 
> Project build complete. To flash, run this command:
> ../../../components/esptool_py/esptool/esptool.py -p (PORT) -b 921600 --after hard_reset write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blink.bin
> or run 'idf.py -p (PORT) flash'

- does not support `defconfig`
sample code:[ble mesh node](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node)
This sample code should do `make defconfig` before build.[IDFGH-1888](https://github.com/espressif/esp-idf/issues/4091#issuecomment-532935283)
But, `idf.py` does not prepare `defconfig` option.[idf.py option](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#idf-py)

kumekay, 2019-09-19:
Hi @tatsutaigu,
Thank you for the issue, 

- We will add missing `--before` parameter
-  If you just run `idf.py build` it will create sdkconfig automatically. If you want only create sdkconfig `idf.py reconfigure` can be used. This part of the issue is duplication of https://github.com/espressif/esp-idf/issues/3965

tatsutaigu, 2019-09-19:
Hi, @kumekay ,

Thank you for your reply.

About defconfig,
I'm sorry, I cannot understand it.
when `idf.py build`, value of `sdkconfig.default` set to `sdkconfig`?
How do I let `sdkconfig` file reflect default value of  `sdkconfig.default`?

kumekay, 2019-09-19:
If `sdkconfig` is missing  project build targets  will automatically create ``sdkconfig`` file, populated with the settings from ``sdkconfig.defaults`` file, and the rest of the settings will be set to their default values. 

If sdkconfig is present, settings that are already in sdkconfig will not be overriden by the ones in `sdkconfig.defaults`

tatsutaigu, 2019-09-19:
Thank you for your advice.

I have solved the problem.
When `sdkconfig` is missing and `idf.py menuconfig`, `idf.py menuconfig` does not read default value from `sdkconfig.defaults`.
If `sdkconfig` already exists, I remove it and copy `sdkconfig.defaults` to `sdkconfig`.
after that, `idf.py menuconfig` reads copied `sdkconfig`.

> $ rm sdkconfig
> $ rm sdkconfig.old
> $ cp sdkconfig.defaults sdkconfig
> $ idf.py menuconfig

kumekay, 2019-09-19:
`idf.py menuconfig` should read values of sdkconfig.defaults. Could you, please share your `sdkconfig.defaults` and `Kconfig.projbuild` (if you have it)?

tatsutaigu, 2019-09-19:
I already wrote it on **Problem Description**.
[https://github.com/espressif/esp-idf/issues/4092#issuecomment-532944861](https://github.com/espressif/esp-idf/issues/4092#issuecomment-532944861)

I use below example.
I did not modify code.
can you duplicate this issue?
[https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node)

kumekay, 2019-09-19:
@tatsutaigu 
Yes, there is an issue with some parameters in sdkconfig.defaults that were recently renamed, for example `CONFIG_BTU_TASK_STACK_SIZE ` -> `CONFIG_BT_BTU_TASK_STACK_SIZE`, while both old and new names are updated through menuconfig and stored in `sdkconfig`, old names are ignored in `sdkconfig.defaults` by cmake.
We will fix this.

tatsutaigu, 2019-09-19:
@kumekay 
Thank you for checking.

When I did below command sequence,
`CONFIG_BTU_TASK_STACK_SIZE` and `CONFIG_BT_BTU_TASK_STACK_SIZE` are defined automatically.

> $ rm sdkconfig
> $ rm sdkconfig.old
> $ cp sdkconfig.defaults sdkconfig
> $ idf.py menuconfig

> $ grep CONFIG_BTU_TASK_STACK_SIZE sdkconfig
> CONFIG_BTU_TASK_STACK_SIZE=4512

> $ grep CONFIG_BT_BTU_TASK_STACK_SIZE sdkconfig
> CONFIG_BT_BTU_TASK_STACK_SIZE=4512

kumekay, 2019-09-20:
Yes, the issue is that if you have deprecated name in sdkconfig.defaults i.e. `CONFIG_BTU_TASK_STACK_SIZE ` it's ignored,
 for compatibility both `CONFIG_BTU_TASK_STACK_SIZE` and `CONFIG_BT_BTU_TASK_STACK_SIZE ` are set in sdkconfig to default value of CONFIG_BT_BTU_TASK_STACK_SIZE that is 4096. 

As I mentioned before, we are working on fix.

dobairoland, 2019-09-27:
So there are three issues here mentioned:
- missing `before` for flashing,
- missing `defconfig`, and
- problem with deprecated options in `sdkconfig.defaults`.

Since the missing defconfig is a duplicate of https://github.com/espressif/esp-idf/issues/3965, I propose to ignore it here. @tatsutaigu you can follow that issue and it will be solved when it gets closed. I will reference this current issue in the commit where I fix the `sdkconfig.defaults`. And https://github.com/espressif/esp-idf/pull/4125 could close this issue.

kumekay, 2020-05-19:
@tatsutaigu  If all problems described in the issue are solved, could you please close it?

closed 2020-05-19 by tatsutaigu