espressif/esp-idf #4094: dose it can not ping to esp32 with mdns example in idf 4.0?

huazhenjiang, 2019-09-19:
### Environment
Development Kit: ESP32-DevKitC
Kit version (for WroverKit/PicoKit/DevKitC): v2
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-1287-gd7e659d
Build System: CMake
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: Windows 10
Power Supply: USB

### Problem Description
I copy the mdns example from sdk for testing. (esp-idf/examples/protocols/mdns/)
But I can not ping to the esp32, just like the following graph.

ping esp32-mdns.local (failed)
ping 192.168.200.9 (ok)

[image]

negativekelvin, 2019-09-19:
Can you ping any .local address from Windows command line?  I don't think it is supported.

huazhenjiang, 2019-09-20:
Hi @negativekelvin , 

I can ping another esp32 mdns example which build with arudino, but it only support ipv4.

closed 2019-09-20 by huazhenjiang

jaguy, 2019-11-28:
I'm seeing the same issue with V4.0 beta 2 (worked in IDF V3.3). See https://esp32.com/viewtopic.php?f=13&t=13255

jaguy, 2019-12-04:
My testing shows the issue I was seeing with mDNS was fixed by @david-cermak  See esp32 forum thread above.