espressif/esp-idf #4095: Missing libusb dependency on ubuntu

mirronelli, 2019-09-19:
- IDF version: v4.1-dev-281-g96b96ae24
- Operating System:     Linux

## Problem Description

The toolchain setup on a ubuntu:latest container doesn't produce a working build environment

### Expected Behavior

- export.sh in esp-idf directory does not display errors and warnings
- OpenOcd tool works

### Actual Behavior
export.sh throws this error
```WARNING: tool openocd-esp32 version v0.10.0-esp32-20190708 is installed, but the tool failed to run```

trying to start openocd directly displays:
```root@d483f5c351af:~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin# ./openocd
./openocd: error while loading shared libraries: libusb-1.0.so.0: cannot open shared object file: No such file or directory
```

### Steps to repropduce

```
docker run -it ubuntu:latest
apt-get update
apt-get -y install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache
mkdir /root/esp
cd /root/esp
git clone --recursive https://github.com/espressif/esp-idf.git
cd /root/esp/esp-idf
./install.sh
. ./export.sh 
```

### Steps to fix
apt-get install libusb-1.0

### Suggested change
add the libusb dependency to the documentation
```
apt-get -y install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache  libusb-1.0
```

PerMalmberg, 2019-09-20:
I had this issue too so it's not local to @mirronelli's system.

KaeLL, 2020-04-24:
I'm having this exact problem too running the `v4.1-beta1` snapshot.
```
WARNING: tool openocd-esp32 version v0.10.0-esp32-20191114 is installed, but the tool failed to run
ERROR: tool openocd-esp32 has no installed versions. Please run 'install.sh' to install it.
```

cjds, 2020-12-30:
I couldn't find the package but I did

```
apt-get install libusb-1.0-0-dev
```

on Ubuntu 20.04

projectgus, 2021-01-12:
It should be enough to do `apt install libusb-1.0-0` (no `-dev` needed) on Debian/Ubuntu.

A fix to the docs is in review now, this issue will be updated when it merges.

closed 2021-01-15 by espressif-bot

tyeth, 2023-05-28:
Thanks for the solve already, don't know why I'm getting this, github codespace building circuitpython, using the idf install.sh, but got a non-zero exit code (one would have expected it to be installed):
```
WARNING: tool openocd-esp32 version v0.11.0-esp32-20221026 is installed, but returned non-zero exit code (127) with error message:
/home/codespace/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin/openocd: error while loading shared libraries: libusb-1.0.so.0: cannot open shared object file: No such file or directory
```

SL2021-Dev, 2023-08-12:
NEW to using CODESPACES...
can't run . export.sh

@SL2021-Dev ➜ /workspaces/SL-v2/esp-idf (ac5d805d0e) $ . export.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.10.8
"python" has been detected
Adding ESP-IDF tools to PATH...
WARNING: tool openocd-esp32 version v0.11.0-esp32-20221026 is installed, but returned non-zero exit code (127) with error message:
/home/codespace/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin/openocd: error while loading shared libraries: libusb-1.0.so.0: cannot open shared object file: No such file or directory

Tryed to install... apt install libusb-1.0-0
got
@SL2021-Dev ➜ /workspaces/SL-v2 (main) $ apt install libusb-1.0-0
E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?