espressif/esp-idf #4096: [BLE Mesh]how many the node can be provisioned by device provisioner?

isDerek, 2019-09-19:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

make menuconfig (ble mesh support)

env: ble wifi coexist
csv file:
phy_init,   data,   phy,       0x9000      ,  0x1000
otadata,  data,    ota,                           ,  0x2000
ota_0,      app,     ota_0,    0x10000     ,  0x1C0000
ota_1,       0,    ota_1,          0x210000   ,  0x1C0000
nvs,          data, nvs,                              ,  0x20000

example 1:
CONFIG_BLE_MESH_WAIT_FOR_PROV_MAX_DEV_NUM=20
CONFIG_BLE_MESH_MAX_PROV_NODES=20
CONFIG_BLE_MESH_MAX_STORED_NODES=20
example 2:
CONFIG_BLE_MESH_WAIT_FOR_PROV_MAX_DEV_NUM=50
CONFIG_BLE_MESH_MAX_PROV_NODES=50
CONFIG_BLE_MESH_MAX_STORED_NODES=50

### Expected Behavior

example 1 & example 2
the node can be provisioned

### Actual Behavior

example 1:
the  node can be provisioned
example 2:
the node provisioned fail

## Debug Logs

loop the log
```
W (46942) BLE_MESH: Receive configuration status message timeout
I (46944) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0005, opcode: 0x8008
```
 i think may be  the error is occured by scv file?

WCCWCC, 2019-09-19:
Hi isDerek,

     the node provisioned fail  ->  Can you provide me with the relevant log.

isDerek, 2019-09-20:
```
W (213936) BLE_MESH: Receive configuration status message timeout
I (213938) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0005, opcode: 0x8008
W (215768) BLE_MESH: Receive configuration status message timeout
I (215770) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0009, opcode: 0x8008
W (215783) BLE_MESH: Receive configuration status message timeout
I (215787) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0006, opcode: 0x8008
W (216952) BLE_MESH: Receive configuration status message timeout
I (216954) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0005, opcode: 0x8008
W (218787) BLE_MESH: Receive configuration status message timeout
I (218789) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0009, opcode: 0x8008
W (218802) BLE_MESH: Receive configuration status message timeout
I (218804) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0006, opcode: 0x8008
W (219967) BLE_MESH: Receive configuration status message timeout
I (219969) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0005, opcode: 0x8008
W (221804) BLE_MESH: Receive configuration status message timeout
I (221806) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0009, opcode: 0x8008
W (221819) BLE_MESH: Receive configuration status message timeout
I (221823) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0006, opcode: 0x8008
W (222983) BLE_MESH: Receive configuration status message timeout
I (222985) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0005, opcode: 0x8008
W (224823) BLE_MESH: Receive configuration status message timeout
I (224825) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0009, opcode: 0x8008
W (224838) BLE_MESH: Receive configuration status message timeout
I (224840) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0006, opcode: 0x8008
W (225999) BLE_MESH: Receive configuration status message timeout
I (226001) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0005, opcode: 0x8008
E (226015) BLE_MESH: net_buf_alloc():Failed to get free buffer
E (226016) BLE_MESH: send_unseg, Out of network buffers
E (226027) BLE_MESH: bt_mesh_cfg_comp_data_get, send failed (err -105)
I (226029) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0xffffff97, event = 0x00, addr: 0x0005, opcode: 0x8008
E (226042) ble_mesh_provisioner: Send config client message failed, opcode 0x8008
W (227840) BLE_MESH: Receive configuration status message timeout
I (227842) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0009, opcode: 0x8008
W (227855) BLE_MESH: Receive configuration status message timeout
E (227857) BLE_MESH: net_buf_alloc():Failed to get free buffer
E (227868) BLE_MESH: send_unseg, Out of network buffers
E (227870) BLE_MESH: bt_mesh_cfg_comp_data_get, send failed (err -105)
I (227881) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x03, addr: 0x0006, opcode: 0x8008
I (227894) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0xffffff97, event = 0x00, addr: 0x0009, opcode: 0x8008
E (227907) ble_mesh_provisioner: Send config client message failed, opcode 0x8008
E (227909) BLE_MESH: net_buf_alloc():Failed to get free buffer
E (227920) BLE_MESH: send_unseg, Out of network buffers
E (227921) BLE_MESH: bt_mesh_cfg_comp_data_get, send failed (err -105)
I (227933) ble_mesh_provisioner: esp_ble_mesh_config_client_cb, error_code = 0xffffff97, event = 0x00, addr: 0x0006, opcode: 0x8008
E (227946) ble_mesh_provisioner: Send config client message failed, opcode 0x8008

``

Campou, 2019-09-25:
Hi @isDerek 

The issue happens because BLE Mesh failed to create internal adv task. And when BLE Mesh coexists Wi-Fi, we recommend to use ESP32-DevKitC with PSRAM on board.

We have pushed a bugfix to the branch `ble_mesh_release/esp-ble-mesh-v0.6.1` to check if the task in BLE Mesh is created successfully.

Thanks for your patience

Alvin1Zhang, 2019-10-11:
@isDerek Thanks for reporting, feel free to reopen if the issue still exists. Thanks.

closed 2019-10-11 by Alvin1Zhang