espressif/esp-idf #4099: [BLE Mesh] provisioner doesn't provision with net_idx not equal to 0

jkockx, 2019-09-19:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v4.0-dev-1191-g138c941fa
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

I'm trying to make my provisioner provision devices with a net_idx other than zero.  However, whatever I try, the provisioner keeps provisioning the device with net_key 0.I use following code for this:
```
// to add another network key with net_idx
void add_net_key_local(uint16_t net_idx) {
	esp_err_t err;
	err = esp_ble_mesh_provisioner_add_local_net_key(NULL, net_idx);
	if (err == ESP_OK) {
		ESP_LOGI(TAGPROV, "%s: success", __func__);
		mqtt_prov_add_net_key_local(1, net_idx);
	} else {
		ESP_LOGI(TAGPROV, "%s: failed", __func__);
		mqtt_prov_add_net_key_local(0, net_idx);
	}
}
uint16_t net_key = 1;
add_net_key_local(net_key)

 esp_ble_mesh_prov_data_info_t prov_data_info;
    prov_data_info.net_idx = net_idx;
    prov_data_info.flags = NET_IDX_FLAG;
    esp_ble_mesh_provisioner_set_prov_data_info(&prov_data_info);

    /* Note: If unprovisioned device adv packets have not been received, we should not add
             device with ADD_DEV_START_PROV_NOW_FLAG set. */
    err = esp_ble_mesh_provisioner_add_unprov_dev(add_dev,
            ADD_DEV_RM_AFTER_PROV_FLAG | ADD_DEV_START_PROV_NOW_FLAG | ADD_DEV_FLUSHABLE_DEV_FLAG);
```
## Debug logs
```
I (227228) BLUETOOTH: ESP_BLE_MESH_PROVISIONER_PROV_COMPLETE_EVT
I (227228) BLUETOOTH: node index: 0x1, unicast address: 0x06, element num: 1, netkey index: 0x00
I (227238) BLUETOOTH: device uuid: d00e1fecf2b05302ae8d9b8690fd5ace
I (227248) MQTT: sent publish to topic /topic/provisioner/status unsuccessful, msg_id=0
I (227248) BLUETOOTH: ESP_BLE_MESH_PROVISIONER_SET_NODE_NAME_COMP_EVT, err_code 0
I (227258) BLUETOOTH: Node 1 name is: NODE-1
I (229128) BLUETOOTH: PB-ADV link close, reason 0x00
```

jkockx, 2019-09-20:
OK, I learned from the specification that this behaviour is normal as always a primary network key (net_idx 0) is added during provisioning and I had to add other network key through the configuration model. This leaves me with the following question, what are the following lines of code setting?
```
esp_ble_mesh_prov_data_info_t prov_data_info;
prov_data_info.net_idx = net_idx;
prov_data_info.flags = NET_IDX_FLAG;
esp_ble_mesh_provisioner_set_prov_data_info(&prov_data_info);
```

John-nx, 2019-09-23:
Hi @jkockx,

May I ask you something?
Why should one use a micro as a provisioner while it is easier to use a smartphone or computer device as it is stated in the Bluetooth website? I mean when it can be useful?

Thanks

Campou, 2019-09-25:
Hi @jkockx ,

The API `esp_ble_mesh_provisioner_set_prov_data_info()` is used change the `net_idx`, `IV Index`, etc. which will be used during provisioning.

If you want to use NetKey with NetKeyIndex other than 0x0000 during provisioning, the procedure is:
1. Add local NetKey with proper NetKeyIndex, for example 0x0001
2. Use the API `esp_ble_mesh_provisioner_set_prov_data_info()` to set the NetKeyIndex
3. Start to scan and provision unprovisioned devices

Thanks.

Alvin1Zhang, 2019-10-11:
@jkockx Thanks for reporting, feel free to reopen if the issue still exists. Thanks.

closed 2019-10-11 by Alvin1Zhang