espressif/esp-idf #4101: error provisioning devices with Bluetooth Mesh under Bluetdroid

aircable, 2019-09-20:
We use Wifi and Bluetooth Mesh at the same time, under Bluedroid. 
A mesh node provisioning crashed with a stack overflow.

Turns out that BTU_TASK_STACK_SIZE is configured to small.

Changing the file components/bt/host/bluedroid/stack/btu/btu_init.c
to add another 2k of space, allow the provisioning to run successful.

```
#define BTU_TASK_STACK_SIZE             (2048 + BT_BTU_TASK_STACK_SIZE + BT_TASK_EXTRA_STACK_SIZE)
```

enricop, 2019-09-23:
We have the same issue if we click on "Get Composition Data" on the nRF_Mesh Android app after provisioning.

Increasing the stack size fixes the overflow.

enricop, 2019-09-24:
this can be done with CONFIG_BTU_TASK_STACK_SIZE

gnameg, 2023-08-11:
[image]
Hi,
Maybe, you can adjust BTC or BTU stack size in menuconfig.