espressif/esp-idf #4102: I2S Issues

lachlanleslie1995, 2019-09-20:
Hello, 

I am having some issues playing sound via I2S to a MAX98357A. I'm trying to write a 1KHz sine wave that I've defined in an array and am passing to i2s_write(). 

This is where I'm running into issues. The sound being produced is not even close to a sine wave, it is more akin to a click (repeated at a high rate). And the signal never ends the sound will keep playing till the buffer is cleared, from my understanding of diving into the function itself it should end not make noise forever.

- IDF version: v4.0-beta1-40-gda06e20c1
- Build System:      CMake
- Chip: ESP32-WROOM-32

This is my setup below:
`
int i2s_num = I2S_NUM_0;
	 i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_TX,                                
        .sample_rate = 36000,
        .bits_per_sample = 16,
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,                           
        .communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB,
        .dma_buf_count = 6,
        .dma_buf_len = 60,
        .use_apll = false,
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1                               
    };
	 
	 i2s_pin_config_t pin_config = {
            .bck_io_num = I2S_BCLK_PIN_NUM,
            .ws_io_num = I2S_LRCLK_PIN_NUM,
            .data_out_num = I2S_DIN_PIN_NUM,
            .data_in_num = I2S_PIN_NO_CHANGE
    };
    
	 i2s_driver_install(i2s_num, &i2s_config, 0, NULL);

	 i2s_set_pin(I2S_NUM_0, &pin_config);
    

   size_t i2s_bytes_write = 0;

   i2s_set_clk(0, SAMPLE_RATE, 16, 1);

   i2s_write(0, sine_wav_1kHz, sizeof(sine_wav_1kHz), &i2s_bytes_write, 100);

`

Please advise on what the issue is or if you need any further information.

Cheers

Alvin1Zhang, 2019-09-20:
@lachlanleslie1995 Thanks for reporting, we will look into this. Thanks.

lachlanleslie1995, 2019-09-22:
@Alvin1Zhang Any update on this?

Alvin1Zhang, 2019-09-23:
@lachlanleslie1995 Thanks for the patience, we are still working on this, will update once available. Thanks.

jason-mao, 2019-09-23:
@lachlanleslie1995 Hi, I am not sure what's the root case of  your problem, but there are some suggestions as follow, please test.

- `.dma_buf_len = 60,`->`.dma_buf_len = 200,`
- `i2s_write(0, sine_wav_1kHz, sizeof(sine_wav_1kHz), &i2s_bytes_write, 100);`->`i2s_write(0, sine_wav_1kHz, sizeof(sine_wav_1kHz), &i2s_bytes_write, portMAX_DELAY);`

lachlanleslie1995, 2019-09-23:
Hey @jason-mao,

Tried what you've suggested. The only change these made was the "clicking" sound was repeated slower. To me it almost sounds like the speaker is turning on and off repeatedly. Could you suggest anything else? Or potentially is there more information that would assist you?

Here is the samples I'm trying to output:

`const int16_t sine_wav_1kHz[] = {0,5690,11207,16384,21062,25101,28377,30791,32269,32767,32269,30791,28377,25101,21062,16384,11207,5690,0,-5690,-11207,-16384,-21062,-25101,-28377,-30791,-32269,-32767,-32269,-30791,-28377,-25101,-21062,-16384,-11207,-5690};
`

Cheers

lachlanleslie1995, 2019-09-24:
Hey @Alvin1Zhang & @jason-mao ,

Any follow up on this?

I would like to get this solved urgently if you could please assist.

Cheers

lachlanleslie1995, 2019-09-29:
Any updates @Alvin1Zhang @jason-mao  ?

koobest, 2020-05-28:
@lachlanleslie1995 
Why you set to one channel by calling ` i2s_set_clk(0, SAMPLE_RATE, 16, 1);` ?

MikaBeats, 2022-07-15:
I have the same problem, did you find a solution? On arduino everything works fine but when I go through esp-idf nothing happens except 2 sharp clicks and then nothing, I used the same code as you, my pin configuration is the following:

bck = 26
ws = 25
data out = 19

and I use an esp32 wroom 32.
Thanks!

egnor, 2022-07-27:
[this seems obsolete and should probably be closed?]

It's hard to tell, because the code in the OP is clearly partial (not a complete program), but it looks like they are only calling `i2c_write()` for one single instance of the waveform, rather than repeatedly/continuously outputting the waveform to generate an ongoing tone.

This may be because the old [i2s basic example](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/peripherals/i2s/i2s_basic/main/i2s_example_main.c) did something sneaky and tricky where it allocated precisely enough buffer space for precisely one waveform, filled it, and then never wrote any more data, allowing the DMA engine to just repeatedly loop over that one waveform. If you attempt something similar but don't have a buffer size that precisely matches your waveform, you'll get your single waveform repeated with a lot of silence between invocations.

The [new basic example](https://github.com/espressif/esp-idf/blob/release/v5.0/examples/peripherals/i2s/i2s_basic/main/i2s_example_main.c) in v5.0+ has separate read and write tasks, and the write task continuously generates the waveform, to be less confusing.