espressif/esp-idf #4105: idf.py fails when git init is ran but not committed

anozaki, 2019-09-21:
idf.py fails when `git init` is call in the directly but if you have not committed the initialized git repository.

```bash
cd esp-idf/example/hello_world
git init
idf.py menuconfig
```
You will get the following error.
`CMake Error: File /Volumes/ExternalSSD/Temp/hello_world/.git/packed-refs does not exist.`

### Work Around
Commit the repository with something before running `idf.py`

```bash
git add README.md
git commit -m"Test"
idf.py menuconfig
```

Alvin1Zhang, 2019-09-23:
@anozaki Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

morganrallen, 2019-09-28:
I believe this is a duplicate #3920 and has been addressed.

Alvin1Zhang, 2019-09-29:
@morganrallen Thanks for the notes. @anozaki Will close this due to short of feedback. Feel free to reopen if the issue still exists. Thanks.

closed 2019-09-29 by Alvin1Zhang