espressif/esp-idf #4108: Clear WIFI without clearing NV

devropes, 2019-09-22:
I wanted to know if there is some sample code to erase the ESP32's WIFI information without clear all of the NV memory?

Currently, the only method I know of is `nvs_flash_erase()`.

negativekelvin, 2019-09-23:
Possibly like this

```
nvs_handle_t net80211_handle;
nvs_open("nvs.net80211", NVS_READWRITE, &net80211_handle);
nvs_erase_all(net80211_handle);
nvs_commit(net80211_handle);
nvs_close(net80211_handle);
```

sagb2015, 2019-09-23:
@devropes - @negativekelvin answer should do it. However,  we will add an API to take care of this as users need not know the internal details such as partition name do perform this operation.

sagb2015, 2019-09-26:
@devropes We have _esp_wifi_restore_ API to set the WiFi persistent settings to default values. Do you intend to reset the values to default or cleanup to save on the space?

devropes, 2019-09-27:
Reset values to default.

sagb2015, 2019-09-27:
"esp_wifi_restore" should work for your requirement then. Please try and let us know.

sagb2015, 2019-09-27:
Closing the issue. Feel free to open this again if it does not satisfy your criteria.

closed 2019-09-27 by sagb2015