espressif/esp-idf #4111: dl.espressif.com certificate expired

sieren, 2019-09-24:
`This server could not prove that it is dl.espressif.com; its security certificate expired yesterday. This may be caused by a misconfiguration or an attacker intercepting your connection. Your computer's clock is currently set to Tuesday, 24 September 2019. `

Looks like the download server needs to renew its certificate. Running the installation scripts is therefore currently broken due to an invalid certificate.

Alvin1Zhang, 2019-09-25:
@sieren Thanks for the heads-up. We are now looking into this. Thanks.

Alvin1Zhang, 2019-09-25:
@sieren Hi. We have fixed the certificate, would you please help try again? Thanks.

sieren, 2019-09-25:
@Alvin1Zhang Nice one! It's back! Thank you :)

closed 2019-09-25 by sieren