espressif/esp-idf #4112: Problem building project with components dir

jkockx, 2019-09-24:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
v4.1-dev-256-g9f145ff16
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

I'm working on a second esp-idf embedded project. As with my previous project, the project lacked structure in the project directory. Now I've decided to work with a components directory in which I can further structure my source files in the future.
I tried to start from the project template and added the components folder with a component.mk file with it.
However when I try to build the project (in Eclipse, with configuration correctly set as described in https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/eclipse-setup.html)

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. clone my project
2. set up configuration in Eclipse
3. run make menuconfig in termianl
4. build the project

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

An example of my problem is present in the following git repo:
https://github.com/jkockx/coffee_project 

## Debug Logs

```
/home/jonas/esp/coffee-project/main/main.c:9:10: fatal error: coffee_output.h: No such file or directory
 #include "coffee_output.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
echo "/home/jonas/esp/esp-idf/components/pthread/./pthread_local_storage.c:" >> pthread_local_storage.d
echo "/home/jonas/esp/esp-idf/components/protocomm/src/common/protocomm.c:" >> src/common/protocomm.d
echo "/home/jonas/esp/esp-idf/components/openssl/library/ssl_cert.c:" >> library/ssl_cert.d
/home/jonas/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: Leaving directory '/home/jonas/esp/coffee-project/build/main'
make: *** [component-main-build] Error 2
make: *** Waiting for unfinished jobs....
/home/jonas/esp/esp-idf/make/project.mk:584: recipe for target 'component-main-build' failed
```

igrr, 2019-09-25:
Please put coffee_output.h into include directory of the component, or update the component Makefile to add component root directory (`.`) to the list of include directories.

igrr, 2019-10-29:
@jkockx Please feel free to reopen if the suggestion above didn't work.

closed 2019-10-29 by igrr