espressif/esp-idf #4113: std::stringstream broken

sieren, 2019-09-24:
## Environment

- Development Kit:  ESP32-Wrover-Kit   
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-beta1-134-gef17d8bba
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   1.22.0-80-g6c4433a
- Operating System:    macOS

## Problem Description

std::stringstream is broken.

### Expected Behavior

Output: "4.0" for float = 4;

### Actual Behavior

Output: "f"

### Steps to repropduce

See code.

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"
#include <sstream>
#include <string>
void app_main()
{
    retVal = 4;
    std::stringstream stream;
    stream << std::fixed;
    stream << std::setprecision(1);
    stream << retVal;
    std::string s = stream.str();
    ESP_LOGI("TST", "%s", s.c_str());
}

```

Alvin1Zhang, 2019-09-25:
@sieren Thanks for reporting, we will look into. Thanks.

igrr, 2019-09-25:
I've noticed you are using the toolchain version `1.22.0-80-g6c4433a`. Could you please install the required version by following the Getting Started Guide? https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#installation-step-by-step. The version `1.22.0-80-g6c4433a` is correct for v3.x versions of IDF, but 4.x use a different version.

sieren, 2019-09-25:
Thank you, that was an oversight on my behalf. Had the old version in $PATH as well.
Either way, unfortunately that did not solve the issue.
Occurs with `xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0` as well

igrr, 2019-09-26:
Sorry, i couldn't reproduce the issue with esp32-2019r1 toolchain and IDF commit ef17d8bba1.

My test is a slightly modified version of yours (which did not compile):
```c++
#include <string>
#include <sstream>
#include <iomanip>
#include "esp_log.h"

extern "C" void app_main()
{
    // volatile to make sure the following steps are not done at compile time
    volatile float retVal = 4;
    std::stringstream stream;
    stream << std::fixed;
    stream << std::setprecision(1);
    stream << retVal;
    ESP_LOGI(__func__, "%s", stream.str().c_str());
}
```
Prints:
```
I (10969) app_main: 4.0 
```

Can you please try to run this exact code and check that you get the same output?

> Thank you, that was an oversight on my behalf. Had the old version in $PATH as well.

By the way, after changing your PATH to point to the new compiler, make sure you delete `build/CMakeCache.txt`, as CMake stores the compiler path there. Or do `idf.py fullclean` just to be sure...

sieren, 2019-09-26:
I suppose it was an issue in the toolchain/sdkconfig. But it resolved itself. Thanks for your assistance.

closed 2019-09-26 by sieren