espressif/esp-idf #4114: sntp_getserver retrieves blank ip with DNS

sieren, 2019-09-24:
https://github.com/espressif/esp-idf/issues/3756
## Environment

- Development Kit:  ESP32-Wrover-Kit   
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-beta1-134-gef17d8bba
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   1.22.0-80-g6c4433a
- Operating System:    macOS

## Problem Description

SNTP does not retrieve an IP address.
This was fixed in https://github.com/espressif/esp-idf/commit/fa02598b5c978402c1013b22ca81919beded97f3 but forgotten to be backported to `release/v4.0`

The commit title says it was meant to be for 4.0, but the merge happened to 4.1 (see: https://github.com/espressif/esp-idf/commit/fa02598b5c978402c1013b22ca81919beded97f3)

Alvin1Zhang, 2019-09-25:
@sieren Thanks for reporting, we will look into. Thanks,

KonstantinKondrashov, 2019-09-25:
Hi @sieren!
I have checked the behavior of `sntp_getserver(0) should return valid ip` on v4.0. It works as expected. I think your branch was not updated whole (See [updating-to-a-release-branch](https://docs.espressif.com/projects/esp-idf/en/latest/versions.html#updating-to-a-release-branch)). Please run `git submodule update --init --recursive`.
(To fix this issue we updated the lwip submodule).

sieren, 2019-09-25:
@KonstantinKondrashov youre right, it's on the current lwip. However I'm still running into the same issue.

```cpp
namespace ntp
{

  NTPSync::NTPSync(const std::string timeZone) :
    mTimeZone(timeZone)
  {
    initialize_sntp();
  }

  void NTPSync::syncTime()
  {
    time_t now;
    struct tm timeinfo;
    time(&now);
    localtime_r(&now, &timeinfo);

    obtain_time();
    // update 'now' variable with current time
    time(&now);

    char strftime_buf[64];
    setenv("TZ", mTimeZone.c_str(), 1);
    tzset();
    localtime_r(&now, &timeinfo);
    strftime(strftime_buf, sizeof(strftime_buf), "%c", &timeinfo);
  }
} // namespace ntp

static void obtain_time(void)
{
    // wait for time to be set
    time_t now = 0;
    struct tm timeinfo = { };
    int retry = 0;
    const int retry_count = 10;
    while (sntp_get_sync_status() == SNTP_SYNC_STATUS_RESET && ++retry < retry_count) {
        ESP_LOGI(TAG, "Waiting for system time to be set... (%d/%d)", retry, retry_count);
        const ip_addr_t*ip = sntp_getserver(0);
        printf("SNTP Server IP: %s\n", ipaddr_ntoa(ip));
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
    time(&now);
    localtime_r(&now, &timeinfo);
}

static void time_sync_notification_cb(struct timeval *tv)
{
    ESP_LOGI(TAG, "Notification of a time synchronization event");
}

static void initialize_sntp(void)
{
    ESP_LOGI(TAG, "Initializing SNTP");
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    char serverName[] = "pool.ntp.org"; // Should be const, blame the sntp API.
    sntp_set_time_sync_notification_cb(time_sync_notification_cb);
    sntp_setservername(0, serverName);
    sntp_init();
}```

The code in question is pretty much the one taken from https://github.com/espressif/esp-idf/blob/master/examples/protocols/sntp/main/sntp_example_main.c but wrapped in C++.

```
I (5304) tcpip_adapter: sta ip: 192.168.8.174, mask: 255.255.255.0, gw: 192.168.8.1
I (5305) wifi station: got ip:192.168.8.174
I (5307) NTP: Waiting for system time to be set... (1/10)
SNTP Server IP: 0.0.0.0
I (7313) NTP: Waiting for system time to be set... (2/10)
SNTP Server IP: 0.0.0.0
I (9313) NTP: Waiting for system time to be set... (3/10)
SNTP Server IP: 0.0.0.0
I (11313) NTP: Waiting for system time to be set... (4/10)
...```

KonstantinKondrashov, 2019-09-25:
@sieren 
Probably you have to use `sntp_getserver(0)` after updated of time.
My test - [test_sntp_getserver.zip] (https://github.com/espressif/esp-idf/files/3651007/test_sntp_getserver.zip) - patch for sntp example.
Can you use `sntp_getserver (0)` after getting the time? What is the purpose to get this in a loop?

KonstantinKondrashov, 2019-09-25:
Can you use like this?
```c
static void obtain_time(void)
{
    // wait for time to be set
    time_t now = 0;
    struct tm timeinfo = { };
    int retry = 0;
    const int retry_count = 10;
    while (sntp_get_sync_status() == SNTP_SYNC_STATUS_RESET && ++retry < retry_count) {
        ESP_LOGI(TAG, "Waiting for system time to be set... (%d/%d)", retry, retry_count);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
    time(&now);
    localtime_r(&now, &timeinfo);

    // get ip after that loop
    const ip_addr_t*ip = sntp_getserver(0);
    printf("SNTP Server IP: %s\n", ipaddr_ntoa(ip));
}
```

sieren, 2019-09-25:
Thanks for the assistance. The example shipped with ESP-IDF works for me, so I suppose the culprit is somewhere in my code or sdkconfig. I'll keep on digging and will close this.

closed 2019-09-25 by sieren

sieren, 2019-09-25:
The issue was 
```cpp
     char serverName[] = "pool.ntp.org"; 
    sntp_set_time_sync_notification_cb(time_sync_notification_cb);
    sntp_setservername(0, serverName);
```

serverName probably goes out of scope and the sntp client only holds a reference.

```
-    char serverName[] = "pool.ntp.org"; // Should be const, blame the sntp API.
-    sntp_setservername(0, serverName);
+    sntp_setservername(0, "pool.ntp.org");
```
was the fix.