espressif/esp-idf #4115: esp_restart() fails when called from timer callback when WiFi is enabled

vonnieda, 2019-09-24:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROVER-IB
- IDF version (run ``git describe --tags`` to find it): v3.3
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: macOS
- Power Supply: USB

## Problem Description

Once WiFi has been initialized, if esp_restart() is called from an esp_timer callback the chip does not restart.

### Expected Behavior

The chip restarts.

### Actual Behavior

The chip appears to start to shutdown WiFi and then does not restart. The chip is still responsive and other tasks can run.

### Steps to repropduce

1. Run attached example based on hello_world and observe that esp_restart() does not work.
2. Modify hello_world_main.c and comment out init_wifi().
3. Run again and observe that esp_restart() now works.

### Code to reproduce this issue

See attached example project.

## Debug Logs

None apply.

## Other items if possible

See attached example project.

[esp_restart_issue.zip](https://github.com/espressif/esp-idf/files/3649455/esp_restart_issue.zip)

Note that this appears to be related to #2101, but since I can now reproduce it with a simple example I thought a new issue was warranted.

Alvin1Zhang, 2019-09-25:
@vonnieda Thanks for reporting, we will look into this. Thanks.

igrr, 2019-09-26:
I think this was originally reported in https://github.com/espressif/esp-idf/issues/3458 and fixed (in two independent ways) in https://github.com/espressif/esp-idf/commit/6b08e8b44966fadba83e09cbc47f6f86bac03e51 and https://github.com/espressif/esp-idf/commit/197e0ae147d7b150d0ffaa482ed670e7d6c2b561. It seems that these fixes never made it to the v3.3 release. Can you please try cherry-picking the former commit and see if it fixes the issue you are observing?

vonnieda, 2019-09-26:
Thanks for the response @igrr; which commit do you want me to try to cherry pick? There are two commits with the message "fix some wifi bugs",  but they seem to affect different parts of the tree, and do not apply cleanly, and there is the " esp_timer: handle esp_timer_delete in timer task " one. I was able to apply that one, and then experienced a deadlock in esp_timer during normal operation, so I am not sure if I need both. Can you provide some more guidance?

Thanks,
Jason

igrr, 2019-09-26:
The "esp_timer: handle esp_timer_delete in timer task" one should be sufficient, I think. Can you please give any more information about the deadlock? If you are able to attach a JTAG debugger, you can interrupt the program and get the backtrace.

vonnieda, 2019-09-26:
@igrr I've confirmed that https://github.com/espressif/esp-idf/commit/6b08e8b44966fadba83e09cbc47f6f86bac03e51 fixes the esp_restart() issue. The deadlock may be unrelated, I will monitor and let you know if it happens again. I had a I2C bus contention issue that may have caused it.

vonnieda, 2019-09-27:
@igrr I have not seen the deadlock again across several devices, so I think that was just my side. So, end result is that https://github.com/espressif/esp-idf/commit/6b08e8b44966fadba83e09cbc47f6f86bac03e51 seems to resolve this issue.

igrr, 2019-09-30:
@vonnieda Ok, we will backport 6b08e8b to release/v3.3.

vonnieda, 2019-09-30:
Sounds good, thank you!

jack4k, 2019-12-13:
This worked for me, if it helps anyone else until this is fixed.

work around code:

```
taskENTER_CRITICAL();
esp_restart();	
taskEXIT_CRITICAL();
```

Alvin1Zhang, 2020-05-28:
@vonnieda Thanks for reporting and sorry for slow turnaround. The fix on release/3.3 is available at https://github.com/espressif/esp-idf/commit/dbd2cd2f7a4f1907e5aa3a46df64dfdf7e66eca0. Feel free to reopen if the issue still happens. Thanks.

closed 2020-05-28 by Alvin1Zhang