espressif/esp-idf #4116: nvs_flash_init()  ESP_ERR_NOT_FOUND

lachlanleslie1995, 2019-09-25:
Hello,

I'm experiencing some issues with some nvs initialisation for ble. It it saying it can't find nvs partition in the partition table but I've tried to use either of the partition tables that are defaulted and also a custom table but all of them are still unable to find nvs.

IDF ver: v4.1-dev-281-g96b96ae24

Expected 
NVS initialises

Actual
nvs_flash_init fails to find nvs paritions.

Cheers

igrr, 2019-09-25:
Hi @lachlanleslie1995, could you please add more details, such as: the code you are running, sdkconfig file, partition table file (if it is a custom partition table)? Also please attach the logs you get from UART, starting from the bootloader output. Thanks.

lachlanleslie1995, 2019-09-26:
Hey @igrr,

here is the particular section of code that is failing

`ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );`

Here is the UART output:

> Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory c:\users\lachlan\documents\talkiplay\sourcecode
Executing "C:\Users\lachlan\.espressif\python_env\idf4.0_py2.7_env\Scripts\python.exe C:\esp\esp-idf\tools/idf_monitor.py -p COM12 -b 115200 c:\users\lachlan\documents\talkiplay\sourcecode\build\SourceCode.elf -m 'C:\Users\lachlan\.espressif\python_env\idf4.0_py2.7_env\Scripts\python.exe' 'C:\esp\esp-idf\tools\idf.py'"...
--- idf_monitor on COM12 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
05) bootlo ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8716
load:0x40078000,len:14748
ho 0 tail 12 room 4
load:0x40080400,len:4636
entry 0x4008071c
D (75) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (37) boot: ESP-IDF v4.1-dev-281-g96b96ae24-dirty 2nd stage bootloader
I (37) boot: compile time 09:37:01
D (38) boot: Enabling RTCWDT(9000 ms)
I (42) boot: Enabling RNG early entropy source...
D (48) boot: magic e9
D (50) boot: segments 04
D (53) boot: spi_mode 02
D (55) boot: spi_speed 00
D (58) boot: spi_size 02
I (60) boot: SPI Speed      : 40MHz
I (65) boot: SPI Mode       : DIO
I (69) boot: SPI Flash Size : 4MB
D (73) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (82) boot: mapped partition table 0x8000 at 0x3f408000
D (87) flash_parts: partition table verified, 6 entries
I (92) boot: Partition Table:
I (96) boot: ## Label            Usage          Type ST Offset   Length
D (103) boot: load partition table entry 0x3f408000
D (108) boot: type=1 subtype=2
I (111) boot:  0 nvs              WiFi data        01 02 00009000 00004000
D (119) boot: load partition table entry 0x3f408020
D (124) boot: type=1 subtype=0
I (127) boot:  1 otadata          OTA data         01 00 0000d000 00002000
D (134) boot: load partition table entry 0x3f408040
D (139) boot: type=1 subtype=1
I (142) boot:  2 phy_init         RF data          01 01 0000f000 00001000
D (150) boot: load partition table entry 0x3f408060
D (155) boot: type=0 subtype=0
I (158) boot:  3 factory          factory app      00 00 00010000 00100000
D (166) boot: load partition table entry 0x3f408080
D (171) boot: type=0 subtype=10
I (174) boot:  4 ota_0            OTA app          00 10 00110000 00100000
D (181) boot: load partition table entry 0x3f4080a0
D (186) boot: type=0 subtype=11
I (190) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (197) boot: End of partition table
D (201) boot: OTA data offset 0xd000
[... 241 lines cut ...]

0x40090d93: _esp_error_check_failed at C:/esp/esp-idf/components/esp32/panic.c:740

0x400d68f9: initBluetooth at c:\users\lachlan\documents\talkiplay\sourcecode\build/../main/bluetooth.c:695 (discriminator 1)

0x400d611b: activateBluetooth at c:\users\lachlan\documents\talkiplay\sourcecode\build/../main/talkiPlayer.c:29

0x400d60ae: app_main at c:\users\lachlan\documents\talkiplay\sourcecode\build/../main/main.c:120

0x400d307a: main_task at C:/esp/esp-idf/components/esp32/cpu_start.c:563

0x40094119: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

Here is the sdfconfig:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3660272/sdkconfig.zip)

As to the partition table I'm using I've used both the default ones provided by espressif and haven't needed to change until this has occurred. I tried using a custom one but the same issue occurred so I'm back to using the default one.

To debug this I've tried erasing the flash on the chip as well as deleting the build folder and doing a complete rebuild

Cheers,
Lachlan

lachlanleslie1995, 2019-09-29:
Any update @igrr  ?

igrr, 2019-09-30:
Indeed this looks odd, the log at startup indicates that `nvs` partition exists, but later the error happens when looking for the partition. Can you please add the following code into your app, before initializing the NVS? It will list the partitions from the partition table:

```c++
    esp_partition_iterator_t it;
    ESP_LOGI(TAG, "Iterating through data partitions...");
    it = esp_partition_find(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, NULL);

    // Loop through all matching partitions, in this case, all with the type 'data' until partition with desired 
    // label is found. Verify if its the same instance as the one found before.
    for (; it != NULL; it = esp_partition_next(it)) {
        const esp_partition_t *part = esp_partition_get(it);
        ESP_LOGI(TAG, "\tfound partition '%s' at offset 0x%x with size 0x%x", part->label, part->address, part->size);
    }

    // Release the partition iterator to release memory allocated for it
    esp_partition_iterator_release(it);
```

(The code is from the examples/storage/partition_api/partition_find example)

negativekelvin, 2019-09-30:
It is possible to have this error if there is no free heap.

igrr, 2019-09-30:
> The only unusual thing is an external flash chip running a fatfs partition.

Indeed, this seems to be what is causing the issue. Well spotted @negativekelvin, thanks. I can now reproduce the issue.

igrr, 2019-09-30:
@lachlanleslie1995 As a workaround, you can add the following line *before* you initialize the external flash:

```
esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, NULL);
```

This will ensure that the main partition table has been loaded, prior to adding the partition from external flash. The proper fix will be merged soon.

lachlanleslie1995, 2019-10-01:
Thanks @igrr and @negativekelvin really appreciate your support in this.

I am trying this workaround now, will let you know if it is successful. Maybe an addition to documentation or some fix could be created to prevent this workaround being required.

Cheers,
Lachlan

lachlanleslie1995, 2019-10-01:
Success! 
That solves my issue, just for my own understanding can you confirm exactly what the issue was @igrr ?
Was it identifying the external flash chip first and looking for the "nvs" partition there instead of the internal storage?|

Cheers,
Lachlan

igrr, 2019-10-01:
The issue is that the partitions from the main partition table were loaded on the first call to esp_partition_find, if the list of partitions was empty. However once an external partition was added to the list, the list was no longer empty, so loading of the main partition table would never be triggered.

The fix is in review at the moment and will be merged soon. This issue will be updated when that happens.

closed 2019-10-13 by igrr

khconservation, 2023-05-31:
You will also get the 0x105 (ESP_ERR_NOT_FOUND) if you have selected that the partition table has an MD5 checksum, and the checksum is missing.