espressif/esp-idf #4117: Voltage spikes at 5Hz in GPIO36 causing interrupts in release v3.3, v3.2.2 works fine

lrodenasSistel, 2019-09-25:
## Environment

- Module used:  [ESP32-WROVER-B in custom pcb]
- IDF version: v3.3-51-g5e321badd-dirty
- Build System:         [Make]
- Compiler version: (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB+Battery]

## Problem Description

ISR attached to interrupt of GPIO36 is being launched repeatedly at 5 Hz aprox.
Pin has external pullup to 3.3V, interrupt is configured to negative edge.
After checking with oscilloscope I see there are some spikes in GPIO36 at 5Hz aprox. These spikes go from 3.3V to 1.5V aprox, but my oscilloscope is not high-end.

My code is just connected to wifi network and doing some http request every 10 seconds or so.

**In IDF v3.2.2 this does not happen. There are no spikes in GPIO36 with same code.**
If I turn off wifi and go to sleep with this interrupt as wakeup, it works fine = no wakeup until real interrupt happens.

Seems related to bug 3.11 of "ECO and workarounds for bugs in ESP32" document.

Seems related to Issue #1953 https://github.com/espressif/esp-idf/issues/1953
I also have this option set up esp_wifi_set_ps(WIFI_PS_MIN_MODEM), but it is also setup in v3.2.2 and does not happen.

### Expected Behavior
No spikes on GPIO36 as in IDFv3.2.2.

Thank you very much for your support and please let me know if you need any extra detail about this.

Alvin1Zhang, 2019-09-26:
@lrodenasSistel Thanks for reporting, we will look into this issue. Thanks.

negativekelvin, 2019-09-26:
https://github.com/espressif/esp-idf/issues/3714#issuecomment-512141448

philippe44, 2020-02-03:
I have a related issue when using 36/39 as digital input for push buttons. When 2 push buttons (push to ground with 33k pull-ups) are connected to these GPI, every time one button is pushed, I see a spurious interrupt on the other input, as if they were connected.

Indeed, the wiring I have creates a small crosstalk  between the inputs (say a few x0mV drop on one GPI when the other is pressed), but that should not matter at all in the digital domain. 

It feels to me that 36/39 still behaves as analogue inputs converted to digital by software and what is being caught is a sort of spurious differential. Could it be the case? To support that idea, if I filter the push button with a 470 nF, the issue does not happen, but that's a hardware modification that is difficult to do atm.

The exact same setup works fine on any other GPIO. I'm using an old IDF (3.1.5)

closed 2020-12-13 by espressif-bot