espressif/esp-idf #4118: install.bat urllib.error.URLerror: <urlopen error unknown url type: https>

AldyKwee, 2019-09-26:
I am trying to get started in ESP-IDF. But when I try to run "install.bat", it gives this error:

Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Installing idf-exe@1.0.1
Downloading idf-exe-v1.0.1.zip to C:\Users\user\.espressif\dist\idf-exe-v1.0.1.zip.tmp
Traceback (most recent call last):
  File "C:\Users\user\esp\esp-idf\tools\idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "C:\Users\user\esp\esp-idf\tools\idf_tools.py", line 1313, in main
    action_func(args)
  File "C:\Users\user\esp\esp-idf\tools\idf_tools.py", line 1107, in action_install
    tool_obj.download(tool_version)
  File "C:\Users\user\esp\esp-idf\tools\idf_tools.py", line 541, in download
    urlretrieve(url, local_temp_path, report_progress if not global_non_interactive else None)
  File "D:\Anaconda\lib\urllib\request.py", line 247, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "D:\Anaconda\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "D:\Anaconda\lib\urllib\request.py", line 525, in open
    response = self._open(req, data)
  File "D:\Anaconda\lib\urllib\request.py", line 548, in _open
    'unknown_open', req)
  File "D:\Anaconda\lib\urllib\request.py", line 503, in _call_chain
    result = func(*args)
  File "D:\Anaconda\lib\urllib\request.py", line 1387, in unknown_open
    raise URLError('unknown url type: %s' % type)
urllib.error.URLError: <urlopen error unknown url type: https>

I tried to run using the setup on windows and tried to install it manually. It seems that the link to download idf-exe is broken. Thus, the error occurs.

Any suggestion or solution regarding to what should I do to install the ESP-IDF tools? Thank you

YesayaS, 2019-09-26:
I've got the same issues. Any help will be appreciated!

Alvin1Zhang, 2019-09-26:
@AldyKwee Thanks for reporting, we have noted this issue, and is now fixing. Will update once it is fixed, thanks.

Alvin1Zhang, 2019-09-26:
@AldyKwee We have fixed the issue, would you please try again? Thanks.

AldyKwee, 2019-09-26:
@Alvin1Zhang Thank you for responding. Unfortunately, I've tried to re-run the install.bat and it still has the same error (urllib error). I've tried to re-clone the repository and run it again but it still give the same error. Did I do something wrong?

AldyKwee, 2019-09-27:
@Alvin1Zhang Sorry! Now it works. I've set the wrong python on my environment variables. (must python2 not python3). Thanks for helping!

closed 2019-09-27 by AldyKwee

Alvin1Zhang, 2019-09-27:
@AldyKwee Thanks.

mcmanus-git, 2020-01-11:
I'm getting a somewhat similar error when trying to run a script through a .bat file although it works when I run through my IDE.  Below is the error thrown in the cmd line.

    page = urllib.request.urlopen(url)
  File "C:\ProgramData\Anaconda3\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "C:\ProgramData\Anaconda3\lib\urllib\request.py", line 525, in open
    response = self._open(req, data)
  File "C:\ProgramData\Anaconda3\lib\urllib\request.py", line 548, in _open
    'unknown_open', req)
  File "C:\ProgramData\Anaconda3\lib\urllib\request.py", line 503, in _call_chain
    result = func(*args)
  File "C:\ProgramData\Anaconda3\lib\urllib\request.py", line 1387, in unknown_open
    raise URLError('unknown url type: %s' % type)
urllib.error.URLError: <urlopen error unknown url type: https>

brenaud75, 2020-03-18:
> I'm getting a somewhat similar error when trying to run a script through a .bat file although it works when I run through my IDE. Below is the error thrown in the cmd line.
>
> ```
> page = [...]

Exact same problem. Did you find a solution ?

igrr, 2020-03-19:
It seems that the version of Python you have installed doesn't support HTTPS protocol in urllib. You can try to go through the installation process again, this time choosing the option to install Python 3.7? This will install a new version of Python that is known to work. Thanks.

brenaud75, 2020-03-19:
> It seems that the version of Python you have installed doesn't support HTTPS protocol in urllib. You can try to go through the installation process again, this time choosing the option to install Python 3.7? This will install a new version of Python that is known to work. Thanks.

Problem resolved by moving 2 dlls in the right directory ->  https://stackoverflow.com/a/60405693