espressif/esp-idf #4119: How Do I find Toolchain Setup ?

MA92, 2019-09-26:
I followed https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup.html for linux but in the step Toolchain Setup ( tar -xzf ~/Downloads/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz),  it give me:  Error is not recoverable:exiting now.
Also I tried  https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html  but in step 3 (cd ~/esp/esp-idf  ./install.sh)  it does not continue with the process.

AldyKwee, 2019-09-26:
what did it output? I got the same issues in windows, when I do install.bat it outputs urllib error. It seems that the link for idf-exe is gone. 

I have raised the issue here: https://github.com/espressif/esp-idf/issues/4118

Alvin1Zhang, 2019-09-26:
@MA92 @AldyKwee Thanks for reporting, we have noted this issue, and is now fixing. Will update once it is fixed, thanks.

Alvin1Zhang, 2019-09-26:
@MA92 @AldyKwee We have fixed the issue, would you please try again? Thanks.

Mario524, 2019-09-26:
it´s working! thanks .

Alvin1Zhang, 2019-09-27:
@Mario524 Thanks for the notes. @MA92 Would you please help confirm whether your issue has been resolved? Thanks.

MA92, 2019-09-27:
Yeah, the issue has been  resolved. Thanks

Alvin1Zhang, 2019-09-27:
@MA92 Thanks for the confirmation, feel free to reopen the issue if it pops up again.

closed 2019-09-27 by Alvin1Zhang