espressif/esp-idf #4121: idf.py menuconfig: failed to run confgen.py

howroyd, 2019-09-26:
Getting the following when trying to run menuconfig.  Have successfully run it in the past.  On v3.3.

If you need any more info, please let me know.

TIA

```
PS C:\Users\howro\Documents\Code\a2dp_source> python C:\Users\howro\.platformio\packages\framework-espidf\tools\idf.py menuconfig     
Checking Python dependencies...
Python requirements from C:\Users\howro\.platformio\packages\framework-espidf\requirements.txt are satisfied.
Running cmake in directory C:\Users\howro\Documents\Code\a2dp_source\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized C:\Users\howro\Documents\Code\a2dp_source"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.23.0.windows.1") 
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/howro/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/howro/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/howro/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/howro/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/howro/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap freertos vfs newlib esp_ringbuf driver esp_event ethernet mbedtls micro-ecc efuse bootloader_support partition_table app_update spi_flash nvs_flash lwip tcpip_adapter pthread smartconfig_ack wpa_supplicant xtensa-debug-module espcoredump esp32 cxx app_trace asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning
-- Component paths: C:/Users/howro/.platformio/packages/framework-espidf/components/soc;C:/Users/howro/.platformio/packages/framework-espidf/components/log;C:/Users/howro/.platformio/packages/framework-espidf/components/heap;C:/Users/howro/.platformio/packages/framework-espidf/components/freertos;C:/Users/howro/.platformio/packages/framework-espidf/components/vfs;C:/Users/howro/.platformio/packages/framework-espidf/components/newlib;C:/Users/howro/.platformio/packages/framework-espidf/components/esp_ringbuf;C:/Users/howro/.platformio/packages/framework-espidf/components/driver;C:/Users/howro/.platformio/packages/framework-espidf/components/esp_event;C:/Users/howro/.platformio/packages/framework-espidf/components/ethernet;C:/Users/howro/.platformio/packages/framework-espidf/components/mbedtls;C:/Users/howro/.platformio/packages/framework-espidf/components/micro-ecc;C:/Users/howro/.platformio/packages/framework-espidf/components/efuse;C:/Users/howro/.platformio/packages/framework-espidf/components/bootloader_support;C:/Users/howro/.platformio/packages/framework-espidf/components/partition_table;C:/Users/howro/.platformio/packages/framework-espidf/components/app_update;C:/Users/howro/.platformio/packages/framework-espidf/components/spi_flash;C:/Users/howro/.platformio/packages/framework-espidf/components/nvs_flash;C:/Users/howro/.platformio/packages/framework-espidf/components/lwip;C:/Users/howro/.platformio/packages/framework-espidf/components/tcpip_adapter;C:/Users/howro/.platformio/packages/framework-espidf/components/pthread;C:/Users/howro/.platformio/packages/framework-espidf/components/smartconfig_ack;C:/Users/howro/.platformio/packages/framework-espidf/components/wpa_supplicant;C:/Users/howro/.platformio/packages/framework-espidf/components/xtensa-debug-module;C:/Users/howro/.platformio/packages/framework-espidf/components/espcoredump;C:/Users/howro/.platformio/packages/framework-espidf/components/esp32;C:/Users/howro/.platformio/packages/framework-espidf/components/cxx;C:/Users/howro/.platformio/packages/framework-espidf/components/app_trace;C:/Users/howro/.platformio/packages/framework-espidf/components/asio;C:/Users/howro/.platformio/packages/framework-espidf/components/jsmn;C:/Users/howro/.platformio/packages/framework-espidf/components/aws_iot;C:/Users/howro/.platformio/packages/framework-espidf/components/bootloader;C:/Users/howro/.platformio/packages/framework-espidf/components/bt;C:/Users/howro/.platformio/packages/framework-espidf/components/coap;C:/Users/howro/.platformio/packages/framework-espidf/components/console;C:/Users/howro/.platformio/packages/framework-espidf/components/nghttp;C:/Users/howro/.platformio/packages/framework-espidf/components/esp-tls;C:/Users/howro/.platformio/packages/framework-espidf/components/esp_adc_cal;C:/Users/howro/.platformio/packages/framework-espidf/components/tcp_transport;C:/Users/howro/.platformio/packages/framework-espidf/components/esp_http_client;C:/Users/howro/.platformio/packages/framework-espidf/components/esp_http_server;C:/Users/howro/.platformio/packages/framework-espidf/components/esp_https_ota;C:/Users/howro/.platformio/packages/framework-espidf/components/openssl;C:/Users/howro/.platformio/packages/framework-espidf/components/esp_https_server;C:/Users/howro/.platformio/packages/framework-espidf/components/esptool_py;C:/Users/howro/.platformio/packages/framework-espidf/components/expat;C:/Users/howro/.platformio/packages/framework-espidf/components/wear_levelling;C:/Users/howro/.platformio/packages/framework-espidf/components/sdmmc;C:/Users/howro/.platformio/packages/framework-espidf/components/fatfs;C:/Users/howro/.platformio/packages/framework-espidf/components/freemodbus;C:/Users/howro/.platformio/packages/framework-espidf/components/idf_test;C:/Users/howro/.platformio/packages/framework-espidf/components/json;C:/Users/howro/.platformio/packages/framework-espidf/components/libsodium;C:/Users/howro/.platformio/packages/framework-espidf/components/mdns;C:/Users/howro/.platformio/packages/framework-espidf/components/mqtt;C:/Users/howro/.platformio/packages/framework-espidf/components/protobuf-c;C:/Users/howro/.platformio/packages/framework-espidf/components/protocomm;C:/Users/howro/.platformio/packages/framework-espidf/components/spiffs;C:/Users/howro/.platformio/packages/framework-espidf/components/ulp;C:/Users/howro/.platformio/packages/framework-espidf/components/unity;C:/Users/howro/.platformio/packages/framework-espidf/components/wifi_provisioning
Traceback (most recent call last):
  File "C:/Users/howro/.platformio/packages/framework-espidf/tools/kconfig_new/confgen.py", line 333, in <module>
    main()
  File "C:/Users/howro/.platformio/packages/framework-espidf/tools/kconfig_new/confgen.py", line 84, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "C:\Users\howro\.platformio\packages\framework-espidf\tools\kconfig_new\kconfiglib.py", line 656, in __init__
    self.top_node)  # prev_node
  File "C:\Users\howro\.platformio\packages\framework-espidf\tools\kconfig_new\kconfiglib.py", line 1583, in _parse_block
    self._enter_file(sourced_file)
  File "C:\Users\howro\.platformio\packages\framework-espidf\tools\kconfig_new\kconfiglib.py", line 1167, in _enter_file
    .format(self._filename, self._linenr, e.message))
IOError: C:/Users/howro/.platformio/packages/framework-espidf/Kconfig:59: Could not open ' C:/Users/howro/.platformio/packages/framework-espidf/components/app_update/Kconfig.projbuild C:/Users/howro/.platformio/packages/framework-espidf/components/bootloader/Kconfig.projbuild C:/Users/howro/.platformio/packages/framework-espidf/components/esptool_py/Kconfig.projbuild C:/Users/howro/.platformio/packages/framework-espidf/components/partition_table/Kconfig.projbuild' (ENOENT: No such file or directory) [... cut ...]
  C:/Users/howro/.platformio/packages/framework-espidf/components/partition_table/Kconfig.projbuild;--env;IDF_CMAKE=y).
  Error 1
Call Stack (most recent call first):
  C:/Users/howro/.platformio/packages/framework-espidf/CMakeLists.txt:68 (kconfig_process_config)

-- Configuring incomplete, errors occurred!
See also "C:/Users/howro/Documents/Code/a2dp_source/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/howro/Documents/Code/a2dp_source/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```

dobairoland, 2019-09-26:
Hi @howroyd. Do you see `kconfiglib` in the output when you run `python -m pip freeze`?

What IDF version is you using? All branches (releases) contain now the fix. 

If you cannot update for some reason and you have `kconfiglib` installed then you could try to remove it: `python -m pip uninstall kconfiglib`

howroyd, 2019-09-26:
I'm on v3.3.

Output from the python command:
```C:\Users\howro>C:\Users\howro\.platformio\python27\python.exe -m pip freeze
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
asn1crypto==0.24.0
cffi==1.12.3
cryptography==2.7
enum34==1.1.6
future==0.17.1
ipaddress==1.0.22
pycparser==2.19
pyparsing==2.3.1
pyserial==3.4
six==1.12.0
virtualenv==16.6.0```

howroyd, 2019-09-26:
So I appear to have figured this out now, so will close the issue.  Thanks for the help @dobairoland .

It turns out thet somehow platformio had overwritten the esp-idf which I pulled from github.

I deleted the contents of the `.platformio\packages\framework-espidf` directory and replaced it with the contents of github v3.3.  All works now.
Cheers

closed 2019-09-26 by howroyd