espressif/esp-idf #4124: Secure bootloader + Flash Encryption in place not working

ManishMehra001, 2019-09-26:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
After enabling Secure boot (Reflashable) + Flash Encryption (In Place) unable to boot successfully. 
//Detailed problem description goes here.

### Expected Behavior
Board should successfully boot

### Actual Behavior
Unable to boot successfully

### Steps to repropduce
1. Enable Secure Boot And Flash Encryption

Select "Security features"
Enable for "Enable secure boot in bootloader"
Select "Secure bootloader mode", select "Reflashable".
Enable "Sign binaries during build"
On "Secure boot privite signing key", input your file name, here is "my_secure_boot_signing_key.pem"
Enable "Allow potentially insecure options"
Select "Potentially insecure option",
Select "Leave ROM BASIC Interpreter available on reset"
Select "Allow JTAG Debugging"
Select "Allow app partition length not 64KB aligned"
Select "Leave UART bootloader encryption enabled"
Select "Leave UART bootloader decryption enabled"
Select "Leave UART bootloader flash cache enabled"
Select "Bootloader config"
Select "Bootloader log verbosity"
Select "No output"

Select "Enable flash encryption on boot (READ DOCS FIRST) - Enable usage mode (Development(NOT SECURE))"
Save and Exit menu config

2. make bootloader

3. Enable Secure boot on board and Burn secure bootloader key

        3.1 Enable Secure boot on board by burn ABS_DONE_0 to 1,
        python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0

        3.2 Burn secure bootloader key (secure-bootloader-key-256.bin),
        python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key secure_boot ./build/bootloader/secure-bootloader-key-256.bin

4. Build and Flash following bin files at their address using esp32 flash download tool:
        
		bootloader-reflash-digest_0x0.bin at address 0x0
		
		( Note :
		I flashed bootloader-reflash-digest_0x0.bin at address 0x0
		
		But Do i need to flash
			bootloader.bin (at Address 0x1000)
				OR
			bootloader-reflash-digest_0x0.bin at address 0x0
		)
		
        Build Partition table and flash at address 0x8000      
        Build factory app and flash at address 0x10000
        Build main app and flash at address 0x100000
        Build ota_data_initial and flash at address 0xd000
        Build otadata_086b0e29f7c5570fb7ee208a8c7f8c0f and flash at address 0x400000

5. Reboot the H/W.

6. Getting below error :

        rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
        configsip: 0, SPIWP:0xee
        clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
        mode:DIO, clock div:2
        load:0x3fff0018,len:4
        load:0x3fff001c,len:1088       
        ho 0 tail 12 room 4
        load:0x40078000,len:16500
        load:0x40080400,len:2996
        entry 0x400805c4
        ets Jun  8 2016 00:22:57

7. Please find the attached log summary.
       

Please have a look on my above steps and if anything is missing here please suggest me.
It would be great help of yours if finally we can work with this approach as we are planning to go this for our production purpose
where we dont need to create flash encryption key.

If I understand correctly, for flash encryption in place, 

- I no need to burn FLASH_CRYPT_CNT and FLASH_CRYPT_CONFIG. Am i right here ?

- And also please suggest, do I need to adjust the flash addresses of my binaries at step 4 and if yes I need to adjust the address then what address values I should use for each of binary at step 4.
  Actually last time when i did successfully flash encryption with pre-generated key procedure my bootloader size was 25KB after encryption with
  pre-generated flash encryption key, so do i need to increase partition_table offset address greater than 0x8000 ?
  
Thanks & Best Regards,
Manish

[image]

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

ManishMehra001, 2019-09-27:
Next I tried with following steps :
1. Erasing flash with ESP32 flash download tool
2. 
 - flashing bootloader.bin (at Address 0x1000)
 - Build Partition table and flash at address 0x8000      
 - Build factory app and flash at address 0x10000
 - Build main app and flash at address 0x100000
 - Build ota_data_initial and flash at address 0xd000
 - Build otadata_086b0e29f7c5570fb7ee208a8c7f8c0f and flash at address 0x400000

Then i am getting below error:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1088
ho 0 tail 12 room 4
load:0x40078000,len:16500
load:0x40080400,len:2996
secure boot check fail
ets_main.c 371
ets Jun  8 2016 00:22:57

And below is the snapshot efuse summary:
[image]

negativekelvin, 2019-09-27:
```
Select "Bootloader config"
Select "Bootloader log verbosity"
Select "No output"
```

When you flash bootloader-reflash-digest_0x0.bin it works, but bootloader exits.  There is no log.  Try turning logs back to info.

ManishMehra001, 2019-09-27:
Hi, 
This time I have enabled - Error in Bootloader configuration and rebuild the bootloader again,
I flashed as using ESP32 flash download tool (3.6.5) 
i. bootloader-reflash-digest_0x0.bin at address 0x0
And other plain text files at their respective address

After boot I see following errors 

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5824
ho 0 tail 12 room 4
load:0x40078000,len:18308
load:0x40080400,len:3204
entry 0x40080604
E (352) esp_image: Image length 789440 doesn't fit in partition length 786432
E (352) boot: Factory app partition is not bootable
E (1029) flash_encrypt: flash encryption is not enabled, and SECURE_FLASH_REQUIRE_ALREADY_ENABLED is set, refusing to boot.
E (1030) boot: Flash encryption check failed (259).
E (1035) boot: OTA app partition slot 0 is not bootable
E (1041) esp_image: image at 0x280000 has invalid magic byte
E (1048) boot: OTA app partition slot 1 is not bootable
E (1054) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5824
ho 0 tail 12 room 4
load:0x40078000,len:18308
load:0x40080400,len:3204
entry 0x40080604

Attaching here my partitions_spiffs.csv file snapshot iamge,
[image]

Can you please suggest me what addresses I need to adjust here with what values and how to edit my 
partitions_spiffs.csv file, because i am not able to edit it with microsoft excel.

Thanks & Best Regards,
Manish

ManishMehra001, 2019-09-27:
Is there any way to send you my partitions_spiffs.csv file to you. Please suggest me.

negativekelvin, 2019-09-27:
Edit with a text editor.  So you have 2 problems, factory partition too small and SECURE_FLASH_REQUIRE_ALREADY_ENABLED set to yes.

ManishMehra001, 2019-09-27:
Thanks for your input negativekelvin, So, 

1. Edit with a text editor. So you have 2 problems, factory partition too small and
My Action: I need to increase the factory partitions and accordingly adjust others also in the partition table

2. For this "SECURE_FLASH_REQUIRE_ALREADY_ENABLED set to yes" :
 My Action : What exactly I need to do here and how to do ?

ManishMehra001, 2019-09-27:
My factory app is 
factory | app | factory | 0x10000 | 0xC0000

so to increase factory app size shall i shift 4096 byes forward for spiffs which is currently, like 
user_conf | data | spiffs | 0xD0000 | 0x30000 

And down the line for the rest of partitions also shift 4096 byte..
Am i right here ? want to confirm before got trouble into another issue :)

negativekelvin, 2019-09-27:
> App partitions have to be at offsets aligned to 0x10000 (64K).
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/partition-tables.html

KonstantinKondrashov, 2019-09-27:
FYI, You can leave the offsets empty. Partition table tool fill them.

See https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/partition-tables.html#offset-size 
> If you want the partitions in the partition table to work with any starting offset (CONFIG_PARTITION_TABLE_OFFSET) of the table itself, leave the offset field (in CSV file) for all partitions blank. Similarly, if changing the partition table offset then be aware that all blank partition offsets may change to match, and that any fixed offsets may now collide with the partition table (causing an error).

As example - https://github.com/espressif/esp-idf/blob/master/components/partition_table/partitions_two_ota.csv
Use the `make partition_table` command to see which offsets are there.

ManishMehra001, 2019-09-27:
Currently i change like shifted all the partitions 4KB space from the Factory app partitions, image below,
[image]

ManishMehra001, 2019-09-27:
Now I am trying with this new updated partition table,

ManishMehra001, 2019-09-27:
Thanks negativekelvin  & KonstantinKondrashov.

Finally it works for me 👍 
Thanks a lot :)

ManishMehra001, 2019-09-28:
Hi negativekelvin,
Now I tried for Ota update after booting up board successfully,

I created my ota file (as I done before while tested successfully ota with approach of generated key for flash encryption )

- ./mkspiffs -c assets -b 4096 -p 256 -s 0x200000 otadata
- ../script/catfile.py panl35L_app 0x180000 otadata 0x200000 otaall

When I started ota this time it is failing,
Do I need to set any efuse ?

ManishMehra001, 2019-09-30:
Now, the last step, I want to go ahead for Production release.

So I have following queries :
**_1. Can i use the same module (where i have previous used configuration for Reflshable + Development mode) for following changes required for Release mode_**
  
**1. Enable Secure Boot And Flash Encryption**
Select "Security features"
Enable for "Enable secure boot in bootloader"
Select "Secure bootloader mode", select "**One-time flash**".
Enable "Sign binaries during build"
On "Secure boot privite signing key", input your file name, here is "my_secure_boot_signing_key.pem"

Enable "Allow potentially insecure options"
Select "Potentially insecure option",
Select "Leave ROM BASIC Interpreter available on reset"
Select "Allow JTAG Debugging"
Select "Allow app partition length not 64KB aligned"
Select "Leave UART bootloader encryption enabled"
Select "Leave UART bootloader decryption enabled"
Select "Leave UART bootloader flash cache enabled"
Un-Select "Require flash encryption to be already enabled"
Select "Bootloader config"
Select "Bootloader log verbosity"
Select "No output"

Select "Enable flash encryption on boot (READ DOCS FIRST) - Enable usage mode (**Release**)"
Save and Exit menu config

2. **make bootloader**

3. **Enable Secure boot on board**

 3.1 **Enable Secure boot on board by burn ABS_DONE_0 to 1,**
 python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0

4. **Build and Flash following bin files at their address using esp32 flash download tool (v3.6.5):**
    Erased flash & Flashed all my binaries on below addresses using ESP32 flash download tool:
    flashing bootloader-reflash-digest_0x0.bin at address 0x0
    Build Partition table and flash at address 0x8000
    Build factory app and flash at address 0x10000
    Build main app and flash at address 0x110000
    Build ota_data_initial and flash at address 0xd000
    Build otadata_086b0e29f7c5570fb7ee208a8c7f8c0f and flash at address 0x410000

5**. Reboot the H/W.**

negativekelvin, 2019-09-30:
No not same module. Why Allow potentially insecure options? No to 3 and 3.1.

ManishMehra001, 2019-09-30:
Thanks negativekelvin.

So I will re-correct my steps as below :

**No not same module.**
Action : Understood and took new module.

**Why Allow potentially insecure options?** 
Action : Disable - potentially insecure options.

**No to 3 and 3.1.** 
Action : No need to explicitly enable Secure boot on board by command
              python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0

**unsigned bootloader goes at 0x1000**

Build and Flash following bin files at their address using esp32 flash download tool (v3.6.5):
Erased flash & Flashed all my binaries on below addresses using ESP32 flash download tool:
**flash bootloader.bin at address 0x1000**
Build Partition table and flash at address 0x8000
Build factory app and flash at address 0x10000
Build main app and flash at address 0x110000
Build ota_data_initial and flash at address 0xd000
Build otadata_086b0e29f7c5570fb7ee208a8c7f8c0f and flash at address 0x410000

Please correct me if still I am wrong in above corrections.

Thanks & Best Regards,
Mansih

negativekelvin, 2019-09-30:
Yes because you still have to sign your OTA updates with this key

ManishMehra001, 2019-09-30:
Thanks negativekelvin,
 Understood this point so already did build and flashing of all those applications.
Finally I could do successfully bootup my module with Release mode.

Thanks a lot more your great help.

ManishMehra001, 2019-10-01:
Hi negativekelvin,

I have another requirement where we need to encrypt our assets also,
can you please guide how can we achieve that:
Below is our partition table :

[image]

negativekelvin, 2019-10-01:
Spiffs does not support encryption.

ManishMehra001, 2019-10-02:
So you mean after adding "encrypted" flag for my spiffs storage (storage, storage1), these will not be encrypted by bootloader. :( 
And it means we need to move all our storage data to nvs partition and then add "encrypted" flag for that.

negativekelvin, 2019-10-02:
https://github.com/espressif/esp-idf/blob/master/components/spiffs/esp_spiffs.c

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html#nvs-encryption

ManishMehra001, 2019-10-11:
Thanks negativekelvin.
I got another query below, 
- **Can i still flash spiffs partition multiple times using esp32 download tool even when i had enabled Secure Boot - One time flash and Flash Encryption - Release mode ..?**

Want to check here to make sure our module does not brick..!

projectgus, 2019-10-16:
Hi @ManishMehra001 ,

You can, because SPIFFS is not encrypted. Just take care not accidentally reflash the bootloader, app, or any other encrypted partitions.

Angus

closed 2020-04-30 by projectgus