espressif/esp-idf #4126: Ninja No work to do

JAICHANGPARK, 2019-09-27:
I had some issue about after flash state.

```Leaving...
Hard resetting via RTS pin...
Executing action: all (aliases: build)
Running ninja in directory c:\users\me\desktop\research\esp32\hello_world\build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Done```

Alvin1Zhang, 2019-09-27:
@JAICHANGPARK Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

negativekelvin, 2019-09-27:
What command did you run exactly?

kumekay, 2019-09-30:
Hi @JAICHANGPARK 

Could you please write exact command you execute?

It looks like a possible duplication of https://github.com/espressif/esp-idf/issues/3948 That was already fixed in 44c89c0e9f846690c9d499d665ae0b9fa5c93514 but not yet published on github. It will be published soon after passing QA.

igrr, 2019-10-01:
Has been merged in https://github.com/espressif/esp-idf/commit/44c89c0e9f846690c9d499d665ae0b9fa5c93514. Please update IDF and reopen this if the issue persists.

closed 2019-10-01 by igrr

JAICHANGPARK, 2019-10-01:
@igrr @kumekay  Thanks to help 
I solved this problem !!