espressif/esp-idf #4128: Tools installation fails if user name includes space, due incorrect arg passing to 7za.exe

kiyrik, 2019-09-27:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
     v3.2-dev-1148-g96cd3b75c
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     1.22.0-80-g6c4433a
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

esp-idf-tools-setup-2.0.exe fails if Username includes spaces. Seems quotes are missed for path when running 7za.exe:

Running command: C:\Users\KIRILB~1\AppData\Local\Temp\is-J02MV.tmp\7za.exe x -oC:\Users\KIRILB~1\AppData\Local\Temp\is-J02MV.tmp -r -aoa C:\Users\Kiril Borodko\.espressif\releases\esp-idf-v3.3.zip
                          
                          7-Zip (a) 19.00 (x86) : Copyright (c) 1999-2018 Igor Pavlov : 2019-02-21
                          
                          Scanning the drive for archives:
                          
                          ERROR: The system cannot find the file specified.
                          C:\Users\Kiril

Must be:
C:\Users\KIRILB~1\AppData\Local\Temp\is-J02MV.tmp\7za.exe x -oC:\Users\KIRILB~1\AppData\Local\Temp\is-J02MV.tmp -r -aoa "C:\Users\Kiril Borodko\.espressif\releases\esp-idf-v3.3.zip"

kiyrik, 2019-09-27:
[Setup Log 2019-09-27 #002.txt](https://github.com/espressif/esp-idf/files/3662458/Setup.Log.2019-09-27.002.txt)

igrr, 2019-09-27:
This seems to be a duplicate of https://github.com/espressif/esp-idf/issues/3807, fixed in https://github.com/espressif/esp-idf/commit/aaf3dcbda01e8c22da88184922465155441cd20a. Please try the new version of the installer and see if it works for you: https://dl.espressif.com/dl/esp-idf-tools-setup-2.1.exe.

kiyrik, 2019-09-30:
I can confirm, with 2.1 extracting with 7za.exe works well. But anyway I cant install if username includes space(s). 

1. Now I have installer message, if selected default directory (user dir)
ESP-IDF build system doesnt not support spaces in path. Please choose a different directory.

2. I got an error, when manually change the path to "E:\GIT_Private\esp32\isp-idf"
In log:
     2019-09-30 08:03:56.687   Moving C:\Users\KIRILB~1\AppData\Local\Temp\is-E5Q11.tmp\esp-idf- 
      v3.3 to E:\GIT_Private\esp32\esp-idf
     2019-09-30 08:03:56.687   Message box (OK):
                          Failed to copy ESP-IDF to the destination directory: E:\GIT_Private\esp32\esp-idf
     2019-09-30 08:03:59.972   User chose OK.

[Setup Log 2019-09-30 #005.txt](https://github.com/espressif/esp-idf/files/3668947/Setup.Log.2019-09-30.005.txt)

So, seems, need create user without spaces or manually install the build system.

igrr, 2019-09-30:
It is possible to install IDF Tools into a path with spaces inside, but IDF itself needs to be installed into a path without spaces. Is there anything unusual with the permissions of E:\GIT_Private\esp32 directory? Does this directory exist before the installer is started?

kiyrik, 2019-09-30:
No any special permissions. Setup launched with Admin rights. I have tried with/without created directory - result is same.

dima0905, 2019-12-08:
I confirm that was said above. Installer delete specified directory but install in default

closed 2020-02-12 by espressif-bot

JoercT, 2022-04-03:
My idf successfully installed in the specified directory, although there was a space in the directory name. Is it possible to move the folder to another directory? If yes,  how?
OS: Windows 11 x64
ESP-IDF version: 4.4

I believe it is the reason why I can't build my blink project.