espressif/esp-idf #4129: Can't exit console by pressing Crtl+]

schrotti12, 2019-09-27:
Good evening,
I have the ESP32 connected on port COM3 on a Windows 10 1903 x64 machine and use the latest 2.0 setup of IDF. I can connect by using "idf.py -p COM3 monitor" and get the serial output, but I can not exit the monitor anymore by pressing "Crtl + ]", which worked on the old make based system. I have a German keyboard and have to press "[Strg] + [Alt Gr] + [9]" but the monitor keeps running. The same happens, if I connect by using PuTTY.
Only way I can exit is by closing the PuTTY window or pressing Crtl+C in the monitor. Then Python throws an exception.
Thanks, bye

dobairoland, 2019-09-30:
Hi @schrotti12. Are you using Windows command line or MSYS2? What is the version of IDF `git describe --tags`?

schrotti12, 2019-09-30:
Good morning,
I am using v. 3.3 and I am using it from the windows command line (cmd.exe). Previously I was working with MSYS2, where the "Crtl+]" command worked.

BlahMadal, 2019-10-02:
I have experienced the same problem too. I am using windows command line instead of msys2. I think windows command line esp-idf is not at the same level of msys esp-idf, so I will install msys version.

schrotti12, 2019-10-03:
I tried to launch a msys2 like bash from the commandline opened by clicking on "ESP-IDF Command Prompt" and tried there "idf.py -p COM3 monitor". The monitor opened and I was able to exit with the mentioned key combination. So I think it's a problem with "cmd.exe". I'll set up the environment with Msys now to continue work on my project.

closed 2019-10-13 by igrr

BB-5000, 2020-06-06:
Sorry for digging out this old issue.
I just installed the esp-idf (v4.2-dev-1660-g7d7521367) in linux and also could not successfully quit the serial monitor as recommended. I started it with `idf.py -p /dev/ttyUSB0 monitor` according to the manual. Hitting CTRL + J did nothing.
For me **however hitting CTRL + 5 did the trick** (in fact it was STRG + 5 on the German keyboard)  and I could gracefully leave the monitor again.

andreasgottardi, 2020-09-18:
Good evening, sorry for my late response but it seems to be fixed. Now it's the same key as on the American keyboard. But on the German keyboard it is the key [+] left besides the big [ENTER] key. Alternative characters are * and ~.
So basically you just have to type [Strg] + [+] (or [Crtl] + [+] in English).
Thanks for the help, I can now gracefully leave the monitor.

MarkusvonStaden, 2022-06-08:
Sorry to dig that out once again.
In the new Windows 11 Terminal and in VS Code `CTRL + [+]` just zooms in. 
I think the Command is used by the UI and not passed to IDF

dobairoland, 2022-06-08:
You can exit by pressing CTRL+T+X (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html#keyboard-shortcuts)

codebrane, 2022-06-10:
Running `v5.0-dev-2586-ga82e6e63d9` on a Raspberry pi 4 `idf.py -p /dev/ttyACM0 monitor` cannot be exited with any key combination. CTRL+], CTRL+X, CTRL+T+X, CTRL+C all do nothing other than halt the output. There doesn't appear to be a way to exit the program.

mik13ST, 2023-06-13:
And if you are using the Fish shell, you should use CTRL + ALT GR + ]. Couldn't we just have the idf.py monitor react to double pressing Ctrl + C like some other programming language shells have? That would be shell agnostic.

shufps, 2024-08-15:
> however hitting CTRL + 5 did the trick

OMG Thank you very much! :exploding_head: