espressif/esp-idf #4130: UART receiving interrupt service routine of ESP32 in eclipse IDE using CMake

surekhaibot, 2019-09-28:
Hi,
i am facing problem at UART receiving based on interrupt method in eclipse(esp_idf 3.3) with CMake, i am using below functions. but it showing uart_rx_isr is not declared. please suggest how to solve the issue
void echo_task()
{
    // Configure parameters of an UART driver, communication pins and install the driver
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
    uart_param_config(UART_NUM_1, &uart_config);
    uart_set_pin(UART_NUM_1, TXD, RXD, RTS, CTS);
    uart_driver_install(UART_NUM_1, BUF_SIZE, BUF_SIZE, 0, NULL, 0);
    uart_isr_free(UART_NUM_1);
    uart_isr_register(UART_NUM_1, uart_rx_isr, NULL, ESP_INTR_FLAG_IRAM, NULL);
    uart_enable_pattern_det_intr(UART_NUM_1, 0x55, 7, 2400, 80000, 120000);
    uart_enable_rx_intr(UART_NUM_1);
}

void IRAM_ATTR uart_rx_isr(void *para)
{
	  uint16_t rx_fifo_len, status;
	  uint16_t is = 0;

	  status = UART0.int_st.val; // read UART interrupt Status
	  rx_fifo_len = UART0.status.rxfifo_cnt; // read number of bytes in UART buffer
	  while(rx_fifo_len)
	  {
	   rxbuf[is++] = UART0.fifo.rw_byte; // read all bytes
	   rx_fifo_len--;
	  }
	  uart_clear_intr_status(UART_NUM_1,UART_AT_CMD_CHAR_DET_INT_CLR);
	 uart_write_bytes(UART_NUM_1, (const char*) rxbuf, 1);
}

negativekelvin, 2019-09-28:
https://en.wikipedia.org/wiki/Forward_declaration

igrr, 2019-10-01:
@surekhaibot as hinted by @negativekelvin, this error occurs because there is no declaration of `uart_rx_isr` function in your code ahead of its usage in `uart_isr_register` call. This is a basic C language rule.

closed 2019-10-01 by igrr