espressif/esp-idf #4132: Duplicate definition in spi_common.h

jonsmirl, 2019-09-28:
The definition of spicommon_bus_using_iomux(spi_host_device_t host); is duplicated in spi_common.h This is causing micropython tools that process this file to fail.

diff --git a/components/driver/include/driver/spi_common.h b/components/driver/include/driver/spi_common.h
index b44159546..15e91cd45 100644
--- a/components/driver/include/driver/spi_common.h
+++ b/components/driver/include/driver/spi_common.h
@@ -319,7 +319,7 @@ bool spicommon_bus_using_iomux(spi_host_device_t host);
  *
  * @return false if any pins are through the GPIO matrix, otherwise true.
  */
-bool spicommon_bus_using_iomux(spi_host_device_t host);
+//bool spicommon_bus_using_iomux(spi_host_device_t host);
 
 /**
  * @brief Get the IRQ source for a specific SPI host

ginkgm, 2019-10-08:
Hi @jonsmirl ,

We plan to make this function invisible from public domain, do you use it?

jonsmirl, 2019-10-08:
We do not use, the duplicate declaration broke a tool used to compile micropython for ESP. The tool has been fixed to all duplicate definitions now.  The tool scans the header files and converts it into Python callable objects.

closed 2019-10-13 by igrr