espressif/esp-idf #4133: All target runs after flash target, instead of before

jkent, 2019-09-28:
## Environment

- Development Kit:      N/A
- Module or chip used:  ESP32-WROVER
- IDF version:  v4.1-dev-281-g96b96ae24
- Build System:        CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Power Supply:         Battery

## Problem Description

Using `idf.py flash` with no build directory fails to build and flash.

Running `idf.py build` and then `idf.py flash` works, however, the all target runs after flashing is complete.

This occurs with both GNU make and ninja build systems.

### Expected Behavior

All target should run before flash with `idf.py flash`

### Actual Behavior

All target is appended to the list of actions, running last.

### Steps to repropduce

```
jkent@pav:~/Projects/esp32$ . ${IDF_PATH}/export.sh
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Python requirements from /home/jkent/usr/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/jkent/usr/esp-idf/components/esptool_py/esptool
  /home/jkent/usr/esp-idf/components/espcoredump
  /home/jkent/usr/esp-idf/components/partition_table/
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

jkent@pav:~/Projects/esp32$ cp -r ${IDF_PATH}/examples/get-started/hello_world .
jkent@pav:~/Projects/esp32$ cd hello_world
jkent@pav:~/Projects/esp32/hello_world$ idf.py flash
Adding flash's dependency "all" to list of actions
Executing action: flash
Choosing default port b'/dev/ttyUSB0' (use '-p PORT' option to set a specific serial port)
Traceback (most recent call last):
  File "/home/jkent/usr/esp-idf/tools/idf.py", line 1520, in <module>
    main()
  File "/home/jkent/usr/esp-idf/tools/idf.py", line 1435, in main
    cli(prog_name=PROG)
  File "/home/jkent/.espressif/python_env/idf4.1_py3.7_env/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/jkent/.espressif/python_env/idf4.1_py3.7_env/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/jkent/.espressif/python_env/idf4.1_py3.7_env/lib/python3.7/site-packages/click/core.py", line 1164, in invoke
    return _process_result(rv)
  File "/home/jkent/.espressif/python_env/idf4.1_py3.7_env/lib/python3.7/site-packages/click/core.py", line 1102, in _process_result
    **ctx.params)
  File "/home/jkent/.espressif/python_env/idf4.1_py3.7_env/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/jkent/usr/esp-idf/tools/idf.py", line 1067, in execute_tasks
    task.run(ctx, global_args, task.action_args)
  File "/home/jkent/usr/esp-idf/tools/idf.py", line 685, in run
    self.callback(self.name, context, global_args, **action_args)
  File "/home/jkent/usr/esp-idf/tools/idf.py", line 362, in flash
    esptool_args = _get_esptool_args(args)
  File "/home/jkent/usr/esp-idf/tools/idf.py", line 340, in _get_esptool_args
    with open(os.path.join(args.build_dir, "flasher_args.json")) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/jkent/Projects/esp32/hello_world/build/flasher_args.json'
```
Also, after having built the target using idf.py build:

```
jkent@pav:~/Projects/esp32/hello_world$ idf.py flash
Adding flash's dependency "all" to list of actions
Executing action: flash
Choosing default port b'/dev/ttyUSB0' (use '-p PORT' option to set a specific serial port)
Running esptool.py in directory /home/jkent/Projects/esp32/hello_world/build
Executing "/home/jkent/.espressif/python_env/idf4.1_py3.7_env/bin/python /home/jkent/usr/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --after hard_reset write_flash @flash_project_args"...
esptool.py -p /dev/ttyUSB0 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v2.8-dev
Serial port /dev/ttyUSB0
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme 3/4
Crystal is 40MHz
MAC: b4:e6:2d:95:df:f9
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 82...
Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 6754.4 kbit/s)...
Hash of data verified.
Compressed 24768 bytes to 14737...
Wrote 24768 bytes (14737 compressed) at 0x00001000 in 0.3 seconds (effective 567.5 kbit/s)...
Hash of data verified.
Compressed 147120 bytes to 76537...
Wrote 147120 bytes (76537 compressed) at 0x00010000 in 1.9 seconds (effective 612.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: all (aliases: build)
Running ninja in directory /home/jkent/Projects/esp32/hello_world/build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Done
```

Notice that `ninja all` runs after the flash.  This is a problem for a more complex project that runs some build steps unconditionally.

negativekelvin, 2019-09-28:
https://github.com/espressif/esp-idf/issues/4126

kumekay, 2019-09-30:
Hi @jkent 
Thank you for the report. 
It looks like a duplication of https://github.com/espressif/esp-idf/issues/3948 

https://github.com/espressif/esp-idf/issues/3948  was already fixed in 44c89c0e9f846690c9d499d665ae0b9fa5c93514 but not yet published on github. It will be published soon after passing QA.

dobairoland, 2019-10-01:
https://github.com/espressif/esp-idf/commit/44c89c0e9f846690c9d499d665ae0b9fa5c93514 closes this issue.

@jkent please checkout the latest `master` branch for the solution. If you still encounter the problem then feel free to reopen this issue or create a new one. Thanks for your report.

closed 2019-10-01 by dobairoland