espressif/esp-idf #4134: Not Getting IP address from ESP32 wrover?

vivekalways, 2019-09-30:


vivekalways, 2019-09-30:
i am using my custom ESP32 wrover with Lan8720A .Currently LAN8720 ic is intialised but i am not getting ip address.
ESP GPIOs  which i connected below:
TXD0 = GPIO19
TXD1 = GPIO22
TX_EN = GPIO21
 RXD0 = GPIO25
 RXD1 = GPIO26
 CLK == GPIO0
Here is my schematic
[image]

vivekalways, 2019-10-03:
Anybody is there?

Alvin1Zhang, 2019-10-08:
@vivekalways Thanks for reporting and sorry for the late reply. We have celebrated China's National Holiday. Thanks.

suda-morris, 2019-10-08:
Hi @vivekalways 
In your case:

> LAN8720 ic is intialised

That means the control panel works, esp32 can read PHY status with SMI interface

>  i am not getting ip address

Acquiring IP address is relevant to data panel, data communication between between ESP32 and PHY may have something wrong. 
1. Please check if GPIO0 got the external RMII clock.
2. If so, what's your sdkconfig?

I would like to suggest you post this question to our [forum](https://www.esp32.com) with a clearer picture so that you will get more answers from there.

vivekalways, 2019-10-08:
> Hi @vivekalways
> In your case:
>
>
> That means the control panel works, esp32 can read PHY status with SMI interface
>
>
> Acquiring IP address is relevant to data panel, data communication between between [...]

suda-morris, 2019-10-08:
@vivekalways 
Configuration seems OK with me.

If the log prints any "link up" or "link down" info? (it would be better if you can enable the DEBUG log, so that we can see what's happening in the background). `Component config > Log output > Default log verbosity > Debug`

 And please make sure, there's at least one DHCP server in your LAN.

vivekalways, 2019-10-09:
@suda-morris 
1) where should i check link up information?
2) Checked DHCP server.enabled in my network.

suda-morris, 2019-10-10:
Link up and Link down event will trigger logs in your monitor.

vivekalways, 2019-10-14:
@all .. its works fine.The problem is clock line.Place clock line nearest to esp32 from LAN chip. Dont use wire for testing or debugging.
conclude ESP32 wrover ethernet is working with external by using GPIO0 pin ...

closed 2019-10-14 by vivekalways