espressif/esp-idf #4137: Option "BT/BLE will first malloc the memory from the PSRAM" will result in corrupted data.

MatthiasJentsch, 2019-09-30:
## Environment

- Development Kit:      none
- Kit version: none
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.1-dev-369-g4dac7c7df
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Windows
- Power Supply:         external 3.3V

## Problem Description

If I enable the option "BT/BLE will first malloc the memory from the PSRAM" / "CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST" some bytes that are received via Bluetooth SPP are set to zero. With option disabled all data is received correctly.

### Expected Behavior

The data that is received shouldn't change if I activate the option.

### Actual Behavior

I'm sending 900kb data (an over the air update) over Bluetooth classic serial port profile (SPP). I'm recording the data that I put into the bluetoothSocket output stream on the android device and I'm recording all the bytes that I receive via read function on the ESP32. I'm receiving the correct number of bytes regardless the option "BT/BLE will first malloc the memory from the PSRAM" is enabled or disabled. But if the option is enabled some bytes that are not zero are zero is the received data. The total number of bytes that are zero but are send as non zero varies from 8 to 20. And there are no more than three bytes in a row set to zero. There are only one, two or three bytes in a row set to zero. If I disable the option all bytes are received correctly.

I think it can have something to do with the PSRAM that is only addressable as 32bit memory. The internal memory is addressable as 8bit memory.

### Steps to repropduce

1. Use the example: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_vfs_acceptor
2. Send a large file (>500kb) and log/record the data that is received
3. Compare the data received byte by byte and see that some bytes are received as zero but are not send as zero.

## Other items if possible

- [x] sdkconfig file: [sdkconfig.zip](https://github.com/espressif/esp-idf/files/3672927/sdkconfig.zip)
- example of bytes that differ. I've compared with "fc" on Windows:
Address: Expected Actual
0002E3FA: 07 00
0002E3FB: 8C 00
0002E3FC: 53 00
0002FF0C: 72 00
0002FF0D: 05 00
0002FF0E: 04 00
0008C450: 8A 00
0008C451: C1 00
0008C452: A0 00
000962DE: 52 00
000962DF: A1 00
000962E0: E7 00
000D63AD: 58 00
000D63AE: 02 00

jeremyherbert, 2019-10-03:
I have this exact same problem with BLE, sending writes quickly (also for OTA). It's always 3 bytes set to zero somewhere in my writes. I did have the PSRAM option enabled, and disabling it has fixed my problem.

jeremyherbert, 2019-10-03:
I have also sniffed the packets out of the air using a different vendors tools, and the packets are transmitted correctly and received correctly by the sniffer (ie there are no spurious zeros)

MatthiasJentsch, 2019-10-03:
Thanks for confirming that bug. Maybe someone with the right knowledge can fix this.

enricop, 2019-10-17:
#2892

neoniousTR, 2019-10-18:
If it is the PSRAM bug, working without PSRAM OR in unicore mode fixes the problem.

xiongweichao, 2022-04-22:
Hi, @MatthiasJentsch 

Sorry, I didn't reproduce this issue. Please update to the latest release/v4.1 to see if the problem still exists.

Thanks

jack0c, 2023-04-12:
Will close this since we have https://github.com/espressif/esp-idf/issues/2892

closed 2023-04-12 by jack0c