espressif/esp-idf #4140: Modbus_master and slave example is not working

Kasuhikdas, 2019-10-01:
Error from modbus_master 

E (458333) SENSE_MAIN: Update failed for cid: 4, Temperature_2(�C) = 0, ESP_ERR_INVALID_STATE
E (458343) MB_CONTROLLER_MASTER: mbc_master_get_parameter(103): Master interface is not correctly initialized.
E (458353) SENSE_MAIN: Update failed for cid: 5, RelayP1(on/off) = 0, ESP_ERR_INVALID_STATE
E (459363) MB_CONTROLLER_MASTER: mbc_master_get_parameter(103): Master interface is not correctly initialized.
E (459363) SENSE_MAIN: Update failed for cid: 0, Data_channel_0(Volts) = 0, ESP_ERR_INVALID_STATE
E (459373) MB_CONTROLLER_MASTER: mbc_master_get_parameter(103): Master interface is not correctly initialized.
E (459383) SENSE_MAIN: Update failed for cid: 1, Humidity_1(%rH) = 0, ESP_ERR_INVALID_STATE
E (459393) MB_CONTROLLER_MASTER: mbc_master_get_parameter(103): Master interface is not correctly initialized.
E (459403) SENSE_MAIN: Update failed for cid: 2, Temperature_1(�C) = 0, ESP_ERR_INVALID_STATE
E (459413) MB_CONTROLLER_MASTER: mbc_master_get_parameter(103): Master interface is not correctly initialized.
E (459423) SENSE_MAIN: Update failed for cid: 3, Humidity_2(%rH) = 0, ESP_ERR_INVALID_STATE

Error from modbus_slave
�������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������

i have tried changing Go to CONFIG_ESP32_XTAL_FREQ_SEL to 26MHz nothing changed.

alisitsyn, 2019-10-01:
Hi Kasuhikdas,

Thanks for reporting the issue. Would you please help provide more details as suggested in the issue template? Information like elf, sdk configuration, backtrace, commit ID (ESP_IDF version), hardware and etc. would help me to reproduce this issue. 

Thank you.

--
Alex

Kasuhikdas, 2019-10-03:
Hi Alex,
Providing the required template parameters
-Commit ID: 4dac7c7df885adaa86a5c79f2adeaf8d68667349
Development Kit: [ESP32-Wrover-Kit]
Core (if using chip or module): ESP-WROOM32
- IDF version: v4.1-dev-369-g4dac7c7df
- Build System:         CMake
- Compiler version :
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0

- Operating System:     [Linux]
- Power Supply:         [USB]

alisitsyn, 2019-10-03:
Hi Kasuhikdas,

Thank you for your information. I was able to reproduce it. It is known memory corruption bag in example which came during rebase and it is already fixed but not merged yet. It will be implemented in ESP-IDF v4.1.
Could you please apply the simple patch below to fix this issue:
diff --git a/examples/protocols/modbus_master/main/sense_modbus.c b/examples/protocols/modbus_master/main/sense_modbus.c
index 8d1ab3539..fdc28a6a8 100644
--- a/examples/protocols/modbus_master/main/sense_modbus.c
+++ b/examples/protocols/modbus_master/main/sense_modbus.c
@@ -59,23 +59,23 @@ static void* sense_modbus_get_param_data(const mb_parameter_descriptor_t* param_
        switch(param_descriptor->mb_param_type)
        {
            case MB_PARAM_HOLDING:
-               instance_ptr = (void*)(&holding_reg_params + param_descriptor->param_offset - 1);
+               instance_ptr = (void*)((uint32_t)&holding_reg_params + param_descriptor->param_offset - 1);
                break;
            case MB_PARAM_INPUT:
-               instance_ptr = (void*)(&input_reg_params + param_descriptor->param_offset - 1);
+               instance_ptr = (void*)((uint32_t)&input_reg_params + param_descriptor->param_offset - 1);
                break;
            case MB_PARAM_COIL:
-               instance_ptr = (void*)(&coil_reg_params + param_descriptor->param_offset - 1);
+               instance_ptr = (void*)((uint32_t)&coil_reg_params + param_descriptor->param_offset - 1);
                break;
            case MB_PARAM_DISCRETE:
-               instance_ptr = (void*)(&discrete_reg_params + param_descriptor->param_offset - 1);
+               instance_ptr = (void*)((uint32_t)&discrete_reg_params + param_descriptor->param_offset - 1);
                break;
            default:
                instance_ptr = NULL;
                break;
        }
     } else {
-        instance_ptr = malloc((size_t)(param_descriptor->param_size));;
+        instance_ptr = malloc((size_t)(param_descriptor->param_size));
     }
     return instance_ptr;
 }

Kasuhikdas, 2019-10-03:
I have tried the changes but it still did not worked
Error is below

I (0) cpu_start: App cpu up.
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (270) heap_init: At 3FFB34A8 len 0002CB58 (178 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (282) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (289) heap_init: At 4008AD78 len 00015288 (84 KiB): IRAM
I (295) cpu_start: Pro cpu start user code
I (313) spi_flash: detected chip: generic
I (314) spi_flash: flash io: dio
W (314) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (324) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (336) uart: queue free spaces: 20
I (396) SENSE_MB: sense_modbus_get_characteristics: Found (6) characteristics in table.
I (396) sense_main: Found (6) characteristics in the table.
E (546) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x107).
E (546) sense_main: Characteristic (Data_channel_0) read value fail, err = 259 (ESP_ERR_INVALID_STATE).
E (706) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x107).
E (706) sense_main: Characteristic (Humidity_1) read value fail, err = 259 (ESP_ERR_INVALID_STATE).

alisitsyn, 2019-10-03:
According to your log the master example works correctly and issue is gone. It does not show the error "E (459393) MB_CONTROLLER_MASTER: mbc_master_get_parameter(103): Master interface is not correctly initialized." that was due to memory corruption.
You see the  'ESP_ERR_INVALID_STATE' because the timeout reading parameters from your slave device (ESP_ERR_TIMEOUT = 0x107). Later version of Modbus propagates more information about Modbus stack errors to higher level but it is clear even for now.
Please make sure your slave board: 1. Correctly flashed with modbus_slave example and its pin configuration matches the configuration in modbus_master example. 2. The RS485 line drivers connected correctly (see the documentation). 3. Please make sure your parameter table in modbus_master  example in main/device_params.c file corresponds to parameters in slave example (modbus_slave/main/deviceparams.h) . The master example table is designed to work with several modbus slaves in the segment with different addresses (not all parameters can be read from slave example device).

Alvin1Zhang, 2020-05-28:
@Kasuhikdas Thanks for reporting and sorry for slow turnaround. The fix has been available at https://github.com/espressif/esp-idf/commit/65c67faa6a84f63dc740111adaabea46ffc3bf89, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-28 by Alvin1Zhang