espressif/esp-idf #4141: Persistent provisioner

jkockx, 2019-10-01:
Hello, I'm working on a project including a BLE mesh provisioner. I would like to be able to reboot the provisioner without losing all BLE mesh provisioning data. However, I've found that the esp-idf doesn't support this functionality (yet?) to store the data in nvs.
Is this to be added in the near future and if not so, are there work arounds for me to implement myself?

WCCWCC, 2019-10-08:
Hi jkockx,
* The provisioner needs to store a lot of information, and we are implementing this feature.
* The specific time has not yet been fixed.

jkockx, 2019-10-15:
Thanks for your reply @WCCWCC ,
Even though you're unable to provide a specific date when this will be implemented, would you have any sight on what measure of time you are speaking of. Will this be available one of the following months or will this have to wait more than a year.
That would be very important for me to know as my project pretty much relies on this persistent provisioning.

Alvin1Zhang, 2019-10-25:
Hi @jkockx Thanks for reporting, since this is duplicate with [4242](https://github.com/espressif/esp-idf/issues/4242), will close this ticket and let us keep tracking in [4242](https://github.com/espressif/esp-idf/issues/4242). Thanks.

closed 2019-10-25 by Alvin1Zhang