espressif/esp-idf #4143: Compiler error in driver/can.c if optimized for performance

zinke-ct-video, 2019-10-02:
Hello,

the latest IDF (4.1) won't compile anymore, if I select compiler optimization = performance and assertions = disabled.
If I select compiler optimization = size then everything is fine.

The mentioned error is below: 

C:/Users/user258/esp/esp-idf/components/driver/can.c: In function 'can_intr_handler_main':
C:/Users/user258/esp/esp-idf/components/driver/can.c:281:9: error: '((unsigned char*)&frame.<Ue318>.<U4ec8>)[0]' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     if (frame->self_reception) {
         ^~~~~
cc1.exe: some warnings being treated as errors

This issue is also very similar to #3591 

Thanks for any help!

zinke-ct-video, 2019-10-02:
I found a temporary solution If somebody needs an immediate fix: Add the line
"-Wno-error=maybe-uninitialized" after line 106 in esp-idf/tools/cmake/build.cmake

Alvin1Zhang, 2019-10-08:
@zinke-ct-video Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

zinke-ct-video, 2019-10-17:
Here are some more information:
esp-idf version is v4.1-dev-474-g2e6398aff
My sdkconfig can be found [here](https://gist.github.com/zinke-ct-video/51218202dc457be74d4c323e0047ebeb).

And here are the full command line prints:

C:\Users\user258\esp\projects\webserver-poe-switch>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\user258\esp\projects\webserver-poe-switch\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project version: 0.2.5
-- Building ESP-IDF components for target esp32
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/Users/user258/esp/projects/webserver-poe-switch/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/user258/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/user258/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/user258/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/user258/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/user258/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/user258/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/user258/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns minimal-ip-stack mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread reader-writer-lock sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/user258/esp/esp-idf/components/app_trace C:/Users/user258/esp/esp-idf/components/app_update C:/Users/user258/esp/esp-idf/components/asio C:/Users/user258/esp/esp-idf/components/bootloader C:/Users/user258/esp/esp-idf/components/bootloader_support C:/Users/user258/esp/esp-idf/components/bt C:/Users/user258/esp/esp-idf/components/cbor C:/Users/user258/esp/esp-idf/components/coap C:/Users/user258/esp/esp-idf/components/console C:/Users/user258/esp/esp-idf/components/cxx C:/Users/user258/esp/esp-idf/components/driver C:/Users/user258/esp/esp-idf/components/efuse C:/Users/user258/esp/esp-idf/components/esp-tls C:/Users/user258/esp/esp-idf/components/esp32 C:/Users/user258/esp/esp-idf/components/esp_adc_cal C:/Users/user258/esp/esp-idf/components/esp_common C:/Users/user258/esp/esp-idf/components/esp_eth C:/Users/user258/esp/esp-idf/components/esp_event C:/Users/user258/esp/esp-idf/components/esp_gdbstub C:/Users/user258/esp/esp-idf/components/esp_http_client C:/Users/user258/esp/esp-idf/components/esp_http_server C:/Users/user258/esp/esp-idf/components/esp_https_ota C:/Users/user258/esp/esp-idf/components/esp_https_server C:/Users/user258/esp/esp-idf/components/esp_local_ctrl C:/Users/user258/esp/esp-idf/components/esp_ringbuf C:/Users/user258/esp/esp-idf/components/esp_rom C:/Users/user258/esp/esp-idf/components/esp_websocket_client C:/Users/user258/esp/esp-idf/components/esp_wifi C:/Users/user258/esp/esp-idf/components/espcoredump C:/Users/user258/esp/esp-idf/components/esptool_py C:/Users/user258/esp/esp-idf/components/expat C:/Users/user258/esp/esp-idf/components/fatfs C:/Users/user258/esp/esp-idf/components/freemodbus C:/Users/user258/esp/esp-idf/components/freertos C:/Users/user258/esp/esp-idf/components/heap C:/Users/user258/esp/esp-idf/components/idf_test C:/Users/user258/esp/esp-idf/components/jsmn C:/Users/user258/esp/esp-idf/components/json C:/Users/user258/esp/esp-idf/components/libsodium C:/Users/user258/esp/esp-idf/components/log C:/Users/user258/esp/esp-idf/components/lwip C:/Users/user258/esp/projects/webserver-poe-switch/main C:/Users/user258/esp/esp-idf/components/mbedtls C:/Users/user258/esp/esp-idf/components/mdns C:/Users/user258/esp/projects/webserver-poe-switch/components/minimal-ip-stack C:/Users/user258/esp/esp-idf/components/mqtt C:/Users/user258/esp/esp-idf/components/newlib C:/Users/user258/esp/esp-idf/components/nghttp C:/Users/user258/esp/esp-idf/components/nvs_flash C:/Users/user258/esp/esp-idf/components/openssl C:/Users/user258/esp/esp-idf/components/partition_table C:/Users/user258/esp/esp-idf/components/protobuf-c C:/Users/user258/esp/esp-idf/components/protocomm C:/Users/user258/esp/esp-idf/components/pthread C:/Users/user258/esp/projects/webserver-poe-switch/components/reader-writer-lock C:/Users/user258/esp/esp-idf/components/sdmmc C:/Users/user258/esp/esp-idf/components/soc C:/Users/user258/esp/esp-idf/components/spi_flash C:/Users/user258/esp/esp-idf/components/spiffs C:/Users/user258/esp/esp-idf/components/tcp_transport C:/Users/user258/esp/esp-idf/components/tcpip_adapter C:/Users/user258/esp/esp-idf/components/ulp C:/Users/user258/esp/esp-idf/components/unity C:/Users/user258/esp/esp-idf/components/vfs C:/Users/user258/esp/esp-idf/components/wear_levelling C:/Users/user258/esp/esp-idf/components/wifi_provisioning C:/Users/user258/esp/esp-idf/components/wpa_supp [... cut ...] oader/subproject/components/micro-ecc C:/Users/user258/esp/esp-idf/components/partition_table C:/Users/user258/esp/esp-idf/components/soc C:/Users/user258/esp/esp-idf/components/spi_flash C:/Users/user258/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/user258/esp/projects/webserver-poe-switch/build/bootloader
ninja: no work to do.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Do you need any other info?

closed 2019-10-25 by igrr