espressif/esp-idf #4144: Compiler error in spiffs/src/spiffs_nucleus.c if optimized for performance

zinke-ct-video, 2019-10-02:
Hello,

similiar to #4143 the following compiler error occurs, if optimization = performance (-O2) is selected in menuconfig.

C:/Users/user258/esp/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c: In function 'spiffs_object_create':
C:/Users/user258/esp/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:942:3: error: 'strncpy' specified bound 32 equals destination size [-Werror=stringop-truncation]
   strncpy((char*)oix_hdr.name, (const char*)name, SPIFFS_OBJ_NAME_LEN);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors

If somebody needs an immediate fix: Add the line
"-Wno-error=stringop-truncation" after line 106 in esp-idf/tools/cmake/build.cmake

Alvin1Zhang, 2019-10-08:
@zinke-ct-video Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

ralfoide, 2019-10-13:
$ cd $IDF_PATH
$ git log
===> commit 4dac7c7df885adaa86a5c79f2adeaf8d68667349 (HEAD -> master, origin/master, origin/HEAD)

$ cd to any other place and try an example
$ cp -rv $IDF_PATH/examples/wifi/iperf .
$ cd iperf
$ idf.py menuconfig
==> compiler options > optimization level > performance (-O2)

generated sdkconfig: https://gist.github.com/ralfoide/9aa6f22deb0c95886a2dd69303d3a9d5

$ idf.py build
[...]
/home/ralf/esp/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c: In function 'spiffs_object_create':
/home/ralf/esp/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:942:3: error: 'strncpy' specified bound 32 equals destination size [-Werror=stringop-truncation]
   strncpy((char*)oix_hdr.name, (const char*)name, SPIFFS_OBJ_NAME_LEN);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ralf/esp/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c: In function 'spiffs_object_update_index_hdr':
/home/ralf/esp/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1005:5: error: 'strncpy' specified bound 32 equals destination size [-Werror=stringop-truncation]
     strncpy((char*)objix_hdr->name, (const char*)name, SPIFFS_OBJ_NAME_LEN);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Workaround:
$ idf.py menuconfig
==> compiler options > optimization level > debug (-Og)
Compiles just fine.

closed 2019-10-25 by igrr