espressif/esp-idf #4148: Data storing in Flash of ESP32

surekhaibot, 2019-10-03:
Hi,
I am write and read data using key based method in NVS flash of ESP32 . Using SPIFLASH i am storing the data in text file. Let me know if possible data storing directly address based from so and so address to so and so address like EEPROM. If possible can you share the reference firmware not for arduino.

stickbreaker, 2019-10-03:
[SPI Flash Access API](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spi_flash.html)

Alvin1Zhang, 2019-10-08:
@stickbreaker Thanks for the notes. @surekhaibot Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

mahavirj, 2019-10-11:
@surekhaibot Please refer to examples provided at https://github.com/espressif/esp-idf/tree/master/examples/storage for reference. Please use esp32.com forum for how-to questions.

closed 2019-10-11 by mahavirj