espressif/esp-idf #4150: [BLE Mesh] what is the difference between different client and server examples?

John-nx, 2019-10-03:
Hello,

Can someone plz tell me what is the difference between the following examples?
P.S I have already read the "tutorial"!

The difference between these two:

-between ble_mesh_fast_prov_client
-ble_mesh_client_model

and the difference between these three:

-ble_mesh_fast_prov_server
-ble_mesh_node
-ble_mesh_node under the ble_mesh_console example 

Thanks

WCCWCC, 2019-10-08:
Hi John-nx,
*  For fast provision solution, we provide two demo. The demo realizes configuring 100 devices into the network within 60s :   `ble_mesh_fast_prov_client` and `ble_mesh_fast_prov_server`

*  ble_mesh_node :  Implemented the basic functions of nodes in the mesh network.

*  ble_mesh_console:  Controlling the provisioner and node through command line interaction

Alvin1Zhang, 2019-10-10:
@John-nx Thanks for reporting, has your issue been resolved? Thanks.

John-nx, 2019-10-11:
Hi @Alvin1Zhang and @WCCWCC ,

Thanks for your responses.
Firstly, my question is mostly about, how these different examples function differently after being provisioned and as a node in a mesh network. 

Secondly, please note that there are major problems with the "ble_mesh_client_model" example
addressed here: https://github.com/espressif/esp-idf/issues/4059

and also the problem common problem between "ble_mesh_client_model" and "ble_mesh_node" which as losing the provisioning information after a power reset. 
addressed here: https://github.com/espressif/esp-idf/issues/4074

I wanted to also check whether the mentioned issues exist int he all examples mentioned in the question or not.

Thanks

Campou, 2019-10-11:
Hi @John-nx 

Firstly:
* **ble_mesh_fast_prov_client** is an example of a Provisioner which tries to provision an unprovisioned device and start fast provisioning.
* **ble_mesh_client_model** is an example of a node which owns a Generic OnOff Client model and a Generic OnOff Server model.

* **ble_mesh_fast_prov_server** is an example of a node which can be provisioned by **ble_mesh_fast_prov_client** and changed into a temporary Provisioner.
* **ble_mesh_node** is an example of a node which owns a Generic OnOff Server model.
* **ble_mesh_node** under the **ble_mesh_console** example is used just for QA test.

Secondly:
* The ble_mesh_client_model example will be modified properly later.

And I have reproduced the issue in https://github.com/espressif/esp-idf/issues/4074, will fix it in release/v4.0 branch.

Thanks.

John-nx, 2019-10-11:
Thanks @Campou.

closed 2019-10-11 by John-nx